The-OpenROAD-Project/OpenLane #1475: Flow fails during writing powered verilog

vijayank88, 2022-11-08:
## Description

This MPW `user_project_wrapper` design: https://github.com/The-OpenROAD-Project/caravel_user_project/blob/Procenne-Digital-Design/secure-memory/openlane/user_project_wrapper/config.tcl

Previously its worked fine. Latest OpenLane throws error and failed to complete writing powered verilog stage.

## Expected behavior

Complete the flow.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: a0b19fa68b79eff3e29b674106b06d20aab637f1
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (grafted, HEAD, tag: 2022.11.02)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)

```

## Reproduction Material

Its quite confusing to generate the test case. 
Please run this MPW CI design by following below steps:

```
git clone https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git
cd OpenLane-MPW-CI
```
Edit and save  `openlane-tag` value to `2022.11.02`

```
./scripts/setup-ci.sh
./scripts/run-design.sh secure-memory secure-memory-wrapper
```

## Logs

```
[STEP 27]
[INFO]: Writing Powered Verilog (logs: logs/signoff/27-write_powered_def.log, logs/signoff/27-write_powered_verilog.log)...
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/power_utils.py write_powered_def --output /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/tmp/signoff/26-user_project_wrapper.p.def --input-lef /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/tmp/merged.nom.lef --power-port vccd1 --ground-port vssd1 --powered-netlist /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/tmp/synthesis/pg_define.v /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/results/routing/user_project_wrapper.def |& tee /dev/null /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/logs/signoff/27-write_powered_def.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
[INFO ODB-0127] Reading DEF file: /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/tmp/synthesis/pg_define.intermediate.def
[WARNING ODB-0148] error: Cannot open DEF file /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.08_10.46.30/tmp/synthesis/pg_define.intermediate.def
[WARNING ODB-0129] Error: Failed to read DEF file
Top-level design name: user_project_wrapper
Found default power net 'vccd1'
Found default ground net 'vssd1'
Found 4 power ports.
Found 4 ground ports.
Modified power connections of 3/3 cells.
child process exited abnormally

[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Skipping Tap/Decap Insertion.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -exit -no_init -python $::env(SCRIPTS_DIR)/odbpy/power_utils.py write_powered_def --output $arg_values(-output_def)  --..."
    (procedure "write_powered_verilog" line 55)
    invoked from within
"write_powered_verilog -output_nl $netlist_name -output_pnl $powered_netlist_name -output_def $powered_def_name"
    (procedure "run_lvs" line 13)
    invoked from within
"run_lvs"
    (procedure "run_lvs_step" line 10)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 411)
```

donn, 2022-11-09:
Gah... The worst script in the codebase

vijayank88, 2022-12-01:
With OpenLane tag: `2022.11.30` the log as follows:
```
[STEP 27]

[INFO]: Writing Powered Verilog (logs: logs/signoff/27-write_powered_def.log, logs/signoff/27-write_powered_verilog.log)...

[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/power_utils.py write_powered_def --output /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.30_22.19.27/tmp/signoff/26-user_project_wrapper.p.def --input-lef /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.30_22.19.27/tmp/merged.nom.lef --power-port vccd1 --ground-port vssd1 --powered-netlist /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.30_22.19.27/tmp/synthesis/pg_define.v /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.30_22.19.27/results/routing/user_project_wrapper.def |& tee /dev/null /openlane/designs/secure-memory-wrapper/runs/RUN_2022.11.30_22.19.27/logs/signoff/27-write_powered_def.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:

  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/odbpy/power_utils.py", line 222, in write_powered_def
    assert power.name == reader.name
AttributeError: 'OdbReader' object has no attribute 'name'
child process exited abnormally
```

Patarimi, 2023-01-10:
Hi everyone,
I get the same issue (the power_utils.py one). Reading the file 27-write_powered_verilog.log, I found this error :
`openroad>         read_verilog /home/mpotereau/DigitalFlowTest/gf_spi_test/openl 
ane/user_project_wrapper/runs/23_01_10_16_10/tmp/synthesis/pg_define.v`
`[ERROR STA-0164] /home/mpotereau/DigitalFlowTest/gf_spi_test/openlane/user_project_wrapper/runs/23_01_10_16_10/tmp/synthesis/pg_define.v line 561, syntax error, unexpected $undefined, expecting ';' or ','`

I look a the line 561 of the pg_define.v file and found :
`defparam spi_reg0.REG_ADDR = 32'sb00000000000000000000000000000000;`
In my opinion, this is a faulty translation of the original line in my code :
`spi_register #(8,3,0) spi_reg1 (`
I rewrite my code as :
`spi_register #(4'b1000,4'b0011,4'b0000) spi_reg1 (`
and the step 27 is now successful.

I don't know what generate the pg_define.v file, but I think the issue is there and not in the power_utils.py part.

@vijayank88 Do you use parameter and do you pass them in decimal format ?

closed 2023-02-13 by donn