The-OpenROAD-Project/OpenLane #1479: Update volare for GF180 PDK

vijayank88, 2022-11-10:
### Description

I've used following command GF180 PDK installation:
```
export PDK_FAMILY=gf180mcu
export PDK=gf180mcuC
cd OpenLane && make
```
Its has successfully installed GF180 pdk and able to complete `make test`
But latest PDK commit not updated from https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/commit/d3798681284ea9b96bfd57bfda20948ac86767e4 as it resolves fanout issue discussed in https://github.com/The-OpenROAD-Project/OpenROAD/issues/2339

### Expected Behavior

Update GF180 PDK volare version to https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/commit/d3798681284ea9b96bfd57bfda20948ac86767e4

`default_fanout_load : 1.0000000000 ; ` to be added in liberty files.

### Environment report

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 8b391c09ae913c7b4e5c64a29a352ef0e9df41e0
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
Failed to verify sky130A.
---
Git Log (Last 3 Commits)

8b391c0 2022-11-09T04:19:46+02:00 Change Issue Templates to Issue Forms (#1476) - Mohamed Gaber -  (HEAD -> master, tag: 2022.11.10, origin/master, origin/HEAD)
ddcc5c9 2022-11-09T04:14:39+02:00 [BOT] Update openroad_app (#1466) - Openlane Bot -  ()
a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (tag: 2022.11.02)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

%./flow.tcl -design aes

### Relevant log output

```shell
Fanout issue log: 16-global.log

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (propagated)
                          8.00    8.00 ^ input external delay
                  1.14    0.60    8.60 ^ reset_n (in)
     2    0.05                           reset_n (net)
                  1.14    0.00    8.60 ^ input10/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  2.13    1.07    9.67 ^ input10/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   450    2.02                           net10 (net)
```

vijayank88, 2022-11-10:
@proppy fyi

proppy, 2022-11-10:
@vijayank88 I believe that will be fixed after merging https://github.com/google/gf180mcu-pdk/pull/91.

The PDK will have to be rebuilt and published manually, until https://github.com/The-OpenROAD-Project/OpenLane/issues/1377 is addressed.

closed 2022-11-23 by vijayank88