The-OpenROAD-Project/OpenLane #1481: Problem with passing parameters of config.tcl

hello-eternity, 2022-11-11:
### Description

When I passed the parameters about verilog_files in config.tcl, I found that the absolute path, or the same way of passing parameters as in json caused the file to not be found, I want to know the reason for this.
I test it on picorv32a, the config.tcl is follow: （absolute path version）
```
set  ::env(DESIGN_NAME)  "picorv32"
set  ::env(VERILOG_FILES)  "/work/OpenLane/designs/picorv32a/src/picorv32a.v"
set  ::env(CLOCK_PORT) "clk"
set  ::env(CLOCK_NET) "clk" 
set  ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) 1
set  ::env(CLOCK_PERIOD) 24
```
and the json verison:
```
set  ::env(VERILOG_FILES)  "dir::src/picorv32a.v"
```
will cause error
```
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/picorv32a/runs/RUN_2022.11.11_04.18.29/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
ERROR: Can't open input file `/work/OpenLane/designs/picorv32a/src/picorv32a.v' for reading: No such file or directory
child process exited abnormally
```

but this parameter in json works well:
`"VERILOG_FILES": "dir::src/picorv32a.v",`

besides there is another sovle method in tcl file:
```
set  script_dir $::env(DESIGN_DIR)
set  ::env(VERILOG_FILES)  "$script_dir/src/picorv32a.v"

```
This is due to file search within docker，But it's not very convenient.

### Expected Behavior

Fix the way the file address is passed

### Environment report

```shell
Kernel: Linux v4.18.0-240.1.1.el8_3.x86_64
Distribution: centos 8
Python: v3.6.8 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: 8b391c09ae913c7b4e5c64a29a352ef0e9df41e0
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

8b391c0 2022-11-09T04:19:46+02:00 Change Issue Templates to Issue Forms (#1476) - Mohamed Gaber -  (grafted, HEAD -> master, tag: 2022.11.10, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

use the official picorv32a and change the config.tcl

### Relevant log output

```shell
on top
```

vijayank88, 2022-11-11:
@hello-eternity 
You've to adopt to docker environment, as absolute path not recommended to use.

hello-eternity, 2022-11-11:
Fine, thank you for answer

closed 2022-11-11 by hello-eternity