The-OpenROAD-Project/OpenLane #1483: Applying DEF template fails

vijayank88, 2022-11-11:
### Description

This is MPW CI design passed successfully with `2022.11.02` and fails with `2022.11.10` and above. 

Between only `OpenROAD` commit got changed. 

MPW CI user repo [config.tcl](https://github.com/thkim2031/efabless_mpw/blob/main/openlane/user_project_wrapper/config.tcl)

CI logs found [here](https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-small-4/detail/PR-100/1/pipeline/291)

### Expected Behavior

Complete IO placement stage.

### Environment report

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 68898e65843d8c9c7fa5fdf609d2ab3ac96b1081
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

68898e6 2022-11-10T14:39:22+02:00 Move flow control from atomic functions to flow functions (#1453) - Mohamed Gaber -  (HEAD -> master, tag: 2022.11.11, origin/master, origin/HEAD)
86615f8 2022-11-10T14:28:56+02:00 Yosys Script Enhancements (#1446) - Mohamed Gaber -  ()
8b391c0 2022-11-09T04:19:46+02:00 Change Issue Templates to Issue Forms (#1476) - Mohamed Gaber -  (tag: 2022.11.10)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Due to file size limit uploaded design directory [here](https://slack-files.com/T01699QAZBQ-F04AWLHHN5P-70569fea90)
Unzip `mpw7_AI_chip.zip` and place it `OpenLane/designs/`
%make mount
%./flow.tcl -design mpw7_AI_chip

### Relevant log output

```shell
[STEP 4]
[INFO]: Running IO Placement...
[INFO]: Applying DEF template...
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/apply_def_template.py --def-template /openlane/designs/mpw7_AI_chip/openlane/user_project_wrapper/fixed_dont_change/user_project_wrapper.def --input-lef /openlane/designs/mpw7_AI_chip/runs/RUN_2022.11.11_05.21.02/tmp/merged.nom.lef --output-def /openlane/designs/mpw7_AI_chip/runs/RUN_2022.11.11_05.21.02/tmp/floorplan/4-io.def --output /openlane/designs/mpw7_AI_chip/runs/RUN_2022.11.11_05.21.02/tmp/floorplan/4-io.odb /openlane/designs/mpw7_AI_chip/runs/RUN_2022.11.11_05.21.02/tmp/floorplan/4-io.odb |& tee /dev/null /openlane/designs/mpw7_AI_chip/runs/RUN_2022.11.11_05.21.02/logs/floorplan/4-apply_def_template.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0099] error: netlist component (mprj) is not defined
[WARNING ODB-0149] DEF parser returns an error!
child process exited abnormally

[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -exit -no_init -python {*}$args  --input-lef $::env(MERGED_LEF)  --output-def $arg_values(-output_def)  --output $arg_v..."
    (procedure "manipulate_layout" line 19)
    invoked from within
"manipulate_layout $::env(SCRIPTS_DIR)/odbpy/apply_def_template.py -indexed_log $log -output_def $::env(CURRENT_DEF) --def-template $::env(FP_DEF_TEMPL..."
    (procedure "apply_def_template" line 7)
    invoked from within
"apply_def_template"
    (procedure "run_floorplan" line 38)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 411)
```

dineshannayya, 2022-11-14:
Is this issue fixed ?

vijayank88, 2022-11-14:
@maliberty fyi

dineshannayya, 2022-11-14:
@maliberty This issue will impact all the digital MPW project top-level integration,Hope issue fix is in higher priority list.

maliberty, 2022-11-14:
Ok I'll give it higher priority

maliberty, 2022-11-14:
I don't see why this should ever have worked.  You have a template def (user_project_wrapper.def) with

COMPONENTS 1 ;
    - mprj user_proj_example + FIXED ( 1175000 1690000 ) N ;
END COMPONENTS

but no lef containing mprj.  What are you expecting to have happen here?

dineshannayya, 2022-11-14:
This reference/fixed def given by efabless team (https://github.com/efabless/caravel_user_project/blob/main/openlane/user_project_wrapper/fixed_dont_change/user_project_wrapper.def) and look like it's generated based on caravel_user_project.

In user project, these hierarchy will not there, Mostly previous apply_def_template logic ignoring these component hierarchy details and picking only IO location.

maliberty, 2022-11-14:
Ok but I don't see this as an OR issue.  We might have been more lax before but the data is wrong and OR is telling you how.  I don't know what the goal of this script is so I don't know if there is a better method.  At a minimum you could make a dummy LEF for the missing macro.

closed 2022-11-14 by maliberty

vijayank88, 2022-11-15:
@dineshannayya 
Have you facing this issue with any of wrapper design?
Only three MPW design am facing this issue with latest OpenLane/OpenROAD commit.
If you're also facing the same, we need to raise this caravel team to correct LEF/DEF for wrapper file.

dineshannayya, 2022-11-15:
@vijayank88 yes, my user_project_wrapper level run are failing with this issue. My understanding most of MPW CI user_project_wrapper run will fail with this issue.
Locally i have deleted COMPONENTS and NETS definition in the def.  Now openlane and pre-check flow are clean for me

vijayank88, 2022-11-15:
@donn Is this issue can be fixed by caravel team or need to raise GH issue on there repo?

donn, 2022-11-15:
@vijayank88 Please raise an issue on their repo.

dineshannayya, 2022-11-15:
I feel desired functionality for apply_def_template.py script is just pick only IO location information and ignore COMPONENT/NET details from the .def files and apply it to user project.

vijayank88, 2022-11-15:
@dineshannayya 
Your suggested changes applied in caravel repo: https://github.com/efabless/caravel_user_project/commit/40af4e002980e38704983475d5c0bed6daf942db