The-OpenROAD-Project/OpenLane #1486: Crash in floorplanning when using asap7

yupferris, 2022-11-11:
### Description

When attempting to use the flow on any design using [asap7](https://github.com/The-OpenROAD-Project/asap7) (using [this version of volare](https://github.com/efabless/volare/pull/29) to install/activate as well as https://github.com/The-OpenROAD-Project/OpenLane/pull/1484 and https://github.com/The-OpenROAD-Project/OpenLane/pull/1485), I get a crash in `openroad` when floorplanning tries to insert tie cells.

To reproduce:

### 1. Use a local volare build with [this patch](https://github.com/efabless/volare/pull/29) to enable asap7 support and build/enable the PDK

```bash
volare build -j$(nproc) --pdk asap7 --clear-build-artifacts --sram -l all 1967b4b2eb2c32b33f948bad4891b8cebd0107f0
volare enable --pdk asap7 1967b4b2eb2c32b33f948bad4891b8cebd0107f0
```
Note that the hashes here simply refer to the latest commit in the `OpenROAD-flow-scripts` repo.

### 2. Try to run (any) design through the OpenLane flow
```bash
PDK=asap7 STD_CELL_LIBRARY=asap7sc7p5t STD_CELL_LIBRARY_OPT=asap7sc7p5t ./flow.tcl -design gcd
```

### Expected Behavior

I wanted to get a simple design through the OpenLane flow using asap7 (or at the very least, past floorplanning, where I could continue to debug further issues I may encounter).

### Environment report

```shell
Kernel: Linux v5.15.0-48-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: floorplanning-crash-dev
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a3bbfe0 2022-11-11T14:40:46+01:00 Use LIB_SYNTH instead of LIB_SYNTH_COMPLETE in some OpenROAD scripts - Jake Taylor -  (HEAD -> floorplanning-crash, ascenium/floorplanning-crash)
7f7ebc1 2022-11-11T14:31:37+01:00 Include all PDK lib files when merging - Jake Taylor -  (ascenium/include-all-pdk-cells, include-all-pdk-cells)
a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (grafted, tag: 2022.11.02, origin/master, origin/HEAD, master)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9990745/reproducible.tar.gz)

### Relevant log output

```shell
<earlier messages omitted>

Stack trace:
 0# 0x0000000000CA52F7 in openroad
 1# 0x00007FA6FEBA1400 in /lib64/libc.so.6
 2# ifp::InitFloorplan::insertTiecells(odb::dbMTerm*, std::string const&) in openroad
 3# 0x0000000000CB0B18 in openroad
 4# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
 5# 0x00007FA70193336C in /lib64/libtcl8.5.so
 6# TclObjInterpProcCore in /lib64/libtcl8.5.so
 7# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
 8# 0x00007FA70193336C in /lib64/libtcl8.5.so
 9# 0x00007FA70193B647 in /lib64/libtcl8.5.so
10# TclEvalObjEx in /lib64/libtcl8.5.so
11# 0x00007FA70197627F in /lib64/libtcl8.5.so
12# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
13# 0x00007FA70193336C in /lib64/libtcl8.5.so
14# 0x00007FA70193B647 in /lib64/libtcl8.5.so
15# TclEvalObjEx in /lib64/libtcl8.5.so
16# 0x00007FA7018F61D0 in /lib64/libtcl8.5.so
17# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
18# 0x00007FA70193336C in /lib64/libtcl8.5.so
19# 0x00007FA70193B647 in /lib64/libtcl8.5.so
20# TclEvalObjEx in /lib64/libtcl8.5.so
21# 0x00007FA7018F9F00 in /lib64/libtcl8.5.so
22# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
23# 0x00007FA70193336C in /lib64/libtcl8.5.so
24# TclObjInterpProcCore in /lib64/libtcl8.5.so
25# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
26# 0x00007FA70193336C in /lib64/libtcl8.5.so
27# TclObjInterpProcCore in /lib64/libtcl8.5.so
28# 0x00007FA7018EEEB2 in /lib64/libtcl8.5.so
29# 0x00007FA7018EFF1E in /lib64/libtcl8.5.so
30# Tcl_EvalEx in /lib64/libtcl8.5.so
31# Tcl_Eval in /lib64/libtcl8.5.so
32# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
33# ord::tclAppInit(Tcl_Interp*) in openroad
34# Tcl_Main in /lib64/libtcl8.5.so
35# main in openroad
36# __libc_start_main in /lib64/libc.so.6
37# 0x0000000000CA05E7 in openroad
[ERROR]: during executing openroad script /openlane/scripts/openroad/floorplan.tcl
[ERROR]: Log: designs/gcd/runs/RUN_2022.11.11_13.58.10/logs/floorplan/3-initial_fp.log
[ERROR]: Last 10 lines:
child killed: segmentation violation
```

donn, 2022-11-11:
ASAP7 is so broken atm unfortunately- volare applies some patches on the fly to the input data from OpenROAD Flow Scripts, but we've been too busy with GF180 to file issues...

yupferris, 2022-11-14:
First of all, thanks for the quick response!

I figured that this may be a bit futile at this point as it's not officially supported, but wanted to try anyways. And, actually, I was able to get `gcd` through the entire flow with some hacks. Specifically:

- https://github.com/The-OpenROAD-Project/OpenROAD/pull/2491 in OpenROAD works around the above crash in floorplanning
- Similar to #1485, several "LIB" environment variables refer to multiple files with asap7, which breaks scripts further on in the flow. #1490 (which currently includes #1485) substitutes `LIB_SYNTH` for all of these, but this is surely not correct; they should probably be set up properly instead.
- `GPL_CELL_PADDING` and `DPL_CELL_PADDING` are both not set in asap7's config (and it seems like they should be; perhaps volare should do this?) and must be provided.

With the above changes, the following appears to work:

`GPL_CELL_PADDING=0 DPL_CELL_PADDING=0 PDK=asap7 STD_CELL_LIBRARY=asap7sc7p5t STD_CELL_LIBRARY_OPT=asap7sc7p5t ./flow.tcl -design gcd -gui`

Curiously, I seem to be unable to get timing information from the GUI. Is this to be expected?

What would be a good next step? Should I try to incorporate these changes more properly? Should I raise separate issue(s)? Or is there a better way/someone I could talk to about it? My team and I would be more than happy to contribute.

maliberty, 2022-11-14:
Note that ORFS does support asap7 so these are likely to be setup issues in OL.

oharboe, 2022-11-15:
With https://github.com/The-OpenROAD-Project/OpenROAD/pull/2495, I get the error message below:

```
make PDK_ROOT="/home/oyvind/.volare/" PDK="asap7" STD_CELL_LIBRARY="asap7sc7p5t" STD_CELL_LIBRARY_OPT="asap7sc7p5t" mount
#
./flow.tcl -design gcd
[deleted]
[ERROR]: during executing openroad script /openlane/scripts/openroad/floorplan.tcl
[ERROR]: Log: designs/gcd/runs/RUN_2022.11.15_08.47.47/logs/floorplan/3-initial_fp.log
[ERROR]: Last 10 lines:
[INFO]: Setting output delay to: 84.0
[INFO]: Setting input delay to: 84.0
[INFO]: Setting load to: 0.0046105700000000005
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[INFO]: Setting timing derate to: 0.5 %
[INFO IFP-0001] Added 52 rows of 260 sites.
[ERROR IFP-0039] Liberty cell or port TIELOx1_ASAP7_75t_R/L not found.
Error: floorplan.tcl, 70 IFP-0039
child process exited abnormally
```

Grepping /home/oyvind/.volare, I find:

```
config.tcl:set ::env(SYNTH_TIELO_PORT) {TIELOx1_ASAP7_75t_R L}
```

maliberty, 2022-11-15:
The cell exists in asap7sc7p5t_SIMPLE_RVT_TT_nldm_201020.lib so OL must not be loading all the .lib

rovinski, 2022-11-15:
> And the macro that does appear to have an "L" pin:
> 
> ```
> MACRO TIELOx1_ASAP7_75t_R
>   CLASS CORE ;
>   ORIGIN 0 0 ;
>   FOREIGN TIELOx1_ASAP7_75t_R 0 0 ;
>   SIZE 0.162 BY 0.27 ;
>   SYMMETRY X Y ;
>   SITE asap7sc7p5t ;
>   PIN L
>     DIRECTION OUTPUT ;
>     USE SIGNAL ;
>     PORT
>       LAYER M1 ;
>         RECT 0.067 0.182 0.144 0.2 ;
>         RECT 0.126 0.027 0.144 0.2 ;
>         RECT 0.094 0.027 0.144 0.045 ;
>     END
>   END L
>   PIN VDD
>     DIRECTION INOUT ;
>     USE POWER ;
>     SHAPE ABUTMENT ;
>     PORT
>       LAYER M1 ;
>         RECT 0 0.261 0.162 0.279 ;
>     END
>   END VDD
>   PIN VSS
>     DIRECTION INOUT ;
>     USE GROUND ;
>     SHAPE ABUTMENT ;
>     PORT
>       LAYER M1 ;
>         RECT 0 -0.009 0.162 0.009 ;
>     END
>   END VSS
>   OBS
>     LAYER M1 ;
>       RECT 0.018 0.225 0.068 0.243 ;
>       RECT 0.018 0.124 0.036 0.243 ;
>       RECT 0.018 0.124 0.095 0.142 ;
>   END
> END TIELOx1_ASAP7_75t_R
> ```

The PDK should probably mark this cell as `CORE TIELOW` instead of just `CORE`. I don't think OpenROAD cares, but it would be good to be correct.

rovinski, 2022-11-15:
Actually, this is not present upstream. This must be an old version of the PDK.
https://github.com/The-OpenROAD-Project/asap7/blob/2cf04dbd8cd4f2d8ca4923196d866616c4b637bc/asap7sc7p5t_28/LEF/asap7sc7p5t_28_L_1x_220121a.lef#L17170

yupferris, 2022-11-16:
I was able to hack this again. In #1485, if we use `LIB_SYNTH_MERGED` instead of `LIB_SYNTH` (which replaced `LIB_SYNTH_COMPLETE`) the tie cells are found in floorplanning and it appears to work as expected. Granted I don't know if that's a proper fix or not (especially after 77fd531edbbdbb17fc16c1631effd7c5b377a803 by @donn) as it may be better to have the scripts deal with multiple .lib files specified by `LIB_SYNTH_COMPLETE` instead, but if I understand correctly `LIB_SYNTH_MERGED` should at least have everything the rest of the flow needs.

yupferris, 2023-01-03:
Closing due to inactivity.

closed 2023-01-03 by yupferris

hanm2019, 2023-03-28:
> Closing due to inactivity.

I'm curious whether OpenLane runs with ASAP7?

oharboe, 2023-03-28:
> > Closing due to inactivity.
> 
> I'm curious whether OpenLane runs with ASAP7?

I created a few PRs, but there was no reactions, we are using ORFS now. ASAP7 works there.

hanm2019, 2023-03-28:
> > > Closing due to inactivity.
> > 
> > 
> > I'm curious whether OpenLane runs with ASAP7?
> 
> I created a few PRs, but there was no reactions, we are using ORFS now. ASAP7 works there.

ORFS is [OpenROAD-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts)?

maliberty, 2023-03-28:
> ORFS is [OpenROAD-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts)?

Yes