The-OpenROAD-Project/OpenLane #1488: Synthesis Fails Every time on both (Mixed Vhdl + V)

naveedabbasi, 2022-11-12:
Hello,

I want to synthesize two simple modules, the (Verilog) file the top module instantiates the VHDL module twice. I am unable to synthesize it. Help me figure other this problem how I can solve this issue, 
Error Logs and files are also available.

Thanks!

Error :>>>>>>>>>>>>>>> Start>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
bash-4.2$ ./flow.tcl -interactive
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.11.23_01.42.34
[INFO]: Running interactively
% prep -design comparator2BitWithVHDL -tag Synthesis_1
[INFO]: Using design configuration at /openlane/designs/comparator2BitWithVHDL/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/comparator2BitWithVHDL/config.tcl
[INFO]: PDKs root directory: /home/shahid/OSPDKs
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/shahid/OSPDKs/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/comparator2BitWithVHDL/config.tcl
[INFO]: Current run directory is /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/shahid/OSPDKs/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
mergeLef.py : Merging LEFs
sky130_fd_sc_hd.lef: SITEs matched found: 0
sky130_fd_sc_hd.lef: MACROs matched found: 441
mergeLef.py : Merging LEFs complete
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparation complete
% run_synthesis
[INFO]: Running Synthesis...
[INFO]: current step index: 1

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+4052 (git sha1 UNKNOWN, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.

1. Executing Verilog-2005 frontend: /openlane/designs/comparator2BitWithVHDL/src/comparator1BitVHDL.vhd
/openlane/designs/comparator2BitWithVHDL/src/comparator1BitVHDL.vhd:1: ERROR: syntax error, unexpected TOK_DECREMENT
[ERROR]: during executing: "yosys -c /openlane/scripts/synth.tcl -l /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1/logs/synthesis/1-yosys.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

_[ERROR]: Please check yosys  log file
[ERROR]: Dumping to /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for comparator2BitWithVHDL/2022.11.12_20.14.29 in 0h0m19s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: comparator2BitWithVHDL
Run Directory: /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1/results/lvs/comparator2BitWithVHDL.lvs_parsed.gds.log
Source not found.
----------------------------------------_

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/comparator2BitWithVHDL/runs/Synthesis_1/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.

% 
:>>>>>>>>>>>>>>>>>>>End here>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

**Design RTL 
CODE**
.....................................................................................................................................................................................  

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

-- Uncomment the following library declaration if using
-- arithmetic functions with Signed or Unsigned values
--use IEEE.NUMERIC_STD.ALL;

-- Uncomment the following library declaration if instantiating
-- any Xilinx leaf cells in this code.
--library UNISIM;
--use UNISIM.VComponents.all;

entity comparator1BitVHDL  is

 port(
   x, y : in std_logic;
   eq : out std_logic
 ); 
end comparator1BitVHDL;

architecture dataflow1Bit of comparator1BitVHDL is
 signal s0, s1: std_logic; 
begin
 s0 <= (not x) and (not y);
 s1 <= x and y;
 
 eq <= s0 or s1;
end dataflow1Bit;
  
--------------------------------------------------------------------------------------------  
**Module 2**  
>>>>>>>>>>>>>>>> MAIN module>>>>>>>>>>>>>>>>
`timescale 1ns / 1ps

module comparator2BitWithVHDL(
    input wire[1:0] a, b,
    output wire eq
);

wire s0, s1;

// instantiate 1 bit comparator
comparator1BitVHDL eq_bit0 (.x(a[0]), .y(b[0]), .eq(s0));
comparator1BitVHDL eq_bit1 (.x(a[1]), .y(b[1]), .eq(s1));

assign eq = s0 & s1;
endmodule

  
.......................................................................................................................................................................................

**# User config**
set ::env(DESIGN_NAME) comparator2BitWithVHDL

**# Change if needed**
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.vhd]

**# Fill this**
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_PORT) "clk"

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}

maliberty, 2022-11-12:
afaik yosys doesn't support vhdl and you would have to use something like ghdl to convert it.  @antonblanchard might have some experience with this.

donn, 2022-11-13:
Please fill out the issue form properly. Thanks.

closed 2022-11-13 by donn