The-OpenROAD-Project/OpenLane #1489: Tcl_SetObjLength: negative length requested: -2147483436

Baungarten-CINVESTAV, 2022-11-13:
### Description

I am working with a small FPGA that was generated with the OpenFPGA tool, when I run OpenLane in step 19 the following message appears: 

_Tcl_SetObjLength: negative length requested: -2147483436 (integer overflow?)
Aborted (core dumped)_ 

How can I fix it?

Additional information: When I ran the synthesis exploration I got the same error

### Expected Behavior

[image]

### Environment report

```shell
Kernel: Linux v5.15.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 41ed0036c3dc77961a34005fc7e4494881f0a07f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (grafted, HEAD -> master, tag: 2022.10.26, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Material:
[FPGA_k4_n4_paper.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9997619/FPGA_k4_n4_paper.zip)

Command: 
`flow.tcl -design FPGA_k4_n4_paper`

### Relevant log output

```shell
Tcl_SetObjLength: negative length requested: -2147483436 (integer overflow?)
Aborted (core dumped)
```

donn, 2022-11-13:
Could you try with the latest version of OpenLane? We've updated OpenROAD since and this looks like an OpenROAD bug.

Baungarten-CINVESTAV, 2022-11-13:
Well, I have tried with the latest version and now the error appears in the first step (Syntehsis)

**Environment report:**
```
Kernel: Linux v5.15.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 9949f5d5f55749dc4b98648a25d355836895dc37
pip: INSTALLED
python-venv: INSTALLED

PDK Version Verification Status: OK

Git Log (Last 3 Commits)

9949f5d 2022-11-11T20:19:38+02:00 [BOT] Update openroad_app (#1478) - Openlane Bot -  (HEAD -> master, tag: 2022.11.12, origin/master, origin/HEAD)
67be872 2022-11-11T19:56:30+02:00 Include all PDK lib files when merging (#1484) - ferris -  ()
ef01337 2022-11-11T19:55:54+02:00 Update CI to include GF180MCU (#1480) - Mohamed Gaber -  ()

Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

**Expected Behavior**

[image]

Baungarten-CINVESTAV, 2022-11-14:
Regarding my last comment, I first ran the -synth_explore command and deleted the `set ::env(SYNTH_NO_FLAT) {1}`, then I ran the command again but I forgot to put the `SYNTH_NO_FLAT`.

So when I set the `SYNTH_NO_FLAT `again and ran openLANE the same problems occurred but now at step 20, As additional information the synthesis exploration failed with the same error.

[image]

vijayank88, 2022-11-14:
@Baungarten-CINVESTAV 
What is the total run time before `step 20` failure?

Baungarten-CINVESTAV, 2022-11-14:
Total run time is between 3 and 4 hours.
I'm working with the processor Intel Core i9-13900k and 128 GB RAM.

maliberty, 2022-11-14:
I was able to reproduce this overnight and will take a look

vijayank88, 2022-11-14:
@maliberty 
I tried with user source files and it got hangs at placement stage.
If I use `SYNTH_NO_FLAT=0` it's failed during synthesis stage itself with `child killed signal`

maliberty, 2022-11-14:
Those are unrelated problems.   Can you reproduce this problem outside of OL in a standalone OR run?

maliberty, 2022-11-14:
When I open the core file in gdb I see
```
Core was generated by `tclsh ./flow.tcl -design FPGA_k4_n4_paper -tag test'.
Program terminated with signal 6, Aborted.
```
so this appears to be a tclsh crash not an OR one.

maliberty, 2022-11-16:
@donn given that this is a crash from tclsh and not OR I think the OL team needs to look at it.

donn, 2022-11-16:
I mean the one thing I can think of that might be causing this is OpenROAD generating a log > 2GB

Baungarten-CINVESTAV, 2022-11-18:
That probably is what donn says, I will close this issue. 

If you have this problem try whit hardening the main modules of the projects separately (as a macro), then hardening the top module, remember to use the commands VERILOG_FILES_BLACKBOX, EXTRA_LEFS , and EXTRA_GDS_FILES.

I think that the variable becomes bigger because the project requires a lot of logic cells and that causes the overflow on the log files.

donn, 2022-11-18:
@Baungarten-CINVESTAV Please do not close the issue! We do want to fix this and keep your workaround documented in the meantime.

Baungarten-CINVESTAV, 2022-11-18:
Great, sorry for closing it.

maliberty, 2022-11-19:
The large log file is because of the well known problems in timing unconfigured fpga.  There are tremendous looping paths that exist structurally.  If you look at the log you'll see many of them.  Timing verification of an fpga is a hard problem and you can't expect generic solutions to work.

closed 2024-09-12 by donn