The-OpenROAD-Project/OpenLane #1491: Global timing report failed to report all paths

vijayank88, 2022-11-15:
### Description

With OpenLane tag [2022.11.12](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.11.12) spm design reports all min/max timing path.
With OpenLane tag [2022.11.15](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.11.15) missing timing path reports.
So its causing signoff timing report as well empty.

### Expected Behavior

Report all timing paths reports during global routing stage to check wns/tns.

### Environment report

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 1298859773c026fc1e852b2727ccd7c26d96e7bb
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1298859 2022-11-14T23:53:27+02:00 `io_place` virtual pins + test design (#1461) - Zeyad Zaki -  (HEAD -> master, tag: 2022.11.15, origin/master, origin/HEAD)
b99fc38 2022-11-14T23:48:29+02:00 Documentation Overhaul - Arman Avetisyan -  ()
0291a96 2022-11-14T22:26:18+02:00 `LIB_SYNTH_COMPLETE` -> `LIB_SYNTH` in some OpenROAD scripts (#1485) - ferris -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Update OpenLane commit to latest and run `make test`

2022.11.12 14-global.log: 
[14-global.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10011077/14-global.log)
2022.11.15 14-global.log: 
[14-global.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10011080/14-global.log)

### Relevant log output

```shell
===========================================================================
report_checks -path_delay min (Hold)
============================================================================
No paths found.
min_report_end
max_report

===========================================================================
report_checks -path_delay max (Setup)
============================================================================
No paths found.
max_report_end
check_report

===========================================================================
report_checks -unconstrained
============================================================================
Startpoint: clk (clock source 'clk')
Endpoint: clkbuf_0_clk/A (internal pin)
Path Group: (none)
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.01    0.01    5.01 v clk (in)
     1    0.00                           clk (net)
                  0.01    0.00    5.01 v clkbuf_0_clk/A (sky130_fd_sc_hd__clkbuf_16)
                                  5.01   data arrival time
-----------------------------------------------------------------------------
(Path is unconstrained)
```

vijayank88, 2022-11-15:
Difference found during global routing stage is wrt `LIB_SYNTH`
With latest commit `set ::env(LIB_SYNTH) {./tmp/synthesis/trimmed.lib};`
With 2022.11.12 tag `set ::env(LIB_SYNTH_COMPLETE) {pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib};`

donn, 2022-11-15:
That'd be https://github.com/The-OpenROAD-Project/OpenLane/pull/1485. However, the trimmed liberty file should contain all the data in this file... ugh, what a mess.

I obviously have to revert 1485 now, but this issue has to be reported in OpenSTA.

donn, 2022-11-15:
Misclick, apologies

closed 2022-11-16 by vijayank88