The-OpenROAD-Project/OpenLane #1494: blackboxed verilog files don't support the SV that yosys supports

mattvenn, 2022-11-15:
### Description

I can build a verilog project with an SV construct like genvar or unsigned.

If I then try to build that project into a larger one with a macro, it fails because the blackbox verilog doesn't support the same verilog used for hardening the sub project.

### Expected Behavior

openlane should be able to read the verilog as a blackbox file if it was able to build the GDS with it.

### Environment report

```shell
ernel: Linux v5.15.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

### Reproduction material

[bb_example.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10014572/bb_example.tar.gz)

tarball contains 2 projects:

52 - the example project that includes 'unsigned'. This was used to create the GDS and LEF.

52_macro - this tries to use 2 of the above project, but it fails with the error reported above.

### Relevant log output

```shell
ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/52_macro/runs/RUN_2022.11.15_16.37.51/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1                             
[ERROR]: Last 10 lines:                                                                                         
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.                        
                                                        
1. Executing Verilog-2005 frontend: /openlane/designs/52_macro/bb_src/counter.v
Parsing SystemVerilog input from `/openlane/designs/52_macro/bb_src/counter.v' to AST representation.
Generating RTLIL representation for module `\xor_shift32_quantamhd'.
Successfully finished Verilog frontend.                                                                                                                                                                                         
                                                                                                                
2. Executing Verilog-2005 frontend: /openlane/designs/52_macro/src/design.v
/openlane/designs/52_macro/src/design.v:22: ERROR: syntax error, unexpected ';', expecting ',' or ')'           
child process exited abnormally
```

donn, 2022-11-15:
Macros should come with nothing but a flattened netlist as far as I'm concerned, but I'm open to comments.

See https://github.com/The-OpenROAD-Project/OpenLane/issues/1291 for more info.

mattvenn, 2022-11-15:
So should I use gate level verilog for the black box files?

donn, 2022-11-15:
I believe you should, yes. Whatever the final netlist OpenLane spits out is the one you want to use.

mattvenn, 2022-11-15:
ok thanks Donn

donn, 2023-04-13:
Sorry for the late reply- I'll fwd this to the Caravel team.

closed 2023-04-13 by donn