The-OpenROAD-Project/OpenLane #1495: yosys can map to cells that don't exist

mattvenn, 2022-11-15:
### Description

if a design tries to make a flop with a non constant reset value, it can get mapped to a cell called _ALDFF_PP_ which doesn't exist in the PDK. The flow fails at floorplanning.

``` 
always @(posedge clk or posedge rst)
    begin
      if (rst) begin
        cnt <= init;
        tick <= 1;
      end else
```

### Expected Behavior

Yosys shouldn't try to map to cells that don't exist. And the flow should fail at synth not floorplanning.

### Environment report

```shell
Kernel: Linux v5.15.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

### Reproduction material

[example.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10014644/example.tar.gz)

No reproducible was made as the flow failed too hard.

### Relevant log output

```shell
[STEP 3]                                                                                                        
[INFO]: Running Initial Floorplanning...         
[ERROR]: Floorplanning failed                  
[ERROR]: module $_ALDFF_PP_ not found in /openlane/designs/binary-clock-asic/runs/RUN_2022.11.15_16.49.45/tmp/merged.nom.lef
[ERROR]: Check whether EXTRA_LEFS is set appropriately
[INFO]: Saving current set of views in 'designs/binary-clock-asic/runs/RUN_2022.11.15_16.49.45/results/final'... 
[INFO]: Generating final set of reports...           
[INFO]: Created manufacturability report at 'designs/binary-clock-asic/runs/RUN_2022.11.15_16.49.45/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/binary-clock-asic/runs/RUN_2022.11.15_16.49.45/reports/metrics.csv'.
[INFO]: Saving runtime environment...                                                                           
[ERROR]: Flow failed.                           
                                                        
    while executing                                 
"flow_fail"                                   
    (procedure "check_floorplan_missing_lef" line 11)
    invoked from within                             
"check_floorplan_missing_lef"                 
    (procedure "init_floorplan" line 12)          
    invoked from within                         
"init_floorplan"                                                                                                
    (procedure "run_floorplan" line 7)                                                                          
    invoked from within                       
"[lindex $step_exe 0] [lindex $step_exe 1] "       
    (procedure "run_non_interactive_mode" line 57)                                                              
    invoked from within                                                                                         
"run_non_interactive_mode {*}$argv"           
    invoked from within                                
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {                                                                    
        run_file [file nor..."                        
    (file "./flow.tcl" line 389)
```

donn, 2022-11-15:
2022-07-01 is not exactly in-date: but this is a known issue.

vijayank88, 2022-11-16:
@mattvenn 
We have discussed here #1070

kareefardi, 2023-08-03:
As for quitting right after synthesis, `[ERROR]: Synthesis failed. There are unmapped cells after synthesis.` this error is being generated now.

closed 2023-08-03 by kareefardi