The-OpenROAD-Project/OpenLane #1497: Running IO Placement stage fails

vijayank88, 2022-11-16:
### Description

This is MPW CI [design](https://github.com/The-OpenROAD-Project/caravel_user_project/blob/SultanShadow/APPROX_MULT/openlane/user_proj_example/pin_order.cfg), follows pin order name with xyz.*

As its MPW6 design, designer specified only few pins in specific side and left some of pins to place randomly.

With latest OpenLane setup its made compulsory to define manual placement of IO pins or go with auto pin placement technique. To skip IO placement error for un-defined pin placement used `set ::env(FP_IO_UNMATCHED_ERROR) 0`.

After the following PR https://github.com/The-OpenROAD-Project/OpenLane/pull/1461 facing error again. Help me how to resolve the error.

### Expected Behavior

Flow pass IO placement stage.

### Environment report

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 1da2eec3d9b5a3612cb026cf9060cb7954e4c24a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1da2eec 2022-11-16T01:37:27+02:00 [BOT] Update netgen (#1452) - Openlane Bot -  (HEAD -> master, tag: 2022.11.16, origin/master, origin/HEAD)
77fd531 2022-11-15T13:32:13+02:00 Revert #1485 (#1492) - Mohamed Gaber -  ()
dbb2487 2022-11-15T13:05:26+02:00 [BOT] Update PDK (#1455) - Openlane Bot -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Unzip `APPROX_MULT.zip` and place it `OpenLane/designs/`

%cd OpenLane
%make mount
%./flow.tcl -design APPROX_MULT -verbose 99

### Relevant log output

```shell
[INFO]: Running IO Placement (log: logs/floorplan/4-place_io.log)...
OpenROAD fc0bc8eb8a2ad14a84e0235578ff94733fc108ab 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Top-level design name: user_proj_example
Block boundaries: 0 0 900000 600000
Horizontal Tracks Origin: 340, Count: 882, Step: 680
Vertical Tracks Origin: 230, Count: 1956, Step: 460
Placement details for the #N side
Virtual pin count:  0
Actual pin count:  114
Total pin count:  114
Tracks count:  1956
Tracks per pin:  17
Used tracks count:  1922
Unused track count:  34
Starting track index:  17
Placement Map:
[230, 690, 1150, 1610, 2070, 2530, 2990, 3450, 3910, 4370, 4830, 5290, 5750, 6210, 6670, 7130, 7590, 8050, 8510, 8970, 9430, 9890, 10350, 10810, 11270, 11730, 12190, 12650, 13110, 13570, 14030, 14490, 14950, 15410, 15870, 16330, 16790, 17250, 17710, 18170, 18630, 19090, 19550, 20010, 20470, 20930, 21390, 21850, 22310, 22770, 23230, 23690, 24150, 24610, 25070, 25530, 25990, 26450, 26910, 27370, 27830, 28290, 28750, 29210, 29670, 30130, 30590, 31050, 31510, 31970, 32430, 32890, 33350, 33810, 34270, 34730, 35190, 35650, 36110, 36570, 37030, 37490, 37950, 38410, 38870, 39330, 39790, 40250, 40710, 41170, 41630, 42090, 42550, 43010, 43470, 43930, 44390, 44850, 45310, 45770, 46230, 46690, 47150, 47610, 48070, 48530, 48990, 49450, 49910, 50370, 50830, 51290, 51750, 52210, 52670, 53130, 53590, 54050, 54510, 54970, 55430, 55890, 56350, 56810, 57270, 57730, 58190, 58650, 59110, 59570, 60030, 60490, 60950, 61410, 61870, 62330, 62790, 63250, 63710, 64170, 64630, 65090, 65550, 66010, 66470, 66930, 67390, 67850, 68310, 68770, 69230, 69690, 70150, 70610, 71070, 71530, 71990, 72450, 72910, 73370, 73830, 74290, 74750, 75210, 75670, 76130, 76590, 77050, 77510, 77970, 78430, 78890, 79350, 79810, 80270, 80730, 81190, 81650, 82110, 82570, 83030, 83490, 83950, 84410, 84870, 85330, 85790, 86250, 86710, 87170, 87630, 88090, 88550, 89010, 89470, 89930, 90390, 90850, 91310, 91770, 92230, 92690, 93150, 93610, 94070, 94530, 94990, 95450, 95910, 96370, 96830, 97290, 97750, 98210, 98670, 99130, 99590, 100050, 100510, 100970, 101430, 101890, 102350, 102810, 103270, 103730, 104190, 104650, 105110, 105570, 106030, 106490, 106950, 107410, 107870, 108330, 108790, 109250, 109710, 110170, 110630, 111090, 111550, 112010, 112470, 112930, 113390, 113850, 114310, 114770, 115230, 115690, 116150, 116610, 117070, 117530, 117990, 118450, 118910, 119370, 119830, 120290, 120750, 121210, 121670, 122130, 122590, 123050, 123510, 123970, 124430, 124890, 125350, 125810, 126270, 126730, 127190, 127650, 128110, 128570, 129030, 129490, 129950, 130410, 130870, 131330, 131790, 132250, 132710, 133170, 133630, 134090, 134550, 135010, 135470, 135930, 136390, 136850, 137310, 137770, 138230, 138690, 139150, 139610, 140070, 140530, 140990, 141450, 141910, 142370, 142830, 143290, 143750, 144210, 144670, 145130, 145590, 146050, 146510, 146970, 147430, 147890, 148350, 148810, 149270, 149730, 150190, 150650, 151110, 151570, 152030, 152490, 152950, 153410, 153870, 154330, 154790, 155250, 155710, 156170, 156630, 157090, 157550, 158010, 158470, 158930, 159390, 159850, 160310, 160770, 161230, 161690, 162150, 162610, 163070, 163530, 163990, 164450, 164910, 165370, 165830, 166290, 166750, 167210, 167670, 168130, 168590, 169050, 169510, 169970, 170430, 170890, 171350, 171810, 172270, 172730, 173190, 173650, 174110, 174570, 175030, 175490, 175950, 176410, 176870, 177330, 177790, 178250, 178710, 179170, 179630, 180090, 180550, 181010, 181470, 181930, 182390, 182850, 183310, 183770, 184230, 184690, 185150, 185610, 186070, 186530, 186990, 187450, 187910, 188370, 188830, 189290, 189750, 190210, 190670, 191130, 191590, 192050, 192510, 192970, 193430, 193890, 194350, 194810, 195270, 195730, 196190, 196650, 197110, 197570, 198030, 198490, 198950, 199410, 199870, 200330, 200790, 201250, 201710, 202170, 202630, 203090, 203550, 204010, 204470, 204930, 205390, 205850, 206310, 206770, 207230, 207690, 208150, 208610, 209070, 209530, 209990, 210450, 210910, 211370, 211830, 212290, 212750, 213210, 213670, 214130, 214590, 215050, 215510, 215970, 216430, 216890, 217350, 217810, 218270, 218730, 219190, 219650, 220110, 220570, 221030, 221490, 221950, 222410, 222870, 223330, 223790, 224250, 224710, 225170, 225630, 226090, 226550, 227010, 227470, 227930, 228390, 228850, 229310, 229770, 230230, 230690, 231150, 231610, 232070, 232530, 232990, 233450, 233910, 234370, 234830, 235290, 235750, 236210, 236670, 237130, 237590, 238050, 238510, 238970, 239430, 239890, 240350, 240810, 241270, 241730, 242190, 242650, 243110, 243570, 244030, 244490, 244950, 245410, 245870, 246330, 246790, 247250, 247710, 248170, 248630, 249090, 249550, 250010, 250470, 250930, 251390, 251850, 252310, 252770, 253230, 253690, 254150, 254610, 255070, 255530, 255990, 256450, 256910, 257370, 257830, 258290, 258750, 259210, 259670, 260130, 260590, 261050, 261510, 261970, 262430, 262890, 263350, 263810, 264270, 264730, 265190, 265650, 266110, 266570, 267030, 267490, 267950, 268410, 268870, 269330, 269790, 270250, 270710, 271170, 271630, 272090, 272550, 273010, 273470, 273930, 274390, 274850, 275310, 275770, 276230, 276690, 277150, 277610, 278070, 278530, 278990, 279450, 279910, 280370, 280830, 281290, 281750, 282210, 282670, 283130, 283590, 284050, 284510, 284970, 285430, 285890, 286350, 286810, 287270, 287730, 288190, 288650, 289110, 289570, 290030, 290490, 290950, 291410, 291870, 292330, 292790, 293250, 293710, 294170, 294630, 295090, 295550, 296010, 296470, 296930, 297390, 297850, 298310, 298770, 299230, 299690, 300150, 300610, 301070, 301530, 301990, 302450, 302910, 303370, 303830, 304290, 304750, 305210, 305670, 306130, 306590, 307050, 307510, 307970, 308430, 308890, 309350, 309810, 310270, 310730, 311190, 311650, 312110, 312570, 313030, 313490, 313950, 314410, 314870, 315330, 315790, 316250, 316710, 317170, 317630, 318090, 318550, 319010, 319470, 319930, 320390, 320850, 321310, 321770, 322230, 322690, 323150, 323610, 324070, 324530, 324990, 325450, 325910, 326370, 326830, 327290, 327750, 328210, 328670, 329130, 329590, 330050, 330510, 330970, 331430, 331890, 332350, 332810, 333270, 333730, 334190, 334650, 335110, 335570, 336030, 336490, 336950, 337410, 337870, 338330, 338790, 339250, 339710, 340170, 340630, 341090, 341550, 342010, 342470, 342930, 343390, 343850, 344310, 344770, 345230, 345690, 346150, 346610, 347070, 347530, 347990, 348450, 348910, 349370, 349830, 350290, 350750, 351210, 351670, 352130, 352590, 353050, 353510, 353970, 354430, 354890, 355350, 355810, 356270, 356730, 357190, 357650, 358110, 358570, 359030, 359490, 359950, 360410, 360870, 361330, 361790, 362250, 362710, [... cut ...] t /openlane/designs/APPROX_MULT/runs/RUN_2022.11.16_09.35.48/logs/floorplan/4-place_io.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Calculating runtime...
[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2022-11-17:
@donn 
Waiting for your feedback how to proceed further.

donn, 2022-11-17:
@ZeyadZaki

vijayank88, 2022-11-18:
@ZeyadZaki 
Its blocking for many of previous MPW CI designs, can you please fix this issue?

closed 2022-11-22 by vijayank88