The-OpenROAD-Project/OpenLane #1499: Slew and cap violations for aes design

vijayank88, 2022-11-16:
### Description

With GF180 PDK facing slew/cap violation during signoff.

### Expected Behavior

Resolve slew/cap violations during signoff stage

### Environment report

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 1da2eec3d9b5a3612cb026cf9060cb7954e4c24a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1da2eec 2022-11-16T01:37:27+02:00 [BOT] Update netgen (#1452) - Openlane Bot -  (HEAD -> master, tag: 2022.11.16, origin/master, origin/HEAD)
77fd531 2022-11-15T13:32:13+02:00 Revert #1485 (#1492) - Mohamed Gaber -  ()
dbb2487 2022-11-15T13:05:26+02:00 [BOT] Update PDK (#1455) - Openlane Bot -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Due to file size limit `issue_reproducible.zip` attached here.

### Relevant log output

```shell
===========================================================================
max slew violation count 123
max fanout violation count 3667
max cap violation count 28
============================================================================
```

vijayank88, 2022-11-16:
@tspyrou fyi

jjcherry56, 2022-11-16:
You can see from the list of violations (not included above) where they are coming from.
They do not come from the sta at the end of the flow, which is the only def or db in the zip file.
They are not resizer issues.

vijayank88, 2022-11-17:
```
max slew

Pin                                    Limit    Slew   Slack
------------------------------------------------------------
ANTENNA__35303__B1/I                    8.60   13.88   -5.28 (VIOLATED)
_35303_/B1                              8.60   13.88   -5.28 (VIOLATED)
ANTENNA__36191__A2/I                    8.60   13.88   -5.28 (VIOLATED)
_36191_/A2                              8.60   13.88   -5.28 (VIOLATED)
_35302_/ZN                              8.60   13.88   -5.28 (VIOLATED)
ANTENNA__35167__B1/I                    8.60   13.03   -4.43 (VIOLATED)
ANTENNA__35457__A2/I                    8.60   13.03   -4.43 (VIOLATED)
_35167_/B1                              8.60   13.03   -4.43 (VIOLATED)
_35457_/A2                              8.60   13.03   -4.43 (VIOLATED)
_35166_/ZN                              8.60   13.03   -4.43 (VIOLATED)
ANTENNA__32581__B1/I                    8.60   11.63   -3.03 (VIOLATED)
_32581_/B1                              8.60   11.63   -3.03 (VIOLATED)
ANTENNA__36098__A2/I                    8.60   11.63   -3.03 (VIOLATED)
_32579_/ZN                              8.60   11.63   -3.03 (VIOLATED)
_36098_/A2                              8.60   11.63   -3.03 (VIOLATED)
ANTENNA__35277__B1/I                    8.60   11.46   -2.86 (VIOLATED)
_35277_/B1                              8.60   11.46   -2.86 (VIOLATED)
ANTENNA__36164__A2/I                    8.60   11.46   -2.86 (VIOLATED)
_36164_/A2                              8.60   11.46   -2.86 (VIOLATED)
_35276_/Z                               8.60   11.46   -2.86 (VIOLATED)
```
Its all due to antenna cells.
@maliberty fyi

maliberty, 2022-11-21:
@jjcherry56 slew on the input of an antenna cell is of no interest.  Is there a way to suppress them?

maliberty, 2022-11-21:
I don't think the antennas are the cause of the problem, they are just noise in the report.  The real problem must happen earlier in the flow and not much can be inferred from this test case at the end of the flow.  

My first guess is that the estimated parasitics don't correlate to the extracted parasitics.  Would you check max slew at the end of repair_design and the end of global_routing and see if the problems are the same or different?

vijayank88, 2022-11-22:
@maliberty 
Till global routing there is no slew/cap violations.
```
===========================================================================
max slew violation count 0
max fanout violation count 66
max cap violation count 0
============================================================================
```

maliberty, 2022-11-22:
That confirms my thinking.  It is unlikely the detailed router suddenly created many violations.  I believe you had some earlier discussion with Tim E. about the OL RCX setup.  I expect this is related to that issue.

vijayank88, 2022-11-22:
yes. GH issue #1493

vijayank88, 2023-07-17:
With latest OpenLane commit slew and cap violation is 0. So closing this issue.

closed 2023-07-17 by vijayank88