The-OpenROAD-Project/OpenLane #1504: GF180 fails at PDN

mattvenn, 2022-11-18:
### Description

flow ends at step 6, tries to make a reproducible, then runs out of disk space.

[image]

### Expected Behavior

not to fail

### Environment report

```shell
Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---

I don't know why this is showing as it is. I followed the instructions to install the pdk and openlane from the main branch of caravel_user_project, then ran make setup which installed the gf180 pdk and openlane. 

So probably this has something to do with my local environment.
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

### Reproduction material

reproducible failed by running out of diskspace.
to reproduce, run make wrapped_vga_clock in this repo:

https://github.com/mattvenn/gf180-vga-clock

### Relevant log output

```shell
[STEP 6]                                                                                             
[INFO]: Generating PDN...                                                                            
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1              
[ERROR]: full log: ../home/matt/work/asic-workshop/shuttle8/caravel_user_project/openlane/wrapped_vga_clock/runs/22_11_18_15_27/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines: 
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.     
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.                                                                                                                          
[WARNING PSM-0019] Voltage on net vccd1 is not explicitly set.                   
[WARNING PSM-0022] Using voltage 0.000V for VDD network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[WARNING PSM-0030] VSRC location at (6.720um, 15.680um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (23.040um, 15.380um).
[INFO PSM-0031] Number of PDN nodes on net vccd1 = 2179.                          
[ERROR PSM-0035] Metal3 resistance not found in DB. Check the LEF or set it using the 'set_layer_rc' command.
Error: pdn.tcl, 57 PSM-0035                                                                          
child process exited abnormally
```

donn, 2022-11-18:
Nothing I can do about this. Caravel needs to update the version of OL used.

QuantamHD, 2022-11-23:
@jeffdi Please take a look.

mattvenn, 2022-11-23:
No, I should have closed this. It was a misconfiguration of openlane versions at my end.