The-OpenROAD-Project/OpenLane #1505: Please help me resolve the following issue in placement

sumanthnimmakayala, 2022-11-18:
My config.tcl file - 
```
# User config
set ::env(DESIGN_NAME) radix2_pro

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this

#clk port #CTS error[INFO]: Running Clock Tree Synthesis...
#[ERROR CTS-0056] Error when finding -clk_nets in DB.
set ::env(DESIGN_IS_CORE) "0"
set ::env(CLOCK_PORT) "clk"
set ::env(CLOCK_NET) " "

set ::env(CLOCK_PERIOD) "90.0"

#test
set ::env(FP_CORE_UTIL) "50"
set ::env(PL_TARGET_DENSITY) "0.225"
set ::env(PL_BASIC_PLACEMENT) "0"
set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) "0"
set ::env(FP_SIZING) "absolute"
set ::env(DIE_AREA) "0.0 0.0 1000.000 1000.000"
#set ::env(CORE_AREA) "15.00 25.00 900.0 900.00"
set ::env(CORE_AREA) "11.00 16.00 900.0 900.00"
set ::env(DIODE_INSERTION_STRATEGY) "4"
set ::env(GLB_RT_MAXLAYER) "5"

set ::env(FP_PDN_AUTO_ADJUST) "0"

#set ::env(VDD_NETS) [list {vccd1}]
#set ::env(GND_NETS) [list {vssd1}]

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
source $filename
}
```

# Utilization exceeds 100%

Error : or_replace.tcl

[image]

My system -
Debian OS, 16GB RAM, 1TB SSD

maliberty, 2022-11-18:
Your design is too big to fit in the area given.  You either need more area or fewer gates.

sumanthnimmakayala, 2022-11-19:
Even though after increasing ``CORE_AREA`` and ```DIE_AREA``` 
I am getting some other issue.

[image]

Config File:

```
# User config
set ::env(DESIGN_NAME) radix2_pro

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this

#clk port #CTS error[INFO]: Running Clock Tree Synthesis...
#[ERROR CTS-0056] Error when finding -clk_nets in DB.
set ::env(DESIGN_IS_CORE) "0"
set ::env(CLOCK_PORT) "clk"

set ::env(CLOCK_PERIOD) "90.0"

#test
set ::env(FP_CORE_UTIL) "30"
set ::env(GLB_RT_ALLOW_CONGESTION) "1"
set ::env(PL_BASIC_PLACEMENT) "0"
set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) "0"
set ::env(FP_SIZING) "absolute"
set ::env(DIE_AREA) "0.0 0.0 6937.49 6948.21"
#set ::env(CORE_AREA) "15.00 25.00 900.0 900.00"
set ::env(CORE_AREA) "5.52 10.88 6931.70 6935.04"
set ::env(DIODE_INSERTION_STRATEGY) "4"
set ::env(GLB_RT_MAXLAYER) "4"

set ::env(FP_PDN_AUTO_ADJUST) "0"

#set ::env(VDD_NETS) [list {vccd1}]
#set ::env(GND_NETS) [list {vssd1}]

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
source $filename
}
```

Is there any new contain to add.

maliberty, 2022-11-19:
Please attached a test case reproducing the problem

sumanthnimmakayala, 2022-11-19:
@maliberty 
I don't have much Idea which test case is producing particular problem.

maliberty, 2022-11-19:
You seem to be confusing line in your config file with a test case.  I need something I can run that produces the same output as you are showing above.

vijayank88, 2022-11-19:
@sumanthnimmakayala 
Read and fill the issue template properly with all details and open the issue again.
Its important to know which OpenLane version currently you're working.

closed 2022-11-19 by vijayank88