The-OpenROAD-Project/OpenLane #1506: JSON config should support non-string vars

proppy, 2022-11-18:
### Description

Currently if encountering a non-string var in an expression:
```
    "UNIT": 2.4,
    "FP_IO_VEXTEND": "expr::2 * $UNIT",
```
variable substitutions will fails with the following error:
```
TypeError: replace() argument 2 must be str, not float
```

### Expected Behavior

non-string variables get substituted in expressions.

### Environment report

```shell
Kernel: Linux v5.10.142-19743-gd301e86c39f7
Distribution: debian 11
Python: v3.7.13 (OK)
Container Engine: docker v20.10.18 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD -> master, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

```
curl -L -O https://github.com/efabless/caravel_user_project/blob/main/openlane/user_project_wrapper/config.json
flow.tcl .
```

### Relevant log output

```shell
Error: : Traceback (most recent call last):
  File "/openlane/scripts/config/tcl.py", line 473, in <module>
    cli()
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/config/tcl.py", line 459, in from_json
    resolved = process_config_dict(config_dict, exposed_dict)
  File "/openlane/scripts/config/tcl.py", line 320, in process_config_dict
    process_config_dict_recursive(config_in, state)
  File "/openlane/scripts/config/tcl.py", line [31](https://github.com/proppy/tiny_user_project/actions/runs/3499668129/jobs/5861479549#step:10:32)2, in process_config_dict_recursive
    value = process_scalar(key, value, state)
  File "/openlane/scripts/config/tcl.py", line 261, in process_scalar
    value = process_string(value, state)
  File "/openlane/scripts/config/tcl.py", line 2[44](https://github.com/proppy/tiny_user_project/actions/runs/3499668129/jobs/5861479549#step:10:45), in process_string
    value = reference.replace(match[0], found)
TypeError: replace() argument 2 must be str, not float
```

proppy, 2022-11-18:
a workaround is to convert the config value to a string, ex:
```
    "UNIT": 2.4,
    "FP_IO_VEXTEND": "expr::2 * $UNIT",
```

closed 2022-11-20 by jeffdi

proppy, 2022-11-20:
@jeffdi I think we should leave this open, the proper fix is to have OpenLane handle non string vars.

donn, 2022-11-22:
Wait, huh? That's not... how expr is supposed to work

Let me test

donn, 2022-11-22:
Seems to be fixed as of #1509, can you check please?