The-OpenROAD-Project/OpenLane #1508: Please help to slove this issue

sumanthnimmakayala, 2022-11-19:
I am getting this error

```
[INFO DRT-0195] Start 64th optimization iteration.
    Completing 10% with 46 violations.
    elapsed time = 00:00:00, memory = 3713.43 (MB).
    Completing 20% with 46 violations.
    elapsed time = 00:00:00, memory = 3713.43 (MB).
    Completing 30% with 46 violations.
    elapsed time = 00:00:00, memory = 3713.43 (MB).
    Completing 40% with 46 violations.
    elapsed time = 00:00:00, memory = 3713.43 (MB).
    Completing 50% with 46 violations.
    elapsed time = 00:00:00, memory = 3713.43 (MB).
    Completing 60% with 44 violations.
    elapsed time = 00:00:01, memory = 3713.43 (MB).
    Completing 70% with 44 violations.
    elapsed time = 00:00:01, memory = 3713.43 (MB).
    Completing 80% with 44 violations.
    elapsed time = 00:05:11, memory = 3713.43 (MB).
    Completing 90% with 44 violations.
    elapsed time = 00:05:11, memory = 3713.43 (MB).
    Completing 100% with 44 violations.
    elapsed time = 00:05:42, memory = 3713.43 (MB).
[INFO DRT-0199]   Number of violations = 44.
[INFO DRT-0267] cpu time = 00:05:42, elapsed time = 00:05:42, memory = 3713.43 (MB), peak = 3834.52 (MB)
Total wire length = 4865255 um.
Total wire length on LAYER li1 = 11643 um.
Total wire length on LAYER met1 = 1992831 um.
Total wire length on LAYER met2 = 2235038 um.
Total wire length on LAYER met3 = 524288 um.
Total wire length on LAYER met4 = 87389 um.
Total wire length on LAYER met5 = 14064 um.
Total number of vias = 1030743.
Up-via summary (total 1030743):.

--------------------------
 FR_MASTERSLICE          0
            li1     502701
           met1     490342
           met2      31526
           met3       5975
           met4        199
--------------------------
                   1030743

[INFO DRT-0198] Complete detail routing.
Total wire length = 4865255 um.
Total wire length on LAYER li1 = 11643 um.
Total wire length on LAYER met1 = 1992831 um.
Total wire length on LAYER met2 = 2235038 um.
Total wire length on LAYER met3 = 524288 um.
Total wire length on LAYER met4 = 87389 um.
Total wire length on LAYER met5 = 14064 um.
Total number of vias = 1030743.
Up-via summary (total 1030743):.

--------------------------
 FR_MASTERSLICE          0
            li1     502701
           met1     490342
           met2      31526
           met3       5975
           met4        199
--------------------------
                   1030743

[INFO DRT-0267] cpu time = 07:14:00, elapsed time = 04:48:39, memory = 3713.43 (MB), peak = 3834.52 (MB)

[INFO DRT-0180] Post processing.
Saving to /openLANE_flow/designs/radix4_mdf/runs/18-11_17-56/results/routing/23-radix4_mdf.def
[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total Number of violations is 44
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for radix4_mdf/18-11_17-56 in 5h7m8s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: radix4_mdf
Run Directory: /openLANE_flow/designs/radix4_mdf/runs/18-11_17-56
Source not found.
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/radix4_mdf/runs/18-11_17-56/results/lvs/radix4_mdf.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openLANE_flow/designs/radix4_mdf/runs/18-11_17-56/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.
```

This is my config file
```
# User config
set ::env(DESIGN_NAME) radix4_mdf

# Change if needed
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

# Fill this
#set ::env(CLOCK_NET) " " 

set ::env(DESIGN_IS_CORE) "0"

set ::env(CLOCK_PERIOD) "30.0"
set ::env(CLOCK_PORT) "clk"

set ::env(DIODE_INSERTION_STRATEGY) "4"
set ::env(FP_CORE_UTIL) "30"
set ::env(CELL_PAD) "2"
set ::env(PL_BASIC_PLACEMENT) "0"
set ::env(DIODE_PADDING) "0"
set ::env(DIODE_INSERTION_STRATEGY) "4"

set filename $::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
```

maliberty, 2022-11-19:
Without a test case not much can be said.

sumanthnimmakayala, 2022-11-19:
@maliberty  
Which test case I was missing.?

maliberty, 2022-11-19:
How can this problem be reproduced?  The log file is insufficient.

sumanthnimmakayala, 2022-11-19:
@maliberty 
Could you please tell me which test cases in need to add sir.?
Are these ..?
```set ::env(FP_SIZING) "absolute"
set ::env(DIE_AREA) "0.0 0.0 6937.49 6948.21"
set ::env(CORE_AREA) "5.52 10.88 6931.70 6935.04"
```
Or some other I need to add..?

maliberty, 2022-11-19:
You can try increasing the die & core area yourself or follow https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/for_developers/using_or_issue.md to generate a test case

vijayank88, 2022-11-19:
@sumanthnimmakayala 
Read and fill the issue template properly with all details and open the issue again.
Its important to know which OpenLane version currently you're working.

closed 2022-11-19 by vijayank88