The-OpenROAD-Project/OpenLane #1512: [PDN] Unable to connect Power Hook Macro with met2/met3 power mesh with top-level power ring met4/met5

dineshannayya, 2022-11-24:
### Description

I have SOC design with multiple Macro with internal macro power met3/met4. Recently i have added a new Macro (u_rp_south) with power mesh with met2/met3.  Look like Top-level PDN run does not recognized this macro and does not hook up Power  and also no warning/error on connectivity issue. 
[image]

Adding additional below command in pdn script also does not help
add_pdn_connect -grid macro -layers "met3 met5"
Looking for option to connect MACRO with met2/met3 power mesh with top-level power ring met4/met5

### Expected Behavior

Looking for option to connect MACRO with met2/met3 power mesh with top-level power ring met4/met5

### Environment report

```shell
<pre>Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (HEAD -&gt; master)
78ed217 2022-11-17T20:29:24+02:00 [BOT] Update openroad_app (#1487) - Openlane Bot -  ()
ceb5288 2022-11-17T20:06:36+02:00 [BOT] Update netgen (#1500) - Openlane Bot -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10081147/issue_reproducible.tar.gz)

### Relevant log output

```shell
<pre>[INFO]: Running Single-Corner Static Timing Analysis (log: user_project_wrapper/runs/user_project_wrapper/logs/synthesis/2-sta.log)...
[INFO]: Creating a netlist with power/ground pins.
[STEP 3]
[INFO]: Running Initial Floorplanning (log: user_project_wrapper/runs/user_project_wrapper/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 2908.58 and height 3497.92.
[STEP 4]
[INFO]: Running IO Placement...
[INFO]: Applying DEF template...
[STEP 5]
[INFO]: Performing Manual Macro Placement (log: user_project_wrapper/runs/user_project_wrapper/logs/placement/5-macro_placement.log)...
[INFO]: Power planning with power {vccd1 vccd2 vdda1 vdda2} and ground {vssd1 vssd2 vssa1 vssa2}...
[STEP 6]
[INFO]: Generating PDN (log: user_project_wrapper/runs/user_project_wrapper/logs/floorplan/6-pdn.log)...
</pre>
```

maliberty, 2022-11-24:
@arlpetergadfort thoughts?

arlpetergadfort, 2022-11-24:
@dineshannayya Adding `add_pdn_connect -grid macro -layers "met3 met5"` will look for ways to connect met3 to met5, but you have met4 straps that would nearly always get in the way. 
What you probably want is `add_pdn_connect -grid macro -layers "met3 met4"` which will connect the pins on met3 to the met4 straps. (looks like it works when I tried)

dineshannayya, 2022-11-25:
@arlpetergadfort When I add the below command 
add_pdn_connect -grid macro -layers "met3 met4"
PDN does not connect the power MET3 to MET4 power strips, but fails with connectivity check  and does not give any additional info
[ERROR PSM-0069] Check connectivity failed.
[ERROR GUI-0070] Error: pdn.tcl, 133 PSM-00

arlpetergadfort, 2022-11-25:
@dineshannayya that appears to be an issue with your macros. There are a TON of warnings that tell you the x/y coordinates that are not connected the error message is just letting you know there is a problem.

The image below is for one of the disconnections. For some reason this pin on this macro is broken into segments instead of one pin on met4, this is causing parts to appear to be disconnected when they are not really. In the picture there is a met5 strap right over it so there is no way to connect to that point, but that pin is connected at the next met5 strap 40um above.
[image]

dineshannayya, 2022-11-25:
@arlpetergadfort I feel your are looking to wrong macro - u_dache_2b is SRAM module and this module is part of Sky130A PDK only. Magic Tool lot's of violation inside SRAM which are not valid as per efabless team.
I am facing issue in u_rp_south macro.  This block has only buffer to fix the long route.

arlpetergadfort, 2022-11-25:
@dineshannayya As far as I can tell `u_rp_south` is getting connected (see picture, you can see the via3 vias).
[image]
The PSM error does not appear to be related to PDNGEN, but the macro pins instead, atleast my spot checking of them.

dineshannayya, 2022-11-25:
My local run does not shows Via3 connectivity .. Which OpenROAD version you are using .. Is there any new PDNGEN latest fixes ? 
My Openlane version is
<pre>Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (HEAD -&gt; master)
78ed217 2022-11-17T20:29:24+02:00 [BOT] Update openroad_app (#1487) - Openlane Bot -  ()
ceb5288 2022-11-17T20:06:36+02:00 [BOT] Update netgen (#1500) - Openlane Bot -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

arlpetergadfort, 2022-11-25:
There are no recent changes to PDNGEN, I'm using the HEAD of OpenROAD. I do not run Openlane so I don't know what version you are on.

dineshannayya, 2022-11-26:
I have installed latest openlane, but still no Via3 connectivity,  Can you share me your pdn_cfg file 

[image]

dineshannayya, 2022-11-26:
I am facing issue locally installing latest Openroad, It's look asking me to install ortools

<pre><font color="#4E9A06"><b>dinesha@lenovo-i3-10100-07IMB05</b></font>:<font color="#3465A4"><b>~/workarea/tools/source/OpenROAD_10012020/build</b></font>$ cmake ..
-- OpenROAD version: v2.0-5825-g7c85c1403
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- boost: 1.72.0
-- spdlog: 1.8.1
-- STA version: 2.3.1
-- STA git sha: 937efddf5ecade9b927829074ab3bb274290a949
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/dinesha/workarea/tools/source/OpenROAD_10012020/src/sta/app/sta
-- GPU is not enabled
CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  By not providing &quot;Findortools.cmake&quot; in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by &quot;ortools&quot;, but
  CMake did not find one.

  Could not find a package configuration file provided by &quot;ortools&quot; with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of &quot;ortools&quot; to CMAKE_PREFIX_PATH or set
  &quot;ortools_DIR&quot; to a directory containing one of the above files.  If
  &quot;ortools&quot; provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also &quot;/home/dinesha/workarea/tools/source/OpenROAD_10012020/build/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/dinesha/workarea/tools/source/OpenROAD_10012020/build/CMakeFiles/CMakeError.log&quot;.
</pre>

arlpetergadfort, 2022-11-26:
@dineshannayya I'm using the pdn_cfg.tcl from your second upload. I made no edits.
@maliberty can probably help you with the ortools issue.

maliberty, 2022-11-26:
or_tools is a new dependency so you should run the dependency installer again  (see https://github.com/The-OpenROAD-Project/OpenROAD/blob/7c85c140308f01b73f57ea1117f3e43f39abd437/etc/DependencyInstaller.sh#L118)

dineshannayya, 2022-11-26:
@arlpetergadfort  Now I am able to see the VIA-3 connectivity
[image]

But, with adding the new PDN command MET3 to MET4 connection [add_pdn_connect -grid macro -layers "met3 met4"]
Now Check Node/IRDROP Enable function [set ::env(FP_PDN_CHECK_NODES) {1} & set ::env(FP_PDN_IRDROP) {1}]  PDNGEN fails with 

[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 39 PSM-0069

This issue look to SRAM macro related one and PDNGEN tool try to connect MET3 POWER Strip inside SRAM and facing issue (As you pointed out -For some reason this pin on this macro is broken into segments instead of one pin on met4, this is causing parts to appear to be disconnected when they are not really. In the picture there is a met5 strap right over it so there is no way to connect to that point, but that pin is connected at the next met5 strap 40um above.)

Is it possible to constraint met-3 to met-4 hook command to specific marco instead of global definition ?

Other wise I need to disable the FP_PDN_CHECK_NODES & FP_PDN_IRDRO in my flow.

arlpetergadfort, 2022-11-26:
@dineshannayya yes you can constrain the connect statements to particular instances, just create a macro grid for that instance and add the connect statements to that. The method of just creating a generic macro grid is not great because different macros may have particular connection requirements. So creating the specific grid would be better.
The connectivity issue isnt something that pdngen controls, so that is a separate issue. I also think the SRAM pins should be addressed so they appear as a single pin. 
Unless there are actual connectivity issues I would suggest turning off the steps you suggested and revisiting if something comes up.

dineshannayya, 2022-11-27:
@arlpetergadfort I am able to setup flow with two separate macro grid:- one for met4 => met5 and other for met3 => met4 connection. With this change Openlane flow is clean with  FP_PDN_CHECK_NODES & FP_PDN_IRDROP check enabled. 
I see this approach is one of the cool and professional thought feature supported in OpenRoad/openlane. 
Appreciate your quick follow up and suggestion.  We can close this ticket

closed 2022-11-27 by maliberty