The-OpenROAD-Project/OpenLane #1514: `yosys-abc` never finishes

xobs, 2022-11-24:
### Description

I thought I'd try synthesizing something other than the inverter, so I took a VexRiscv verilog file and tried replacing `inverter.v`. The resulting repo is at https://github.com/xobs/openlane-cpu-synth-test

The build has been running for about seven hours now:

```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
11209 user      20   0 2010644   1.9g   9516 R  99.7   3.0 411:02.80 yosys-abc
```

This was reported in #openlane, and I was instructed to open an Issue report.

### Expected Behavior

`yosys-abc` shouldn't take so long, particularly since the synthesis only took a few minutes. It would be very useful to know what's going on.

### Environment report

```shell
Kernel: Linux v5.15.74.2-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: flow-overview-typo-dev
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

dbc0495 2022-11-24T09:47:36+08:00 docs: correct small typo in flow_overview.md - Sean Cross -  (HEAD -> flow-overview-typo, xobs/flow-overview-typo)
ff85171 2022-11-21T11:23:51+02:00 Document Error in JSON Config Files (#1509) - Mohamed Gaber -  (origin/master, origin/HEAD, master)
cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (tag: 2022.11.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
xobs    git@github.com:xobs/OpenLane.git (fetch)
xobs    git@github.com:xobs/OpenLane.git (push)
```

### Reproduction material

The repo was posted in "Description"

### Relevant log output

```shell
Running synthesis (run_synthesis).
[STEP 1]
[INFO]: Running Synthesis (log: ../../work/vexriscv-betrusted/runs/RUN_2022.11.24_10.12.11/logs/synthesis/1-synthesis.log)...
Thu Nov 24 10:12:12 +08 2022 - Executing "yosys -c /opt/Si/src/OpenLane/scripts/yosys/synth.tcl |& tee /dev/null /opt/Si/work/vexriscv-betrusted/runs/RUN_2022.11.24_10.12.11/logs/synthesis/1-synthesis.log"
```

vijayank88, 2022-11-24:
@xobs 
Update the `config.json` as per your design.
```
{
    "DESIGN_NAME": "VexRiscv",
    "VERILOG_FILES": "dir::VexRiscv_BetrustedSoC_Debug.v",
    "CLOCK_PORT": clk,
    "DIODE_INSERTION_STRATEGY": 3,
    "SYNTHESIS_STRATEGY": "DELAY 0"
}
```

Refer this setup [file](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/designs/picorv32a/config.json)

xobs, 2022-11-24:
The previous run finished in 7.5 hours, and then complained about `[ERROR PPL-0024] Number of pins 361 exceeds max possible 106.` A fair gripe. Probably not the best idea to expose a Harvard CPU directly to pins.

Let me create a `top.v` that reduces the number of pins that are exposed.

Note that I do have `config.json` configured as you suggested, and `yosys-abc` has been running for 40 minutes now.

I'm also running `flow.tcl -synth_explore` as documented in [hardening_macros.md](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/usage/hardening_macros.md#synthesis) and `yosys-abc` has been running for 21 minutes so far...

I'll try again after putting a debug connector on this and see if that doesn't improve synthesis, but it definitely seems like `yosys-abc` takes a very, very long time.

antonblanchard, 2022-11-24:
I'm not sure it's the problem but you have some pretty big arrays in that design:

```
  (* ram_style = "block" *) reg [31:0] memory_AesPlugin_rom_storage [0:511];
  (* ram_style = "block" *) reg [21:0] ways_0_tags [0:127];
  (* ram_style = "block" *) reg [7:0] ways_0_data_symbol0 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_0_data_symbol1 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_0_data_symbol2 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_0_data_symbol3 [0:1023];
  (* ram_style = "block" *) reg [21:0] ways_1_tags [0:127];
  (* ram_style = "block" *) reg [7:0] ways_1_data_symbol0 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_1_data_symbol1 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_1_data_symbol2 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_1_data_symbol3 [0:1023];
  (* ram_style = "block" *) reg [21:0] ways_2_tags [0:127];
  (* ram_style = "block" *) reg [7:0] ways_2_data_symbol0 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_2_data_symbol1 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_2_data_symbol2 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_2_data_symbol3 [0:1023];
  (* ram_style = "block" *) reg [21:0] ways_3_tags [0:127];
  (* ram_style = "block" *) reg [7:0] ways_3_data_symbol0 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_3_data_symbol1 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_3_data_symbol2 [0:1023];
  (* ram_style = "block" *) reg [7:0] ways_3_data_symbol3 [0:1023];
  (* ram_style = "block" *) reg [31:0] banks_0 [0:1023];
  (* ram_style = "block" *) reg [31:0] banks_1 [0:1023];
  (* ram_style = "block" *) reg [21:0] ways_0_tags [0:127];
  (* ram_style = "block" *) reg [21:0] ways_1_tags [0:127];
```

Reducing the size of these or using DFFRAM macros should help.

xobs, 2022-11-24:
That was one of the theories as to what could be causing the runtime. Is there any way to get Yosys to infer the correct macro? Or will I need to add a step before Yosys is run that replaces these registers with a module? How would I know which DFFRAM macros are available or which one I should use?

Is there any way to see what it's doing? The logfile just ends with `17.1.1. Executing ABC.` without any progress as to what ABC is doing.

Is it just a bad idea to directly adapt this design from an FPGA, and instead I should pick a smaller cache with fewer ways?

xobs, 2022-11-25:
@antonblanchard That's a good find. It seems as though OpenLane isn't currently inferring RAMs due to a missing `memory_libmap` pass. I've opened https://github.com/VLSIDA/OpenRAM/issues/163 to get the libmap and techmap files generated, and it may make sense to add a `map_ram` pass to https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/yosys/synth.tcl

antonblanchard, 2022-11-30:
> I've opened [VLSIDA/OpenRAM#163](https://github.com/VLSIDA/OpenRAM/issues/163) to get the libmap and techmap files generated, and it may make sense to add a `map_ram` pass to https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/yosys/synth.tcl

Great! We could also build and publish a set of https://github.com/AUCOHL/DFFRAM RAMs and add yosys inference for them.

I did take a quick look at your verilog, and it is large, but I have no idea why we got stuck in yosys for so long. There are just under a million standard cells:

```
=== VexRiscv ===

   Number of wires:             916484
   Number of wire bits:         916809
   Number of public wires:      217117
   Number of public wire bits:  217442
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:             916630
     sky130_fd_sc_hd__a2111o_1       2
     sky130_fd_sc_hd__a2111oi_0     26
     sky130_fd_sc_hd__a211o_1       16
     sky130_fd_sc_hd__a211oi_1     180
     sky130_fd_sc_hd__a21bo_1        3
     sky130_fd_sc_hd__a21boi_0      27
     sky130_fd_sc_hd__a21o_1        92
     sky130_fd_sc_hd__a21oi_1    44843
     sky130_fd_sc_hd__a221o_1       13
     sky130_fd_sc_hd__a221oi_1   22038
     sky130_fd_sc_hd__a222oi_1     118
     sky130_fd_sc_hd__a22o_1       136
     sky130_fd_sc_hd__a22oi_1      250
     sky130_fd_sc_hd__a2bb2o_1       1
     sky130_fd_sc_hd__a2bb2oi_1      6
     sky130_fd_sc_hd__a311o_1        1
     sky130_fd_sc_hd__a311oi_1      81
     sky130_fd_sc_hd__a31o_1        21
     sky130_fd_sc_hd__a31oi_1     9511
     sky130_fd_sc_hd__a32o_1       137
     sky130_fd_sc_hd__a32oi_1      217
     sky130_fd_sc_hd__a41o_1         4
     sky130_fd_sc_hd__a41oi_1       12
     sky130_fd_sc_hd__and2_0       222
     sky130_fd_sc_hd__and3_1       146
     sky130_fd_sc_hd__and3b_1       10
     sky130_fd_sc_hd__and4_1        19
     sky130_fd_sc_hd__and4b_1        3
     sky130_fd_sc_hd__buf_4        128
     sky130_fd_sc_hd__conb_1         2
     sky130_fd_sc_hd__dfxtp_1   216931
     sky130_fd_sc_hd__inv_1        372
     sky130_fd_sc_hd__maj3_1       729
     sky130_fd_sc_hd__mux2_1     81682
     sky130_fd_sc_hd__mux2i_1      372
     sky130_fd_sc_hd__mux4_2       512
     sky130_fd_sc_hd__nand2_1   197101
     sky130_fd_sc_hd__nand2b_1     249
     sky130_fd_sc_hd__nand3_1    10903
     sky130_fd_sc_hd__nand3b_1      25
     sky130_fd_sc_hd__nand4_1      166
     sky130_fd_sc_hd__nand4b_1       9
     sky130_fd_sc_hd__nand4bb_1      2
     sky130_fd_sc_hd__nor2_1     71297
     sky130_fd_sc_hd__nor2b_1      202
     sky130_fd_sc_hd__nor3_1      6136
     sky130_fd_sc_hd__nor3b_1       33
     sky130_fd_sc_hd__nor4_1       108
     sky130_fd_sc_hd__nor4b_1        6
     sky130_fd_sc_hd__nor4bb_1       8
     sky130_fd_sc_hd__o2111ai_1     25
     sky130_fd_sc_hd__o211a_1        8
     sky130_fd_sc_hd__o211ai_1      64
     sky130_fd_sc_hd__o21a_1        83
     sky130_fd_sc_hd__o21ai_0   212335
     sky130_fd_sc_hd__o21ba_1        4
     sky130_fd_sc_hd__o21bai_1      10
     sky130_fd_sc_hd__o221a_1       19
     sky130_fd_sc_hd__o221ai_1   33248
     sky130_fd_sc_hd__o22a_1       164
     sky130_fd_sc_hd__o22ai_1      759
     sky130_fd_sc_hd__o2bb2ai_1     26
     sky130_fd_sc_hd__o311ai_0     394
     sky130_fd_sc_hd__o31a_1         5
     sky130_fd_sc_hd__o31ai_1     1171
     sky130_fd_sc_hd__o32a_1         3
     sky130_fd_sc_hd__o32ai_1       79
     sky130_fd_sc_hd__o41ai_1       12
     sky130_fd_sc_hd__or2_0        106
     sky130_fd_sc_hd__or3_1         50
     sky130_fd_sc_hd__or3b_1        13
     sky130_fd_sc_hd__or4_1         14
     sky130_fd_sc_hd__or4b_1         3
     sky130_fd_sc_hd__xnor2_1     1531
     sky130_fd_sc_hd__xnor3_1      236
     sky130_fd_sc_hd__xor2_1       967
     sky130_fd_sc_hd__xor3_1       193

   Chip area for module '\VexRiscv': 8284565.555213
```

It looks to be getting stuck in ABC. Switching to the ORFS synth strategy makes it go much faster for me, it might be worth a try:

```
set ::env(SYNTH_STRATEGY) {DELAY 4}
```

donn, 2022-11-30:
I guess what's actionable here is RAM mapping..? I really recommend you just use a RAM macro though instead of just letting Yosys try to infer one...

xobs, 2022-12-01:
I'm at the stage where I'm just getting an understanding of the project, and I'm making lots of obvious mistakes. I'm discovering that one of those mistakes is that relying on implicit memories like I do when designing an FPGA is not a good idea because it really blows up synthesis time.

There's a lot of documentation, so perhaps I missed something saying that's a bad thing to do.

I do agree that the lowest-hanging of fruit would be to create a set of DFFRAMs and have Yosys infer them. That would improve routing and layout for caches.

Another thing that I'm having trouble understanding is how to diagnose what the problem is. There doesn't appear to be any sort of log. All I have to go on is seeing `yosys-abc` at the top of my process table and the log entry "19.1.1. Executing ABC.". I'm not sure how often this is an issue, so I'm not sure how much effort should be put into diagnosing this, but how can I better understand where the bottlenecks are and how to address them? I.e. how could I have identified the caches as being the source of the long synthesis times? And is there any way to add a check for that to alert people about it before they start synthesizing?

vijayank88, 2022-12-01:
Try `./flow.tcl -design VexRiscv -synth_explore` will give PPA based on synthesis exploration. Choose the best PPA as per design requirement.
Next important thing is use of OpenRAM macros

xobs, 2022-12-01:
I did try `-synth_explore`, but I want to say it took about eight hours to run. I'll try it again and get an exact number for you, but it'll be a few hours.

Setting `SYNTH_STRATEGY` to `DELAY 4` did allow Yosys to finish in about 25 minutes, which is much improved! I still should find a way to extract the caches to a RAM macro. Is there any documentation on how to select a `SYNTH_STRATEGY`? What does it mean to prioritize area or delay? And what does the index actually mean?

vijayank88, 2022-12-01:
Refer this: https://openlane.readthedocs.io/en/latest/reference/configuration.html#synthesis

xobs, 2022-12-01:
Does a `SYNTH_STRATEGY` of `DELAY` mean that it will try to synthesize logic to have the lowest delay, whereas `AREA` will try to use the fewest number of logic elements? What does the index actually mean? I don't see it defined anywhere.

vijayank88, 2022-12-01:
I asked same question #959

antonblanchard, 2022-12-01:
> Does a `SYNTH_STRATEGY` of `DELAY` mean that it will try to synthesize logic to have the lowest delay, whereas `AREA` will try to use the fewest number of logic elements? What does the index actually mean? I don't see it defined anywhere.

While that may have been the original intent, it's basically just different ABC scripts, and I'm not convinced a lot of thought has gone into them.

I've found DELAY 4 (the strategy that came from ORFS - OpenROAD-flow-scripts) has produced the best results on a number of my designs. That includes it completing much quicker, hence why I suggested giving it a go.

maliberty, 2022-12-01:
In general it is hard to predict what script will produce the best results on a given design.  The script names should suggest the common outcome (delay or area minimization) though on any given design it will vary.

I think reporting the problem to [abc](https://github.com/berkeley-abc/abc) makes sense.

xobs, 2022-12-04:
> Try `./flow.tcl -design VexRiscv -synth_explore` will give PPA based on synthesis exploration. Choose the best PPA as per design requirement. Next important thing is use of OpenRAM macros

I don't think that `-synth_explore` is very useful here:

```
[SUCCESS]: Done with synthesis exploration: See report at 'runs/RUN_2022.12.01_15.25.24/reports/synthesis/0-exploration_analysis.html'.

real    4504m8.770s
user    4260m7.903s
sys     13m23.285s
(pyenv) $
```

The good news is that `DELAY 1` produced the best result. The bad news is that it took 3.1 days to get the result. `-synth_explore` looks like it actually runs the synthesis 10x, and is therefore not very useful when the thing you want to optimize is the synthesis itself.

As far as actionable things go:

1. Perhaps a FAQ entry can be added about the synthesis / `yosys-abc` step taking a long time, asking users to check for memories?
2. Maybe the default `flow.tcl` could add a `memory_libmap` step? I've created https://github.com/VLSIDA/OpenRAM/issues/163 to try and get this.
3. Related to (2), perhaps a dummy `memory_libmap` step could be added that does nothing but error out on inferred memories, suggesting users wire up their own RAMs explicitly. I get that it's generally a good idea to do that, but making it explicit (along with an override to use logic for RAM) might be useful for newcomers to let them know that something funny is going on.

antonblanchard, 2022-12-04:
> 2. Maybe the default `flow.tcl` could add a `memory_libmap` step? I've created [Create yosys memlib mapping VLSIDA/OpenRAM#163](https://github.com/VLSIDA/OpenRAM/issues/163) to try and get this.

While this is a great long term aim, keep in mind that OpenRAM needs to characterised on every new process (via a shuttle) so it would likely be a non default option (even sky130hd at this stage).

We could however work on adding DFFRAM based macros now, since they are just standard cell based. Vendor SRAM macros are another possibility (the GF180 PDK has them for example).

antonblanchard, 2022-12-04:
> I don't think that `-synth_explore` is very useful here:

Did the log file show how long `DELAY 4` took? If it's much faster then it's another data point for changing the default synth strategy.

xobs, 2022-12-05:
It doesn't look like the logfiles include any sort of timestamp, so I can't tell. Furthermore, yosys cleans up its /tmp/yosys-abc-xxxxxx directory, so I can't look at the timestamps for the temp files.

In looking at the generated files, I suspect this means that `DELAY 4` is indeed fastest:

```
$ ls -l results/synthesis/ logs/synthesis/
logs/synthesis/:
total 47568
-rw-r--r-- 1 user user 48707526 Dec  4 18:29 0-synthesis.log

results/synthesis/:
total 1381288
-rw-r--r-- 1 user user 161679010 Dec  3 15:32 VexRiscv.AREA_0.nl.v
-rw-r--r-- 1 user user 156315215 Dec  4 03:38 VexRiscv.AREA_1.nl.v
-rw-r--r-- 1 user user 156328097 Dec  4 18:16 VexRiscv.AREA_2.nl.v
-rw-r--r-- 1 user user 171204295 Dec  4 18:29 VexRiscv.AREA_3.nl.v
-rw-r--r-- 1 user user 159082635 Dec  1 21:07 VexRiscv.DELAY_0.nl.v
-rw-r--r-- 1 user user 141429117 Dec  2 09:14 VexRiscv.DELAY_1.nl.v
-rw-r--r-- 1 user user 156176367 Dec  2 19:27 VexRiscv.DELAY_2.nl.v
-rw-r--r-- 1 user user 163662714 Dec  3 07:53 VexRiscv.DELAY_3.nl.v
-rw-r--r-- 1 user user 148509841 Dec  3 08:12 VexRiscv.DELAY_4.nl.v
$
```

I think DFFRAM macros could be a great addition, particularly for helping newcomers to avoid pitfalls like the ones I've fallen into.