The-OpenROAD-Project/OpenLane #1515: unable to resolve hold timing on a simple design

mattvenn, 2022-11-24:
### Description

This is from a Tiny Tapeout design: https://wokwi.com/projects/341571228858843732
The verilog is fetched from: https://wokwi.com/api/projects/341571228858843732/verilog

The flow fails at CTS.

It might be because there are 2 outputs that are driven directly by the clock? 

Also, I am unable to resolve the PDK mismatch, even with a completely fresh install I get this.

### Expected Behavior

Not fail CTS.

### Environment report

```shell
Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3af133706e554a740cfe60f21e773d9eaa41838c)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10084230/issue.tar.gz)

### Relevant log output

```shell
[INFO RSZ-0046] Found 1 endpoints with hold violations.
[WARNING RSZ-0064] Unable to repair all hold checks within margin.
[INFO RSZ-0032] Inserted 254 hold buffers.
[ERROR RSZ-0060] Max buffer count reached.
Error: resizer_timing.tcl, 48 RSZ-0060
child process exited abnormally
```

vijayank88, 2022-11-24:
@mattvenn 
May I get placement density value used?
Also share original configuration used to run this design

vijayank88, 2022-11-24:
@maliberty fyi

mattvenn, 2022-11-24:
Is it not in the reproducible?
Here's the config: 
```# DO NOT EDIT THIS FILE!
# We will rebuild your GDS for the final submission using the information given in the yaml file only
# If you get stuck with this config, please open an issue or get in touch via the discord.

# User config
set script_dir [file dirname [file normalize [info script]]]

# read some user config that is written by the setup.py program.
# - the name of the module is defined
# - the list of source files
source $::env(DESIGN_DIR)/user_config.tcl

# save some time
set ::env(RUN_KLAYOUT_XOR) 0
set ::env(RUN_KLAYOUT_DRC) 0

# don't put clock buffers on the outputs
set ::env(PL_RESIZER_BUFFER_OUTPUT_PORTS) 0

# allow use of specific sky130 cells
set ::env(SYNTH_READ_BLACKBOX_LIB) 1

# put all the pins on the left
set ::env(FP_PIN_ORDER_CFG) $::env(DESIGN_DIR)/pin_order.cfg

# reduce wasted space
set ::env(TOP_MARGIN_MULT) 2
set ::env(BOTTOM_MARGIN_MULT) 2

# absolute die size
set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 150 170"
set ::env(FP_CORE_UTIL) 55
set ::env(PL_BASIC_PLACEMENT) {1}

set ::env(FP_IO_HLENGTH) 2
set ::env(FP_IO_VLENGTH) 2

# use alternative efabless decap cells to solve LI density issue
set ::env(DECAP_CELL) "\
    sky130_fd_sc_hd__decap_3 \
    sky130_fd_sc_hd__decap_4 \
    sky130_fd_sc_hd__decap_6 \
    sky130_fd_sc_hd__decap_8 \
    sky130_ef_sc_hd__decap_12"

# clock
set ::env(CLOCK_TREE_SYNTH) 1
set ::env(CLOCK_PERIOD) "100000"
set ::env(CLOCK_PORT) {io_in[0]}

# hold/slack margin
# set ::env(PL_RESIZER_HOLD_SLACK_MARGIN) 0.8 
# set ::env(GLB_RESIZER_HOLD_SLACK_MARGIN) 0.8 

# don't use power rings or met5
set ::env(DESIGN_IS_CORE) 0
set ::env(RT_MAX_LAYER) {met4}

# connect to first digital rails
set ::env(VDD_NETS) [list {vccd1}]
set ::env(GND_NETS) [list {vssd1}]
```

and user_config.tcl:

```set ::env(DESIGN_NAME) user_module_349209305274122835
set ::env(VERILOG_FILES) "\
    $::env(DESIGN_DIR)/user_module_349209305274122835.v \
    $::env(DESIGN_DIR)/cells.v"
```

vijayank88, 2022-11-24:
@mattvenn 
Following only path causing the issue:
```
Startpoint: io_in[1] (input port clocked by io_in[0])
Endpoint: _01_ (rising clock gating-check end-point clocked by io_in[0])
Path Group: **clock_gating_default**
Path Type: min

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock io_in[0] (rise edge)
                          0.00    0.00   clock network delay (ideal)
                       20000.00 20000.00 ^ input external delay
                  0.01    0.01 20000.01 ^ io_in[1] (in)
     1    0.00                           io_in[1] (net)
                  0.01    0.00 20000.01 ^ _01_/A (sky130_fd_sc_hd__and2_2)
                               20000.01   data arrival time

                  0.15 50000.00 50000.00   clock io_in[0] (fall edge)
                          0.00 50000.00   clock network delay (ideal)
                          0.25 50000.25   clock uncertainty
                          0.00 50000.25   clock reconvergence pessimism
                               50000.25 v _01_/B (sky130_fd_sc_hd__and2_2)
                          0.00 50000.25   clock gating hold time
                               50000.25   data required time
-----------------------------------------------------------------------------
                               50000.25   data required time
                               -20000.01   data arrival time
-----------------------------------------------------------------------------
                               -30000.25   slack (VIOLATED)
```
By updating sdc file with input delay 
`set_input_delay 60000.0000 -clock [get_clocks {io_in[0]}] -add_delay [get_ports {io_in[1]}]` 
passed without violations.

maliberty, 2022-11-24:
@mattvenn did you intend for this to be a 1/2 cycle path?

mattvenn, 2022-11-25:
@maliberty did you see the design? 
[image]

I can't see how this would be half cycle.

maliberty, 2022-11-25:
You can see it in the timing report you posted:
clock io_in[0] (**rise edge**)
...
0.15 50000.00 50000.00   clock io_in[0] (**fall edge**)

That's what the SDC says and so what OR sees.

mattvenn, 2022-11-25:
but why would STA get that from the schematic above?

maliberty, 2022-11-25:
You have to hold the clock gating signal until the clock falls.  This check is automatically generated for a clock gate (note group clock_gating_default).

What is valid period for the clock gate control signal?  Is it just a constant?

maliberty, 2022-11-25:
If you flip the switch while the clock button is depressed you could generate a bad clock pulse from the clock gate.

mattvenn, 2022-11-25:
Of course thanks Matt.

closed 2023-05-10 by donn