The-OpenROAD-Project/OpenLane #1518: JSON config: FP_PDN_MACRO_HOOKS unexpected behavior

mole99, 2022-11-26:
### Description

Previously I set `FP_PDN_MACRO_HOOKS` in the tcl configuration as follows:

```
set ::env(FP_PDN_MACRO_HOOKS) "\
	wfg_top_inst                            vccd1 vssd1 vccd1 vssd1, \
	merge_memory_inst                       vccd1 vssd1 vccd1 vssd1, \
	wb_mux_inst                             vccd1 vssd1 vccd1 vssd1, \
	wb_memory_inst                          vccd1 vssd1 vccd1 vssd1, \
	sky130_sram_2kbyte_1rw1r_32x512_8_inst0 vccd1 vssd1 vccd1 vssd1, \
	sky130_sram_2kbyte_1rw1r_32x512_8_inst1 vccd1 vssd1 vccd1 vssd1"
```

Now that OpenLane is moving to json configurations, I tried setting FP_PDN_MACRO_HOOKS in a similar way, which was wrong.

```
"FP_PDN_MACRO_HOOKS": [
    "wfg_top_inst                            vccd1 vssd1 vccd1 vssd1",
    "merge_memory_inst                       vccd1 vssd1 vccd1 vssd1",
    "wb_mux_inst                             vccd1 vssd1 vccd1 vssd1",
    "wb_memory_inst                          vccd1 vssd1 vccd1 vssd1",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst0 vccd1 vssd1 vccd1 vssd1",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst1 vccd1 vssd1 vccd1 vssd1"
],
```

Only the entry for `wfg_top_inst` will be connected to vccd1/vssd1.

The correct way is to also use a comma at the end of the string:

```
"FP_PDN_MACRO_HOOKS": [
    "wfg_top_inst                            vccd1 vssd1 vccd1 vssd1,",
    "merge_memory_inst                       vccd1 vssd1 vccd1 vssd1,",
    "wb_mux_inst                             vccd1 vssd1 vccd1 vssd1,",
    "wb_memory_inst                          vccd1 vssd1 vccd1 vssd1,",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst0 vccd1 vssd1 vccd1 vssd1,",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst1 vccd1 vssd1 vccd1 vssd1"
],
```

This looks a bit strange and is not what I expected. Is this intentional? If yes, maybe add an example json config to the documentation. Thanks!

### Expected Behavior

I expect this to be the correct way how to set `FP_PDN_MACRO_HOOKS`:

```
"FP_PDN_MACRO_HOOKS": [
    "wfg_top_inst                            vccd1 vssd1 vccd1 vssd1",
    "merge_memory_inst                       vccd1 vssd1 vccd1 vssd1",
    "wb_mux_inst                             vccd1 vssd1 vccd1 vssd1",
    "wb_memory_inst                          vccd1 vssd1 vccd1 vssd1",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst0 vccd1 vssd1 vccd1 vssd1",
    "sky130_sram_2kbyte_1rw1r_32x512_8_inst1 vccd1 vssd1 vccd1 vssd1"
],
```

### Environment report

```shell
Non applicable
```

### Reproduction material

Non applicable

### Relevant log output

```shell
Non applicable
```

mole99, 2022-11-26:
I want to add, that in the tutorial [Designing a chip with an OpenRAM (sky130)](https://openlane.readthedocs.io/en/latest/tutorials/openram.html?highlight=fp_pdn_macro_hooks) `FP_PDN_MACRO_HOOKS` is specified as one long string:

`"FP_PDN_MACRO_HOOKS": "submodule.sram0 vccd1 vssd1 vccd1 vssd1, submodule.sram1 vccd1 vssd1 vccd1 vssd1",`

This is also a correct way to set `FP_PDN_MACRO_HOOKS`, but for a lot of macros this line becomes rather long.

donn, 2022-11-30:
Hi- this is unfortunately a limitation of how the config files work because of the Tcl programming language continuing to be an embarrassment: arrays are just space-delimited strings by the time they reach the OpenLane code. Meaning the commas are still needed to delimit the strings unfortunately, as unintuitive as it is. I can add a documentation example while we try to figure something out.

PR #1536 (2022-11-30): Add test for #1506, Isolated Installation in Docs
\+ Add test for JSON config files
~ Updated various `rm` execs to `file delete -force`
~ Test script now outputs to stdout as well as a log file
~ Moved post-run hook runner to `all.tcl`
~ Moved installation docs to their own subcategory
~ Add note on comma delimited variables to docs

---
Resolves #1518
closed 2022-11-30 by donn