The-OpenROAD-Project/OpenLane #1519: Unable to sintencie previous desing

jaquerinte, 2022-11-26:
### Description

I'm trying to synthesize  a previous MPW-2 design in the new Openlane for the gf180 PDK but always gets to:
 "Reached 20 congestion iterations with less than 15% of reduction between iterations. Routing congestion too high. Check the congestion heatmap in the GUI."
And I try to use as much space as possible and the lowest possible PL_TARGET_DENSITY but always gets to that message. 

### Expected Behavior

Openlane finish execution

### Environment report

```shell
Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.20 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/storage/programs/chip_making/space_shuttle_gf_node/dependencies/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/storage/programs/chip_making/space_shuttle_gf_node/dependencies/openlane_src/dependencies/verify_versions.py", line 72, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/storage/programs/chip_making/space_shuttle_gf_node/dependencies/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[Uploading issue.tar.gz…]()

### Relevant log output

```shell
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 392828
[INFO GRT-0198] Via related Steiner nodes: 33050
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 652988
[INFO GRT-0112] Final usage 3D: 4188682
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_timing.tcl, 39 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
```

maliberty, 2022-11-26:
You need to actually attach a test case

jaquerinte, 2022-11-26:
Just added

maliberty, 2022-11-27:
Is there a reason you eliminated metal5 for routing?

set ::env(RT_MAX_LAYER) {Metal4};

maliberty, 2022-11-27:
Also I see
```
>>> report_design_area
Design area 2029853 u^2 58% utilization.
```
I don't know what you mean by "lowest possible PL_TARGET_DENSITY" but that isn't low utilization.

jaquerinte, 2022-11-27:
Hi, for metal5, I think is needed for the other hardening (user_project_wrapper). For the "lowest possible PL_TARGET_DENSITY", what I mean, is that if I use a lower PL_TARGET_DENSITY ( in order to reduce congestion for the routing) , openlane fails in early stages for not having enough area. I will try to reproduce also that.

jaquerinte, 2022-11-27:
This is what I mean, the smollest PL_TARGET_DENSITY that I can use is 0.54
###Output Error
`[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: ../home/storage/programs/chip_making/space_shuttle_gf_node/openlane/user_proj_example/runs/22_11_27_09_43/logs/placement/7-global.log
[ERROR]: Last 10 lines:
[InitialPlace]  Iter: 16 CG residual: 0.00003643 HPWL: 5435014147
[InitialPlace]  Iter: 17 CG residual: 0.00003801 HPWL: 5424392408
[InitialPlace]  Iter: 18 CG residual: 0.00003557 HPWL: 5408534568
[InitialPlace]  Iter: 19 CG residual: 0.00003596 HPWL: 5394566423
[InitialPlace]  Iter: 20 CG residual: 0.00003553 HPWL: 5400590082
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.42
Suggested target density: 0.54
Error: gpl.tcl, 69 GPL-0302
child process exited abnormally`

###Issue Reproducible 
[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10097865/issue.tar.gz)

vijayank88, 2022-11-28:
@jaquerinte 
Can you try increase die area https://github.com/jaquerinte/space_shuttle_gf_node/blob/a106f6c28fb2e1e16b5a4e5e60e127136ecc0e62/openlane/state_counters/config.json#L41
`"DIE_AREA": "0 0 1850 1850",`

jaquerinte, 2022-11-28:
Hi, @vijayank88,  the state_counters is a sub part of the design, that I divided in multiple hardening macros just to test if it will work having the most dense parts separated. (been the most dense the state_counters.v and the register_data.v). But yes I try with the size that you provided, and I get the same thing too much congestion. 
```
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: ../home/storage/programs/chip_making/space_shuttle_gf_node/openlane/state_counters/runs/22_11_28_09_25/logs/routing/12-resizer.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 175395
[INFO GRT-0198] Via related Steiner nodes: 13954
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 327430
[INFO GRT-0112] Final usage 3D: 2086755
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_timing.tcl, 39 GRT-0118
child process exited abnormally
```

jaquerinte, 2022-11-28:
There is any way to disble the limitation of less that 15% in 20 iterations?

vijayank88, 2022-11-28:
@jaquerinte Attach `issue_reproducible` generated for above failure

jaquerinte, 2022-11-28:
Attach is the issue_reproducible for last test.

[issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10102866/issue.tar.gz)

For the congestion, I will check how to do it and I will pass the report.

maliberty, 2022-11-28:
> There is any way to disble the limitation of less that 15% in 20 iterations?

-allow_overflow

maliberty, 2022-11-28:
What does "Hi, for metal5, I think is needed for the other hardening (user_project_wrapper)." mean?  I haven't observed other mpw designs to require eliminating metal5.

jaquerinte, 2022-11-28:
> What does "Hi, for metal5, I think is needed for the other hardening (user_project_wrapper)." mean? I haven't observed other mpw designs to require eliminating metal5.
I use the user project example for GF180MCU, in this one they only use up to metal4, I try to rise it to metal5 but same result. 

I will try to use the allow_overflow and I will comment with that

armleo, 2022-11-29:
The floorplan has 59% utilization.
The target density need to be like 64%.

Setting it lower than 59% is gonna always fail the placement.

Increase the die area or optimize your design.

jaquerinte, 2022-11-29:
What is the outcome of the project if -allow_overflow is enable? I mean the output will still be correct or only makes more difficult the next steps?

armleo, 2022-11-29:
MPW-2 is sky130, you are expecting the same floorplan to be used for GF180? Despite GF180 being area-wise bigger than sky130?

jaquerinte, 2022-11-29:
As close as it can be with the original, in the MPW-2 was only using 50% of the area

armleo, 2022-11-29:
I would definitely try decreasing the util and density to around util=45%/density=50% to see if it helps.

jaquerinte, 2022-11-29:
ok, I will try that. I try with -allow_overflow this night and that made it to global routing, but were like 1.777.550 violations, and each iteration was between 40 minutes and 6 hours. So I don't know if that will generate a valid desing.

closed 2023-04-13 by donn