The-OpenROAD-Project/OpenLane #1523: Synthesis - ABC: + rewrite failure when running yosys-abc

ramrajrl, 2022-11-28:
### Description

Yosys Synthesis Fails when running ABC, this error happens on a relatively big arithmetic circuit optimization. 

Please check this thread from YosysHQ for the details of the issue.
https://github.com/YosysHQ/yosys/issues/174#issue-157182697

In Openlane setup -- it is this file --> `scripts/yosys/synth.tcl`

### Expected Behavior

> This error happen on a relatively big arithmetic circuit optimization. I have simplified code and post few examples here. They should lead to equivalent synthesis flow. But in some cases ABC fail and in other it is not. Theoretically it could be ABC issue, 

> I've been in email contact with Alan Mishchenko, the author of ABC, regarding this problem.
> 
> This seems to be a known problem with ABC's `rewrite` command with "highly redundant netlists with deep logic". Alan says this is probably not going to be fixed anytime soon because it means a lot of work in obsolete code. He suggests to simply use the newer `drw` command instead of `rewrite`.
> 
> I've now added transforming divide and modulo by a constant power of two to simpler operations in commit [11f7b8a](https://github.com/YosysHQ/yosys/commit/11f7b8a2a1dd6e5c47e8081e4485127331fd80be). This avoids generating "highly redundant netlists" for the example you gave in the first place.

Please use the newer `drw` command instead of `rewrite` in `scripts/yosys/synth.tcl`

### Environment report

```shell
openlane.log:ERROR: ABC: execution of command ""/build/bin/yosys-abc" -s -f /tmp/yosys-abc-IhLev5/abc.script 2>&1" failed: return code 134.

71.1.1. Executing ABC.
Running ABC command: "/build/bin/yosys-abc" -s -f /tmp/yosys-abc-tRZss0/abc.script 2>&1

ABC: + refactor
ABC: An error occurred during computation. The original network is restored.
ABC: + balance
ABC: + rewrite
ABC: yosys-abc: src/base/abc/abcAig.c:167: void abc::Abc_AigFree(abc::Abc_Aig_t*): Assertion `Vec_PtrSize( pMan->vStackReplaceOld ) == 0' failed.
ABC: Flags A, B, or C are not zero.
ERROR: ABC: execution of command ""/build/bin/yosys-abc" -s -f /tmp/yosys-abc-tRZss0/abc.script 2>&1" failed: return code 134.
```

### Reproduction material

Please refer to this for the specific testcase, attached makefile and sources ([testcase.zip](https://github.com/cliffordwolf/yosys/files/286310/testcase.zip)) to reproduce problem.

https://github.com/YosysHQ/yosys/issues/174#issue-157182697

### Relevant log output

```shell
+ fx 
Abc_NtkFastExtract: Nodes have duplicated fanins. FX is not performed.
+ mfs 
+ strash 
+ refactor 
An error occurred during computation. The original network is restored.
+ balance 
+ drw 
+ refactor 
+ balance 
+ drw 
+ drw -z 
+ balance 
+ refactor -z 
+ drw -z 
+ balance 
+ retime -D -D 10000 -M 6 
+ scleanup 
Error: The network is combinational.
+ map -p -B 0.2 -A 0.9 -M 0 
The cell areas are multiplied by the factor: <num_fanins> ^ (0.90).
The cell delays are multiplied by the factor: <num_fanins> ^ (0.20).
+ retime -D -D 10000 
+ &get -n 
+ &st 
+ &dch 
+ &nf 
+ &put 
+ buffer -N 25 -S 750.0 
+ upsize -D 10000 
Current delay (5733.16 ps) does not exceed the target delay (10000.00 ps). Upsizing is not performed.
+ dnsize -D 10000                                                                                                                                 
+ stime -p    

With drw instead of rewrite, ABC works fine
```

antonblanchard, 2022-11-28:
This would also likely work:

```
set ::env(SYNTH_STRATEGY) {DELAY 4}
```

Which is the ORFS synth strategy. @maliberty perhaps we should make it the default?

Do we even know where the other strategies came from? We could incorporate the fix suggested, but there are other things like register retiming which never worked (we don't pass the FFs into abc so it has no ability to retime). Makes me wonder if they got copied blindly from somewhere else.

maliberty, 2022-11-28:
I think that is a decision for the OL team (@donn @shalan )

I know I added in the ORFS strategies to the existing ones but I don't know their origin.

vijayank88, 2022-11-29:
@ramrajrl 
Based your requirement you can choose following `SYNTH_STRATEGY` design focus to meet area or delay.

https://openlane.readthedocs.io/en/latest/reference/configuration.html#optional-variables

```
Strategies for abc logic synthesis and technology mapping
Possible values are DELAY/AREA 0-4/0-3; the first part refers to the optimization target of the synthesis strategy (area vs. delay) and the second one is an index.
(Default: AREA 0)
```

ramrajrl, 2022-11-29:
> @ramrajrl Based on your requirement you can choose following `SYNTH_STRATEGY` for design focus to meet area or delay.
None of the strategies use `drw` instead of `rewrite` in scripts/yosys/synth.tcl - please see https://github.com/YosysHQ/yosys/issues/174#issuecomment-222353448 on why `drw` is recommended by yosys-abc

antonblanchard, 2022-11-29:
> None of the strategies use `drw` instead of `rewrite` in scripts/yosys/synth.tcl - please see [YosysHQ/yosys#174 (comment)](https://github.com/YosysHQ/yosys/issues/174#issuecomment-222353448) on why `drw` is recommended by yosys-abc

Did you try `set ::env(SYNTH_STRATEGY) {DELAY 4}`?

ramrajrl, 2022-11-29:
> Did you try `set ::env(SYNTH_STRATEGY) {DELAY 4}`?

Sorry, yes I did. The synthesis/yosys-abc does work with this setting/strategy, I'm waiting for the PPA to come through on this. I assumed ` DELAY` was a 0-3 setting. Could you please share a link to some details on these `SYNTH_STRATEGY` settings?

vijayank88, 2022-11-29:
> @ramrajrl Based on your requirement you can choose following `SYNTH_STRATEGY` for design focus to meet area or delay.
> 
> https://openlane.readthedocs.io/en/latest/reference/configuration.html#optional-variables
> 
> ```
> Strategies for abc logic synthesis and technology mapping
> Possible values are DELAY/AREA 0-4/0-3; the first part refers to the optimization target of the synthesis strategy (area vs. delay) and the second one is an index.
> (Default: AREA 0)
> ```

@ramrajrl

ramrajrl, 2022-11-29:
Thanks - I looked at it again, I get the options are `DELAY 0-4 and AREA 0-3` this format has clarity - may I request the document to be updated to this please? Looking for details on what the different strategies are? I can see that DELAY-4 does not run `rewrite` obviously synthesis wont fail.

donn, 2023-04-10:
You can find the commands here: https://people.eecs.berkeley.edu/~alanmi/abc/

jix, 2023-11-19:
@donn What was done to address this issue? While looking into https://github.com/YosysHQ/yosys/issues/4039 I noticed that OpenLane still uses the obsolete and known broken `rewrite` command as reported in this issue. The new issue is for a crash in `refactor` before reaching `rewrite`, but given that `rewrite` is obsolete and got replaced by `drw` and there is a `drf` command that performs refactoring I wouldn't be surprised if `refactor` is also considered obsolete and is not maintained by upstream abc anymore. In any case, yosys uses neither command so they are not covered by our tests. (Note that the abc website you linked has been last updated in 2012 and abc's author mentioned in 2016 that `rewrite` has known issues with no plans to fix them as `drw` can be used as replacement.)

donn, 2023-11-20:
I think this was closed by mistake- going to keep track of this in #2051.

closed 2023-11-20 by donn