The-OpenROAD-Project/OpenLane #1525: openroad or_gui is not working

ParvanehS12, 2022-11-28:
### Description

I want to open the openroad gui environment but after writing or_gui in interactive mode I am seeing this note, and no gui environment is opening:

can't read "global_verbose_level": no such variable.

Will appreciate it if anyone could help with this.

maliberty, 2022-11-28:
What exactly are you doing that produces that message?

ParvanehS12, 2022-11-28:
> What exactly are you doing that produces that message?

Just want to follow the steps of this example:
https://openlane.readthedocs.io/en/latest/tutorials/digital_guide.html

and I am facing this issue at the "Analyzing the flow-generated files" step:

./flow.tcl -design mem_1r1w -tag full_guide -interactive
package require  openlane
or_gui

maliberty, 2022-11-28:
does "./flow.tcl -design mem_1r1w -tag full_guide -gui" work?

ParvanehS12, 2022-11-28:
> does "./flow.tcl -design mem_1r1w -tag full_guide -gui" work?

It is failing too due to this issue:

"Could not connect to any X display".

maliberty, 2022-11-28:
That's most likely the same as https://github.com/The-OpenROAD-Project/OpenLane/issues/1149

closed 2022-11-30 by donn

yunuseryilmaz18, 2022-12-28:
Hello, I've been dealing with the same problem for the same reason. When I've used `./flow.tcl -design mem_1r1w -tag full_guide -gui`, it gives this error:
```
[ERROR]: during executing openroad script /openlane/scripts/openroad/gui.tcl
[ERROR]: Log: ../dev/null
[ERROR]: Last 10 lines:
13# 0x00007F02982BD3F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F02A03C23B5 in /lib64/libQt5DBus.so.5
19# 0x00007F029C6DEE71 in /lib64/libQt5Core.so.5
20# 0x00007F029FE43EA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

[ERR] 0 not found.
[ERROR]: Failed to package reproducible.
```

What should I do?

ParvanehS12, 2022-12-28:
> [EXTERNAL EMAIL] DO NOT CLICK unless you know the content is safe..

Hi,

I switched to windows 11 and it is solved.

yunuseryilmaz18, 2022-12-29:
Thanks for the answer but I use OpenLane in Ubuntu, don't have Windows.

maliberty, 2022-12-29:
Now that https://github.com/The-OpenROAD-Project/OpenLane/issues/1149 is fixed can you update and try again?

yunuseryilmaz18, 2022-12-30:
At first, didn't solve it but now `openroad -gui` works for me when mounting the image with --privileged switch. Thank you.