The-OpenROAD-Project/OpenLane #1530: Yosys verilog reader enhancement request

shakouri, 2022-11-29:
### Description

The yosys verilog reader is not able to parse verilog code that other synthesis tools are able to parse.

1.  Allow parameter passing w/o use of parantheses ('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]

2. Allow use of ? during async portion of always block ('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]
[image]

### Proposal

The yosys verilog reader will need to be enhanced.

mithro, 2022-11-29:
This is just an FYI;
> There are a couple of potential options for improve language support in Yosys;
>  * The TabbyCAD suite from YosysHQ (https://www.yosyshq.com/tabby-cad-datasheet) which includes the proprietary Verific frontend.
>  * The Surelog->UHDM->Yosys flow found @ https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin
>  * ZachJS's sv2v tool found @ https://github.com/ZachJS/sv2v
> 
> I have a diagram of the last two @ https://j.mp/sv-flow-diagram

shakouri, 2022-11-29:
I tried sv2v, but the tool is not also able to read it either. This is error that it gives:
   Parse error: unexpected token '4' (Lit_number)

I am getting the following error when I try to use system verilog plugin:
yosys> plugin -i systemverilog
ERROR: Can't load module `./systemverilog': dlopen(/usr/local/bin/../share/yosys/plugins/systemverilog.so, 5): no suitable image found.  Did find:
	/usr/local/bin/../share/yosys/plugins/systemverilog.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03
	/usr/local/Cellar/yosys/0.23/share/yosys/plugins/systemverilog.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03

mithro, 2022-11-29:
@shakouri - Please report the systemverilog issues to the issue tracker @ https://github.com/chipsalliance/yosys-f4pga-plugins/issues

shakouri, 2022-11-30:
Ok. Here is the link:
https://github.com/chipsalliance/yosys-f4pga-plugins/issues/412