The-OpenROAD-Project/OpenLane #1531: Step-26, [ERROR]: during executing magic script

Baungarten-CINVESTAV, 2022-11-29:
### Description

Hello,
I am trying to harden a configurable logic block for the new MPW-8 from efabless but I get the following error. It is strange because this error does not appear with the previous version of OpenLane (c40bcaca582bb1926062656265d49a582fb9ee3ba1be). I mean this block was hardened with this earlier version.

I would like to know if the problem is a bug or if I am doing a wrong configuration.

### Expected Behavior

[image]

### Environment report

```shell
OpenLane Container (5035e1e):/openlane$ python3 ./env.py issue-survey
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-53-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 5035e1e6e2a58783683b6ca5b4e010f76394a3be
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

5035e1e 2022-11-29T10:39:55+02:00 Magic lef write script cleanup (#1527) - Anton Blanchard -  (HEAD -> master, origin/master, origin/HEAD)
235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (tag: 2022.11.29)
78fee2a 2022-11-28T12:44:22+02:00 [BOT] Update openroad_app (#1501) - Openlane Bot -  ()
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10117199/issue_reproducible.zip)

### Relevant log output

```shell
[ERROR]: during executing magic script /openlane/scripts/magic/def/mag_gds.tcl
[ERROR]: Log: designs/grid_clb/runs/RUN_2022.11.29_20.33.59/logs/signoff/26-gdsii.log
[ERROR]: Last 10 lines:
Moving label "net61" from metal1 to viali in cell grid_clb.
Moving label "net82" from metal2 to via2 in cell grid_clb.
Moving label "net90" from metal2 to via2 in cell grid_clb.
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )  area (units^2)

microns:  230.000 x 230.000  ( 0.000,  0.000), ( 230.000,  230.000)  52900.000 
lambda:   23000.00 x 23000.00  (  0.00,  0.00 ), ( 23000.00,  23000.00)  529000000.00
internal:  46000 x 46000   (     0,  0    ), ( 46000,  46000)  2116000000
child killed: segmentation violation

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
```

antonblanchard, 2022-11-30:
You might have a problem, but magic should never SEGV. Reproduces on latest magic:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007fffe965aea2 in DBCellWriteFile (cellDef=0x9e41c0, f=0x9da3a0) at DBio.c:3204
3204		    if (lab->lab_flags & PORT_DIR_MASK)
(gdb) backtrace 
#0  0x00007fffe965aea2 in DBCellWriteFile (cellDef=0x9e41c0, f=0x9da3a0) at DBio.c:3204
#1  0x00007fffe965bd9e in DBCellWrite (cellDef=0x9e41c0, fileName=0x5ad9c0 "./results/signoff/grid_clb") at DBio.c:3573
#2  0x00007fffe961df03 in cmdSaveCell (cellDef=0x9e41c0, newName=0x0, noninteractive=0 '\000', tryRename=1 '\001') at CmdSubrs.c:641
#3  0x00007fffe9636460 in CmdSave (w=0x4ad550, cmd=0x9f22c0) at CmdRS.c:231
#4  0x00007fffe970fdd7 in WindExecute (w=0x4ad550, rc=5561936, cmd=0x9f22c0) at windMain.c:418
#5  0x00007fffe968335a in DBWcommands (w=0x4ad550, cmd=0x9f22c0) at DBWprocs.c:639
#6  0x00007fffe970c934 in WindSendCommand (w=0x4ad550, cmd=0x9f22c0, quiet=1 '\001') at windSend.c:263
#7  0x00007fffe9704c46 in TxTclDispatch (clientData=0x0, argc=1, argv=0x420090, quiet=1 '\001') at txCommands.c:1184
#8  0x00007fffe9829d8e in _tcl_dispatch (clientData=0x0, interp=0x4129c0, argc=1, argv=0x420090) at tclmagic.c:400
#9  0x00007ffff7e297a2 in TclInvokeStringCommand () from /lib64/libtcl8.6.so
#10 0x00007ffff7e2a970 in TclNRRunCallbacks () from /lib64/libtcl8.6.so
#11 0x00007ffff7eb8e5c in Tcl_RecordAndEvalObj () from /lib64/libtcl8.6.so
#12 0x00007ffff7ee2b77 in Tcl_MainEx () from /lib64/libtcl8.6.so
#13 0x0000000000401234 in main (argc=5, argv=0x7fffffffd7c8) at magicdnull.c:49
```

Could you please submit an issue against magic https://github.com/RTimothyEdwards/magic

Baungarten-CINVESTAV, 2022-12-01:
Hi @antonblanchard , I have opened an issue at magic repo [(https://github.com/RTimothyEdwards/magic/issues/205)](https://github.com/RTimothyEdwards/magic/issues/205) they have done an update (version 8.3.348) and they say they have fixed the problems.

The problem now is that when I run the OpenLane environment it works with an old version of Magic 8.3.346 and not with the recent version 8.3.348. I honestly don't know how to tell OpanLane to work with the version of Magic I have installed on my computer.

Additionally, I don't know if I have to make some changes or if the changes have to happen in openLane saying that it has to use the most recent version of magic.

Attached a screenshot of magic version on my computer and on OpenLane
[image]

antonblanchard, 2022-12-01:
@Baungarten-CINVESTAV I've requested for magic to be updated, once that happens and a new docker build completes you should be good.

Baungarten-CINVESTAV, 2022-12-01:
I can confirm that the problem was solved with the magic update.

closed 2022-12-01 by Baungarten-CINVESTAV

msaideroglu, 2022-12-07:
When I clone latest OpenLane repo and run "make test" spm, it runs with Magic's recent version 8.3.348. However when I run make mount, it pulls additional docker images, and Magic's version turns into 8.3.331. And I face same problem as described above, during Caravel flow.