The-OpenROAD-Project/OpenLane #1533: GF180MCU PEX extraction with OpenROAD is likely incorrect

QuantamHD, 2022-11-30:
### Description

Parasitic extraction through OpenRCX is likely incorrect because of a limitation with OpenRCX at the moment. The OpenROAD were able to generate a good deck located at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/656

OpenLane should adopt the same set RCs

### Expected Behavior

OpenLANE PEX should be accurate

### Environment report

```shell
...
```

### Reproduction material

...

### Relevant log output

```shell
...
```

QuantamHD, 2022-11-30:
@jeffdi @mkkassem

donn, 2022-11-30:
s/OpenLANE/OpenLane/g

antonblanchard, 2023-01-03:
I think this was resolved in https://github.com/RTimothyEdwards/open_pdks/commit/9c6fd95d4d02c02af005dc344804841fe40d7f3c

I see some small differences between what went into ORFS and open_pdks, (eg the corner via resistances).

closed 2024-09-12 by donn