The-OpenROAD-Project/OpenLane #1534: PDNSim: comparsion with Cadence Voltus

ZFTurbo, 2022-11-30:
### Description

Hello. 

We were experimenting with Openlane and we were interested in the results of IRDrop analysis obtained with PDNSim tool. Results of PDNSim has extremely small values of voltage (nV). To compare the obtained results, we performed a similar static analysis using the Cadence Voltus tool based on the same library (SkyWater130a) and the resulting output topology from Openlane. We got completely different results, with a different order of voltage (mV). The results of both analyzes are attached. What can be the source of such big difference?

We prepeared archive with our results. 
https://www.dropbox.com/s/fyt2uxz851favst/voltus_vs_pdnsim.zip?dl=1

Archive contains the results of IRdrop analysis for the PPU project from the OpenLane test suite obtained by two programs: PDNSim and Cadence Voltus.

Archive consists of:
1) pdk folder contains files from PDK SKYWATER130a, which were used for PDNSim and Voltus analysis
2) pdnsim folder contains results, obtained in standard pipeline of OpenLane in final_results folder. Also it contains log and report files for IRDrop. File analyze_irdrop_openroad.rpt is contained results of command "analyze_power_grid -net VPWR" in OpenROAD (with more digits after dot). 
3) Folder voltus_results contains results, which were obtained in Cadence Voltus on the basis of topology created in OpenLane (def/lef files) with similar parameters and input files as PDNSim. We did 2 runs in Voltus with different parameters.
4) check_*_match.csv contains gathered data about voltage on instances, which were found both in PDNSim and Voltus reports.
5) check_*_all.csv contains gathered data about voltage on all instances.

### Expected Behavior

We expect results of PDNSim and Voltus to be similar

### Environment report

```shell
Kernel: Linux v3.10.0-1160.76.1.el7.x86_64
Distribution: centos 7
Python: v3.10.5 (OK)
Container Engine: docker v20.10.20 (OK)
OpenLane Git Version: c40bcaca582bb192606265d49a582fb9ee3ba1be
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c40bcac 2022-09-22T07:26:28+02:00 [BOT] Update PDK (#1389) - Openlane Bot -  (HEAD -> master, tag: 2022.09.23, origin/master, origin/HEAD)
c8f917e 2022-09-22T07:23:38+02:00 Correctly decorate wrapper in click_odb (#1381) - Matt Liberty -  ()
f81bd81 2022-09-22T07:21:39+02:00 Correct the installation link (#1384) - Arman Avetisyan -  ()
---
Git Remotes

origin    https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin    https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

https://www.dropbox.com/s/fyt2uxz851favst/voltus_vs_pdnsim.zip?dl=1

### Relevant log output

```shell
---
```

ZFTurbo, 2022-11-30:
PDNSim:
[image]

Voltus:
[image]

vijayank88, 2022-11-30:
@dralabeing fyi

donn, 2022-11-30:
@maliberty I'm guessing this has something to do with how OpenLane calls OpenROAD- but any insight?

maliberty, 2022-11-30:
@VidyaChhabria would you take a look?

tspyrou, 2022-11-30:
@ZFTurbo its a violation to post proprietary tool information. I have to delete this issue.  cc @maliberty @VidyaChhabria