The-OpenROAD-Project/OpenLane #1535: yosys exit with gf180mcu full liberty files

kareefardi, 2022-11-30:
### Description

yosys exits with a syntax error

### Expected Behavior

yosys passing

### Environment report

```shell
❯ python3 ./env.py issue-survey
Kernel: Linux v5.14.0-1054-oem
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (HEAD -> master, origin/master, origin/HEAD)
78fee2a 2022-11-28T12:44:22+02:00 [BOT] Update openroad_app (#1501) - Openlane Bot -  ()
0858705 2022-11-28T10:12:20+02:00 Fix manual macro placement on GF180 (#1521) - Anton Blanchard -  ()
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10122909/issue_reproducible.tar.gz)

### Relevant log output

```shell
Reading /openlane/designs/spm/runs/missing-func/tmp/synthesis/1-gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.no_pg.lib as a blackbox

1. Executing Liberty frontend.
ERROR: Syntax error in liberty file on line 274514.
```
```

vijayank88, 2022-11-30:
@kareefardi 
Is that built in design `spm` not able to run with gf180 pdk?
I tried with latest version `OpenLane 5035e1e6e2a58783683b6ca5b4e010f76394a3be` flow completed.
`./flow.tcl -design spm`

growly, 2022-12-03:
I have this issue too. I have a dead-simple design which tries to instantiate a single mux4_1:

```
module bfg_mux_test #(
    parameter INPUTS=4, SELECTORS=2
) (
    // IO
    input [INPUTS-1:0] inputs, 
    input [SELECTORS-1:0] selectors,
    output bfg_out,
    output baseline_out,

    inout VDD,
    inout VSS
);

gf180mcu_fd_sc_mcu7t5v0__mux4_1 baseline_mux4(
  .I0(inputs[0]),
  .I1(inputs[1]),
  .I2(inputs[2]),
  .I3(inputs[3]),
  .S0(selectors[0]),
  .S1(selectors[1]),

  .Z(baseline_out),

  .VDD(VDD),
  .VSS(VSS)
);

endmodule
```

Following the OpenLane [instructions](https://openlane-docs.readthedocs.io/en/rtd-develop/doc/chip_integration.html) I set 
```
    "SYNTH_READ_BLACKBOX_LIB": 1,
```

and the flow crashes with:

```
[INFO]: Running Synthesis (log: designs/bfg_mux_test/runs/RUN_2022.12.03_09.04.23/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/bfg_mux_test/runs/RUN_2022.12.03_09.04.23/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
Reading /openlane/designs/bfg_mux_test/runs/RUN_2022.12.03_09.04.23/tmp/synthesis/1-gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.no_pg.lib as a blackbox

1. Executing Liberty frontend.
ERROR: Syntax error in liberty file on line 274514.
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

kareefardi, 2022-12-03:
@growly A fix was deployed in the pdk sources and will be propagated to openlane. Untill then, you can use `(* keep *)` for instances that you don't want yosys to remove theme.

thesourcerer8, 2022-12-04:
2 of my submissions for GF180MPW#0 seem to be blocked by this issue, is there a way that I can fix that liberty file it in my local environment directly?

growly, 2022-12-04:
@kareefardi adding `(* keep *)` doesnt' workaround the problem, I get the same error about an problem in the liberty file

What version of OpenLane will include the fix?

gregdavill, 2022-12-04:
A workaround I've been using is to add `VERILOG_FILES_BLACKBOX` to your config.

eg in config.tcl
```
set ::env(VERILOG_FILES_BLACKBOX) "\
	$::env(DESIGN_DIR)/rtl/bb/gf180mcu_fd_sc_mcu7t5v0.v"
```

Populate the verilog file with blackbox module of the cells you're using like so, this should keep yosys happy.
```
(*blackbox*)
module gf180mcu_fd_sc_mcu7t5v0__buf_1( I, Z, VDD, VSS );
input I;
inout VDD, VSS;
output Z;
endmodule
```

You could also add this file under you standard verilog sources due to `(* blackbox *)` attribute.

kareefardi, 2022-12-04:
The current pdk and openlane versions should fix the problem. Let me know if it sill persists.

thesourcerer8, 2022-12-04:
I wasn't solved me yet, so I have manually replaced !RN with "!RN" and !SETN with "!SETN" in all the *.lib files in the gf180mcuC PDK, and that solved the issue.

thesourcerer8, 2022-12-04:
It seems that Liberty files can be parsed with just a alphanumeric names, but as soon as you use the ! operator you need "" around the expression.

kareefardi, 2022-12-04:
@thesourcerer8 what you are describing is the patch that got applied to the pdk. Are you sure you are using the latest version of the pdk?

thesourcerer8, 2022-12-04:
Ah, so I somehow didn't succeed to update the pdk properly.

closed 2022-12-28 by donn