The-OpenROAD-Project/OpenLane #1538: Source json files

marwaneltoukhy, 2022-11-30:
### Description

In the new `config.json` we need to have a way to source another `config.json`

mattvenn, 2022-12-12:
yes, I depend on this for Zero to ASIC group submissions and TinyTapeout

vijayank88, 2022-12-26:
In [MPW CI](https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI) in top level we've `config.tcl` and source caravel flow `config.tcl`. https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/blob/dfdfdb7052f0b24a0fffaebe08d12ee3517dbf69/scripts/AC_top/config.tcl#L4
In latest OpenLane/ MPW-8 shuttle forced to use `config.json`.
How to source `json` file through `tcl` file?

vijayank88, 2022-12-29:
@kareefardi 
Any workaround to source `json` file inside `tcl` file?

mattvenn, 2023-02-15:
@donn can we get an update on this please?

donn, 2023-02-15:
Not for a while- I'll elaborate in DM.

donn, 2024-09-12:
Somewhat addressed in https://github.com/efabless/openlane2/pull/351- you can now simply pass multiple JSON files to OpenLane and the config will be the composition of all of these.