The-OpenROAD-Project/OpenLane #1543: Crash during Global Routing Resizer Timing Optimizations

mole99, 2022-12-03:
### Description

I am currently trying to reharden a design for submission to GFMPW-0. While this previously worked well, now I get a crash during Global Routing Resizer Timing Optimizations. The design can be found under [https://github.com/mole99/caravel_wfg_gf180/tree/ol-issue](https://github.com/mole99/caravel_wfg_gf180/tree/ol-issue) in the branch `ol-issue`.

### Expected Behavior

To successfully finish the flow.

### Environment report

```shell
$ python3 ./env.py issue-survey
Kernel: Linux v6.0.0-5-amd64
Distribution: debian 
Python: v3.10.8 (OK)
Container Engine: docker v20.10.21+dfsg1 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10147356/issue_reproducible.tar.gz)

Command to run: `make user_project_wrapper`

### Relevant log output

```shell
[STEP 9]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: ../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/logs/routing/9-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: ../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/logs/routing/9-resizer.log
[ERROR]: Last 10 lines:
35# 0x00007F98D5761F1E in /lib64/libtcl8.5.so
36# Tcl_EvalEx in /lib64/libtcl8.5.so
37# Tcl_Eval in /lib64/libtcl8.5.so
38# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
39# ord::tclAppInit(Tcl_Interp*) in openroad
40# Tcl_Main in /lib64/libtcl8.5.so
41# main in openroad
42# __libc_start_main in /lib64/libc.so.6
43# 0x0000000000CABE77 in openroad
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/issue_reproducible…
[WRN] /home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/tmp/9-user_project_wrapper.sdc was not found, might be a product. Skipping
Done.
[INFO]: Reproducible packaged: Please tarball and upload '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/issue_reproducible' if you're going to submit an issue.
[INFO]: Saving current set of views in '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_03_21_14/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

maliberty, 2022-12-04:
This test case is not properly packaged:
Error: resizer_routing_timing.tcl, 15 cannot read file /home/leo/Dokumente/workspace_gf_mpw0/dependencies/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_5v00.lib.

maliberty, 2022-12-04:
I got it working and I'll fix the crash.  However your macro placement is outside the core area which you need to fix:
[image]

closed 2022-12-04 by maliberty