The-OpenROAD-Project/OpenLane #1546: Crash during Detailed Routing

mole99, 2022-12-04:
### Description

I am currently trying to reharden a design for submission to GFMPW-0. I get a crash during Detailed Routing.

First of all thanks @maliberty for taking care of #1543

The reason I had the coordinates of the macros outside of the boundary was that I had to double the actual value for the tag `gfmpw-0c` of [caravel_user_project](https://github.com/efabless/caravel_user_project). This seems to be fixed now with the new tag.

Therefore I set the coords back to their original value inside the core area.
I compiled OpenROAD with your patch and rerun the flow. (I hopefully did everything correct there.) Now I get a bit further in the flow but crash during Detailed Routing.

As before, the design can be found at [https://github.com/mole99/caravel_wfg_gf180/tree/ol-issue](https://github.com/mole99/caravel_wfg_gf180/tree/ol-issue) in the branch `ol-issue`.

### Expected Behavior

To successfully finish the flow.

### Environment report

```shell
$ python3 ./env.py issue-survey
Kernel: Linux v6.0.0-5-amd64
Distribution: debian 
Python: v3.10.8 (OK)
Container Engine: docker v20.10.21+dfsg1 (OK)
OpenLane Git Version: b2bd5f68d29aa4bde1e7f7394745568e09e73e54
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 35c7265f51749ad8d9fdbb575af22c7c8fab974e, tested: 141eea4d1bb8c6d4dd85fcbf2c0bdface7df9cfc)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

b2bd5f6 2022-12-03T15:08:33+02:00 [BOT] Update PDK (#1532) - Openlane Bot -  (grafted, HEAD -> master, tag: 2022.12.04, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10148249/issue_reproducible.tar.gz)

Command to run: `make user_project_wrapper`

### Relevant log output

```shell
[STEP 13]
[INFO]: Running Detailed Routing (log: ../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/logs/routing/13-detailed.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/droute.tcl
[ERROR]: Log: ../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/logs/routing/13-detailed.log
[ERROR]: Last 10 lines:
44# 0x00007FB7BE296F1E in /lib64/libtcl8.5.so
45# Tcl_EvalEx in /lib64/libtcl8.5.so
46# Tcl_Eval in /lib64/libtcl8.5.so
47# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
48# ord::tclAppInit(Tcl_Interp*) in openroad
49# Tcl_Main in /lib64/libtcl8.5.so
50# main in openroad
51# __libc_start_main in /lib64/libc.so.6
52# 0x0000000000CB9777 in openroad
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/issue_reproducible' if you're going to submit an issue.
[INFO]: Saving current set of views in '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../home/leo/Dokumente/workspace_gf_mpw0/caravel_wfg_gf180/openlane/user_project_wrapper/runs/22_12_04_11_04/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

maliberty, 2022-12-04:
@antonblanchard the crash in in ta.  Do you have any time to look at this?

maliberty, 2022-12-04:
I think the problem originates with your pin outside the block bounds:
[image]

maliberty, 2022-12-04:
There is also something wrong with the tracks.  The metal2 wrong-way tracks stop well before the die edge:
[image]

How were the tracks generated?

maliberty, 2022-12-04:
@eder-matheus there is a grt problem here.  grt is putting the guide where there are no m2 tracks:
[image]

maliberty, 2022-12-05:
I have an fix that will error out on this condition rather than crashing.  The underlying problem is upstream of detailed routing and needs a different test case showing how the track pattern was generated.

closed 2022-12-05 by maliberty