The-OpenROAD-Project/OpenLane #1547: ERROR GPL-0130

thesourcerer8, 2022-12-04:
### Description

During hardening my gf180mcu based design, I get the following error message:
[ERROR GPL-0130] No rows defined in design. Use initialize_floorplan to add rows.
Error: gpl.tcl, 69 GPL-0130

### Expected Behavior

floorplan should work

### Environment report

```shell
Kernel: Linux v5.15.0-54-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10149423/issue_reproducible.zip)

### Relevant log output

```shell
STEP 5]
[INFO]: Running Global Placement (log: ../home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_04_15_30/logs/placement/5-global.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: ../home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_04_15_30/logs/placement/5-global.log
[ERROR]: Last 10 lines:
OpenROAD 7f00621cb612fd94e15b35790afe744c89d433a7 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Setting RC values...
[INFO]: Setting signal min routing layer to: Metal2 and clock min routing layer to Metal2. 
[INFO]: Setting signal max routing layer to: Metal4 and clock max routing layer to Metal4. 
[ERROR GPL-0130] No rows defined in design. Use initialize_floorplan to add rows.
Error: gpl.tcl, 69 GPL-0130
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

kareefardi, 2022-12-04:
It there a specific reason for using the 9t version? I don't think its openlane configuration files are ready yet. I can prioritize it if there is a strong need

kareefardi, 2022-12-04:
And I can see there are errors in the pdk sources themselves

vijayank88, 2022-12-05:
@thesourcerer8 
Update following configuration for `9t` library and run the flow: 
```
set ::env(PLACE_SITE) "GF018hv5v_green_sc9"
```
This is an open_pdks setup issue. 
Currently `7t` library properly configured for GF180

kareefardi, 2024-07-17:
open_pdks have been updated long time ago

closed 2024-07-17 by kareefardi