The-OpenROAD-Project/OpenLane #1549: PDN can place vias in ways that break DRC

gregdavill, 2022-12-05:
### Description

I've included the SRAM IP block from GF180 in a design (gf180mcu_fd_ip_sram__sram256x8m8wm1). 
When running DRC on the hardened design it failed with 100x `V3.2a : min. via3 spacing : 0.26µm` errors.

Looking at the GDS I see that the PDN has routed a vertical strap directly over multiple overlapping Metal 3 geometries in the IP block. And in 2 places the via grid it tried to place was offset at 1/2 so the vias did not overlap. 
But they are too close according to DRC.

[image]
[image]

The design is here: `serv_0` this commit is the one that shows the issue: https://github.com/gregdavill/gf180-mpw0-serv/blob/91ddd78fc81e904b4ff1f63cd234433312043c27/gds/serv_0.gds

### Expected Behavior

PDN places vias according to DRC.

### Environment report

```shell
Kernel: Linux v5.19.0-23-generic
Distribution: ubuntu 22.10
Python: v3.10.7 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[gf180-mpw0-serv-91ddd78fc81e904b4ff1f63cd234433312043c27.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10151954/gf180-mpw0-serv-91ddd78fc81e904b4ff1f63cd234433312043c27.zip)

Build instructions:
```
git submodule update --init
make serv_0
```

### Relevant log output

```shell
[STEP 7]
[INFO]: Generating PDN (log: ../home/runner/work/gf180-mpw0-serv/gf180-mpw0-serv/openlane/serv_0/runs/22_12_04_22_51/logs/floorplan/7-pdn.log)...
[INFO]: Executing openroad with Tcl script 'scripts/openroad/pdn.tcl'...
OpenROAD 7c85c140308f01b73f57ea1117f3e43f39abd437 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: stdcell_grid
[INFO PDN-0001] Inserting grid: macro - u_cpu.rf_ram.RAM0
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (82.2400, 411.8850) - (83.0400, 416.8900) on vdd
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (127.2400, 80.0000) - (128.2450, 511.8600) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (127.2400, 232.0150) - (128.2500, 353.1100) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (127.2400, 382.3550) - (128.5000, 511.8600) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (217.2400, 80.0000) - (218.2450, 511.8600) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (217.2400, 232.0150) - (218.2500, 353.1100) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (217.2400, 382.3550) - (218.5000, 511.8600) on vss
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (262.2400, 213.8600) - (263.4500, 371.7550) on vdd
[WARNING PDN-0110] No via inserted between Metal3 and Metal4 at (262.2400, 369.5400) - (263.4500, 371.7550) on vdd
[... 31 lines cut ...]
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[WARNING PSM-0030] VSRC location at (420.000um, 229.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (413.040um, 230.210um).
[INFO PSM-0031] Number of PDN nodes on net vdd = 6074.
[INFO PSM-0064] Number of voltage sources = 3.
[INFO PSM-0040] All PDN stripes on net vdd are connected.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net vss is not explicitly set.
[WARNING PSM-0021] Using voltage 0.000V for ground network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[WARNING PSM-0030] VSRC location at (140.000um, 89.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (127.870um, 88.660um).
[WARNING PSM-0030] VSRC location at (420.000um, 229.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (458.040um, 227.360um).
[WARNING PSM-0030] VSRC location at (140.000um, 509.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (128.040um, 510.565um).
[INFO PSM-0031] Number of PDN nodes on net vss = 6088.
[INFO PSM-0064] Number of voltage sources = 3.
[INFO PSM-0040] All PDN stripes on net vss are connected.
Setting global connections for newly added cells...
Writing OpenROAD database to /home/runner/work/gf180-mpw0-serv/gf180-mpw0-serv/openlane/serv_0/runs/22_12_04_22_51/tmp/floorplan/7-pdn.odb...
Writing layout to /home/runner/work/gf180-mpw0-serv/gf180-mpw0-serv/openlane/serv_0/runs/22_12_04_22_51/tmp/floorplan/7-pdn.def...
[INFO]: Changing layout to '../home/runner/work/gf180-mpw0-serv/gf180-mpw0-serv/openlane/serv_0/runs/22_12_04_22_51/tmp/floorplan/7-pdn.def'...
[INFO]: Changing database to '../home/runner/work/gf180-mpw0-serv/gf180-mpw0-serv/openlane/serv_0/runs/22_12_04_22_51/tmp/floorplan/7-pdn.odb'...
```

vijayank88, 2022-12-05:
Have you tried `GRT_OBS` to block layers that route SRAM macros?

gregdavill, 2022-12-05:
Sorry about that. I used the zip from github, since it was built in CI. 
And they don't seem to include submodule contents. Here is one with it attached. 

[gf180-mpw0-serv-91ddd78fc81e904b4ff1f63cd234433312043c27.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10152959/gf180-mpw0-serv-91ddd78fc81e904b4ff1f63cd234433312043c27.zip)

gregdavill, 2022-12-05:
I've not tried that. I don't think the issue is related to layers in the macro. 

The PDN step has routed on metal 4 and placed both sets of vias down to layers on metal 3.
The SRAM macro should probably be placed into it's own hardened block to give more control around the power routing, but thought this was still worth reporting.

vijayank88, 2022-12-05:
@maliberty fyi.,

If this need to debug further mean attached PDN stage test case for the same. 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10154066/issue_reproducible.zip)

maliberty, 2022-12-05:
PDN only sees the LEF abstract.  Is the area in question blocked in your LEF file?  You can open the design in the OR gui to visualize it.

@gadfort FYI

kareefardi, 2023-01-24:
@maliberty I think this is happening due to overlapping shapes in the lef view of the sram similar to: 
[image]
In theory, it might be possible to re-write an abstract lef view using magic with shapes merged, and added to the pdk. However, might it be possible as well for openroad to handle such shapes?

maliberty, 2023-01-24:
@gadfort any comments?

gadfort, 2023-01-24:
I see the issue, I don't think it matters if the LEF if built right or wrong, PDNGEN should also handle this correctly. I'll try to see if I can implement something soon.

gadfort, 2023-01-25:
This should fix the issue (atleast in the testcase provided) https://github.com/The-OpenROAD-Project/OpenROAD/pull/2814

closed 2023-01-25 by maliberty