The-OpenROAD-Project/OpenLane #1552: Error: resizer.tcl, 15 cannot read file

thesourcerer8, 2022-12-05:
### Description

[STEP 10]
[INFO]: Running Placement Resizer Design Optimizations (log: ../home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_05_08_43/logs/placement/10-resizer.log)...
Error: resizer.tcl, 15 cannot read file /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_05_08_43/tmp/synthesis/resizer_gf180mcu_fd_sc_mcu9t5v0__tt_025C_3v30.lib /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_05_08_43/tmp/synthesis/resizer_opt_gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.lib.
child process exited abnormally

### Expected Behavior

liberty files should be parsed properly.
Both filenames referenced in the error log are readable and seem to be proper liberty files on first sight.

### Environment report

```shell
Kernel: Linux v5.15.0-54-generic
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.12 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10154813/issue_reproducible.zip)

### Relevant log output

```shell
OpenROAD 7f00621cb612fd94e15b35790afe744c89d433a7
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: resizer.tcl, 15 cannot read file /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_05_08_43/tmp/synthesis/resizer_gf180mcu_fd_sc_mcu9t5v0__tt_025C_3v30.lib /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/22_12_05_08_43/tmp/synthesis/resizer_opt_gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.lib.
```

thesourcerer8, 2022-12-05:
Perhaps the problem is that there are 2 filenames instead of one?

kareefardi, 2022-12-05:
Yes somehow `RSZ_LIB` has two files in it. I am checking how that might have happened. Meanwhile I noticed that you are using an older version of openlane. Mind trying that with a more recent version

thesourcerer8, 2022-12-05:
diff --git a/scripts/tcl_commands/all.tcl b/scripts/tcl_commands/all.tcl
index 6072e67..1a865e5 100755
--- a/scripts/tcl_commands/all.tcl
+++ b/scripts/tcl_commands/all.tcl
@@ -762,7 +762,7 @@ proc prep {args} {
                     set fbasename [file rootname [file tail $lib]]
                     set lib_resizer $::env(synthesis_tmpfiles)/resizer_opt_$fbasename.lib
                     file copy -force $lib $lib_resizer
-                    lappend ::env(RSZ_LIB) $lib_resizer
+                    #lappend ::env(RSZ_LIB) $lib_resizer
                 }
             }
         }

thesourcerer8, 2022-12-06:
The reason is there were 2 different standard cell libraries used:
[INFO]: Standard Cell Library: gf180mcu_fd_sc_mcu9t5v0
[INFO]: Optimization Standard Cell Library: gf180mcu_fd_sc_mcu7t5v0
I had not set the   "STD_CELL_LIBRARY_OPT" variable, but now that I have set it to  "gf180mcu_fd_sc_mcu9t5v0" too, the problem seems to be gone.
Another workaround I had developed was the following patch, but there was likely a good reason for this code:
diff --git a/scripts/tcl_commands/all.tcl b/scripts/tcl_commands/all.tcl
index 6072e67..1a865e5 100755
--- a/scripts/tcl_commands/all.tcl
+++ b/scripts/tcl_commands/all.tcl
@@ -762,7 +762,7 @@ proc prep {args} {
                     set fbasename [file rootname [file tail $lib]]
                     set lib_resizer $::env(synthesis_tmpfiles)/resizer_opt_$fbasename.lib
                     file copy -force $lib $lib_resizer
-                    lappend ::env(RSZ_LIB) $lib_resizer
+                    #lappend ::env(RSZ_LIB) $lib_resizer
                 }
             }
         }
~

donn, 2022-12-07:
The solution to failing multiple-file support is to properly support multiple files- so yeah, we'll take a look

kareefardi, 2023-01-24:
Should be fixed by #1570

closed 2023-01-24 by kareefardi

InputBlackBoxOutput, 2023-05-08:
Hi, 

I ran into a similar issue while using OpenLane via Google Colab with modifications to use a different liberty file. The issue was fixed when I set both STD_CELL_LIBRARY & STD_CELL_LIBRARY_OPT in the configuration file 

The header from the logs for the version information
[image]

Thanks