The-OpenROAD-Project/OpenLane #1553: Violations after detailed routing.

Baungarten-CINVESTAV, 2022-12-05:
### Description

Hi, I am trying to harden a small eFPGA (4 x 4 CLB), I have used OpenFPGA to create the FPGA architecture.
This FPGA has basic modules like CLB, switching blocks, and connection blocks. To harden the whole FPGA, I first harden each module separately and then try to harden the whole FPGA but get the following error.

[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total number of violations is 267.

any suggestions on how to fix this error?

### Expected Behavior

[image]

### Environment report

```shell
Kernel: Linux v5.15.0-53-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 7bb988c82671d48395012f13e535c6e7bd5e844c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

7bb988c 2022-12-01T21:31:26+00:00 [BOT] Update magic (#1444) - Openlane Bot -  (HEAD -> master, origin/master, origin/HEAD)
95c7cd1 2022-12-01T17:23:19+02:00 magic updates (#1540) - Kareem Farid -  ()
c98a290 2022-12-01T14:22:16+02:00 Fix variables for setting pin thickness and multiplier (#1539) - Kareem Farid -  ()
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[fpga_top.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10158154/fpga_top.zip)

### Relevant log output

```shell
[STEP 17]
[INFO]: Running Detailed Routing (log: designs/fpga_top/runs/RUN_2022.12.05_19.54.23/logs/routing/17-detailed.log)...
[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total Number of violations is 267
[INFO]: Saving current set of views in 'designs/fpga_top/runs/RUN_2022.12.05_19.54.23/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/fpga_top/runs/RUN_2022.12.05_19.54.23/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/fpga_top/runs/RUN_2022.12.05_19.54.23/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2022-12-06:
@Baungarten-CINVESTAV 
Can you follow steps from here: https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually
and produce test case attach?

vijayank88, 2022-12-06:
Other way you can read `runs/runtag/reports/routing/drt.drc` and check what kind of drc errors.
I ran locally your design and most of them related to  shorts and metal spacing etc.,

Baungarten-CINVESTAV, 2022-12-07:
Hello,
Sorry, I had some problems running OpenROAD. 

I can't run OpenROAD with OpenLANE, so I installed 
OpenROAD-flow-scripts and then run the GUI with the command `openroad -gui `
[image]

but I don't know how to load the .drc file, could you help me with the coman line to load that file.

maliberty, 2022-12-07:
Bring up the DRC Viewer from 
[image]
The click the "Load..." button and select your file.

maliberty, 2022-12-07:
You'll need to load the design first

Baungarten-CINVESTAV, 2022-12-07:
OpenROAD closes after loading my design.

The final message is "Segmentation fault (core dumped)"

Could it be a problem with OpenROAD or the installation?

baungarten@baungarten-Z690-AORUS-ELITE-AX-DDR4:~/Desktop/OpenROAD-flow-scripts$ openroad -gui
OpenROAD v2.0-5852-g2a9ef5fa1 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Signal 11 received
Stack trace:
 0# 0x0000556852B38038 in openroad
 1# 0x00007F4556E12090 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBlock::getDataBase() in openroad
 3# gui::DRCWidget::loadTRReport(QString const&) in openroad
 4# gui::DRCWidget::loadReport(QString const&) in openroad
 5# gui::DRCWidget::selectReport() in openroad
 6# 0x0000556853372658 in openroad
 7# QMetaObject::activate(QObject*, int, int, void**) in /lib/x86_64-linux-gnu/libQt5Core.so.5
 8# 0x00007F45585F8BD1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
 9# 0x00007F45585F9E5F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
11# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
12# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
13# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
15# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x00007F455856235D in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# 0x00007F45585651EC in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
21# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
22# QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
23# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
24# 0x00007F4552AC632E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
25# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
26# 0x00007F4556561400 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
27# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
28# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
29# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
31# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
32# ord::tclAppInit(Tcl_Interp*) in openroad
33# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
34# main in openroad
35# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
36# _start in openroad
Segmentation fault (core dumped)

maliberty, 2022-12-07:
How did you load the design?

Baungarten-CINVESTAV, 2022-12-07:
with the load button

[image]

maliberty, 2022-12-07:
That loads the drc report not the design.

Baungarten-CINVESTAV, 2022-12-07:
ok, but how to load the design?

Sorry I am not familiar with OpenROAD

maliberty, 2022-12-07:
If you are in OL you might just try running flow.tcl as usual with the added flag -gui and see if that works for you.

Baungarten-CINVESTAV, 2022-12-07:
`./flow.tcl -design fpga_top -gui`

The following message apears:

OpenLane Container (7bb988c):/openlane$ ./flow.tcl -design fpga_top -gui
OpenLane 7bb988c82671d48395012f13e535c6e7bd5e844c
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/fpga_top/config.json'...
[INFO]: PDK Root: /home/baungarten/Test_env/OpenLane/pdks
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/fpga_top/runs/RUN_2022.12.07_20.32.04
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[ERROR]: during executing openroad script /openlane/scripts/openroad/gui.tcl
[ERROR]: Log: ../dev/null
[ERROR]: Last 10 lines:
13# 0x00007F1D9EA443F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F1DA6B493B5 in /lib64/libQt5DBus.so.5
19# 0x00007F1DA2E65E71 in /lib64/libQt5Core.so.5
20# 0x00007F1DA65CAEA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

[ERR] 0 not found.
[ERROR]: Failed to package reproducible.
[INFO]: Saving current set of views in 'designs/fpga_top/runs/RUN_2022.12.07_20.32.04/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/fpga_top/runs/RUN_2022.12.07_20.32.04/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/fpga_top/runs/RUN_2022.12.07_20.32.04/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

maliberty, 2022-12-07:
We should fix that.  Can you attach the reproducer that it generated.

Baungarten-CINVESTAV, 2022-12-07:
No reproducible files

at the end of the flow it says "[ERROR]: Failed to package reproducible".

I attach the execution folder.
[RUN_2022.12.07_20.52.03.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10179952/RUN_2022.12.07_20.52.03.zip)

do you want me to create new issues?

maliberty, 2022-12-07:
You should open an OL issue for  "[ERROR]: Failed to package reproducible".

I'll see if I can work with your zip

maliberty, 2022-12-09:
The zip is insufficient as it has no design files, just a few tcl files.  I can't reproduce the problem from this.

vijayank88, 2022-12-09:
@Baungarten-CINVESTAV 
I shared you drc error and type. What exactly you're looking now?
Are you looking for help to load DEF file and drc report to locate your DRC errors?

Baungarten-CINVESTAV, 2022-12-09:
HI,
Yes, you told me where the errors are but I want to know how to load the drc files. I'm going to be working more with OpenLANE so learning how to do that will be really helpful.

maliberty, 2022-12-09:
You need to provide a complete test case so I can reproduce your problem.

Baungarten-CINVESTAV, 2022-12-09:
ok, I will open the new issues called "[ERROR]: Failed to package reproducible" and when I have the reproducible package I will attach the file here.

vijayank88, 2022-12-10:
@maliberty Attached detail route reproducible
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10199886/issue_reproducible.zip)

maliberty, 2022-12-10:
You have completely unaccessible pins like sb_0__0_/right_top_grid_bottom_width_0_height_0_subtile_0__pin_O_6_
[image]

It is block from above, below, and all sides by OBS.  The router can only access it by making a short.

maliberty, 2022-12-10:
You can fix this by extending the pin shape to the macro edge

maliberty, 2022-12-10:
You also have no power stripes for your standard cells so they are unpowered:
[image]

I'm not sure why you aren't using met5 as a routing layer.

Baungarten-CINVESTAV, 2022-12-14:
Hello, sorry for the late response, I was following your answers.

I increase the size and change the position of some blocks (SB blocks and CBX blocks), whit those changes the detailed routing step was successful.

Now, I have some problems with the LVS verification (netlists do not match), when I check the logs files the main mismatch happens with sky130_fd_sc_hd__tapvpwrvgnd_1 and sky130_fd_sc_hd__decap_3.

- No issue reproducible file was generated, I attach the design folder.
[fpga_top.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10231143/fpga_top.zip)

**Expected Behavior**
[image]

Moreover, when I eliminate the met4 restriction the following error appears.
 
[image]

To solve this, I need to harden each submodule again without met4 restriction? 
I attach the issue reproducible for this error.

Baungarten-CINVESTAV, 2022-12-14:
[issue_reproducible_met4_error.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10231176/issue_reproducible_met4_error.zip)

maliberty, 2022-12-14:
tap & decap cells should have nothing to do with LVS.  They are physical only instances

maliberty, 2022-12-14:
Apparently you haven't removed the m5 restriction according to the message:
[image]

Baungarten-CINVESTAV, 2022-12-16:
Hello,

I am now working with met5 I hardened each module again without met 5 restriction, but the LVS error still happens. I mentioned tap & decap because the 33-fpga_top.lef.lvs.log says things like: 

"Net: PHY_2035/VPB sky130_fd_sc_hd__decap_3/VPB = 1 | (no matching net)".                          

Any idea how to fix this LVS error?

[33-fpga_top.lef.lvs.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10247846/33-fpga_top.lef.lvs.log)
[33-lvs.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/10247847/33-lvs.lef.log)

[image]

Thank you in advance

maliberty, 2022-12-17:
As this is now an LVS problem and not an OR one the OL team should take over @shalan

donn, 2022-12-28:
Kindly open a new issue for the new problem.

closed 2022-12-28 by donn