The-OpenROAD-Project/OpenLane #1556: [ERROR]: Failed to compare PDKs.

msaideroglu, 2022-12-07:
### Description

When I setup the environment spm test successfully completes, however when I run "./flow.tcl" on the terminal, it gives this error:
[image]

### Expected Behavior

"./flow.tcl" runs without error

### Environment report

```shell
Kernel: Linux v5.15.0-56-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: b8c6129fb60851c452a3136c2b8c603bb92cb180, tested: 35c7265f51749ad8d9fdbb575af22c7c8fab974e)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

./flow.tcl

### Relevant log output

```shell
OpenLane 235fa7a4a2872e779588919c58fc4fa32568e075
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[ERROR]: Failed to compare PDKs.
'NoneType' object has no attribute 'startswith'
Traceback (most recent call last):
  File "/home/msaid/OpenLane/dependencies/verify_versions.py", line 68, in verify_versions
    if pdk.startswith("sky130"):
AttributeError: 'NoneType' object has no attribute 'startswith'

[ERROR]: Please update your environment. OpenLane will now quit.
```

msaideroglu, 2022-12-07:
Hi again. I'm facing another issue now. When I clone latest version of Openlane and run make test command, it successfully runs with Magic's recent version of 8.3.348. However when I run make mount on terminal, it clones additional docker images:
[image]
and I leave make mount environment and try to re-run make test command on terminal, it gives this error:
[image]

vijayank88, 2022-12-07:
```
python3 ./env.py issue-survey
```
Provide environment report to proceed further

msaideroglu, 2022-12-07:
Kernel: Linux v5.15.0-56-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: b8c6129fb60851c452a3136c2b8c603bb92cb180, tested: 35c7265f51749ad8d9fdbb575af22c7c8fab974e)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)

vijayank88, 2022-12-07:
Update your OpenLane master with following commands and re-try:
```
git checkout master
git pull
make
``

msaideroglu, 2022-12-07:
~/OpenLane$ git checkout master
error: pathspec 'master' did not match any file(s) known to git

vijayank88, 2022-12-07:
Delete complete OpenLane directory or install in new directory location and follow steps from here: https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html#download-and-install-openlane

msaideroglu, 2022-12-07:
I installed OpenLane. OpenLane's the last commit installs with Magic's recent version of 8.3.348 but when make mount runned it converts Magic's version to 8.3.331. Then my design flow is giving this error:
[image]
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10176259/issue_reproducible.zip)

vijayank88, 2022-12-07:
Its mean you want to update Magic version to latest and want to try right? 
Is that got fixed with upstream magic? or its new issue you're facing with your design?

msaideroglu, 2022-12-07:
Yes it is fixed with upstream Magic and OpenLane did an update for it. However when I install OpenLane from master branch Magic's recent version comes. However when I run make mount, OpenLane dos additional docker image clones. Magic's version changes. I think there are disagreements between the last commit and the previous ones. You can see same issue in [https://github.com/The-OpenROAD-Project/OpenLane/issues/1531](url)

donn, 2022-12-07:
This is a different issue and your issue form is invalid- please re-file.

closed 2022-12-07 by donn