The-OpenROAD-Project/OpenLane #1557: Magic segmentation violation

msaideroglu, 2022-12-07:
### Description

I installed OpenLane. OpenLane's the last commit installs with Magic's recent version of 8.3.348 but when make mount runned it converts Magic's version to 8.3.331. Then my design flow is giving this error:
[image]
It is fixed with upstream Magic as mentioned in Issue #1531 and OpenLane did an update for it. However when I install OpenLane from master branch Magic's recent version comes. However when I run make mount, OpenLane does additional docker image clones. Magic's version changes. I think there are disagreements between the last commit and the previous ones. 
@antonblanchard @vijayank88 @donn

### Expected Behavior

Magic's version to be updated

### Environment report

```shell
Kernel: Linux v5.15.0-56-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10176696/issue_reproducible.zip)

### Relevant log output

```shell
child killed: segmentation violation
```

donn, 2022-12-07:
1. I will ask you to please refrain from @'ing OpenLane team members. After filing an issue, I or one of my colleagues will tag and attend to the issue.

2.

> However when I run make mount, OpenLane does additional docker image clones. Magic's version changes. I think there are disagreements between the last commit and the previous ones.

The environment survey shows that you are on commit `235fa7a`. So naturally, `make mount` will pull the corresponding OpenLane version. 

> However when I install OpenLane from master branch Magic's recent version comes

This statement is ambiguous as to how you're "installing" OpenLane. If you want to upgrade an existing installation of OpenLane, follow the guide here: https://openlane.readthedocs.io/en/latest/getting_started/updating.html

msaideroglu, 2022-12-08:
Okay. I will look into it. Thanks.

closed 2022-12-08 by msaideroglu