The-OpenROAD-Project/OpenLane #1559: Small designs need PL_BASIC_PLACEMENT

mattvenn, 2022-12-08:
### Description

Referring to the discussion here:

https://github.com/The-OpenROAD-Project/OpenLane/issues/1183

Many small designs need PL_BASIC_PLACEMENT, or they fail at global placement. However setting it to one is a big performance detrement to almost all other designs.

Here's a list of designs that need the option:

https://github.com/TinyTapeout/tt02-test-invert
https://github.com/argunda/tt02-TinyPFD
https://github.com/13arn/tt02_counter_steamdeck
https://github.com/NYIT-CNS/cns001-tt02-submission1
https://github.com/shaos/tt02-submission-shaos
https://github.com/mole99/tt02-1bit-alu
https://github.com/jglim/tt02-bcd-7seg
https://github.com/mmolteni-secpat/tinytapeout02_chi3shares
https://github.com/aiunderstand/tt02-4bit-tristate-loadable-counter
https://github.com/ctag/tt02-submission-ctag
https://github.com/AsmaMohsin1507/tt02-channel-coding
https://github.com/ternary-info/tt02-submission-shaos3
https://github.com/saicharan0112/tt02-submission-template
https://github.com/theFestest/tt02-4x4-ram
https://github.com/jeanthom/tinytapout-lock
https://github.com/cmu-stuco-98154/f22-tt02-qilins
https://github.com/prabaldutta/tt02-adi-demo
https://github.com/TinyTapeout/tt02-test-7seg

Here's a test case I prepared to make it easy to check:

https://github.com/mattvenn/openlane-test-invert

### Expected Behavior

should work fine. 

### Environment report

```shell
Kernel: Linux v5.13.0-28-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3af133706e554a740cfe60f21e773d9eaa41838c)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

download the repo, mount openlane, then run

./flow.tcl -ignore_mismatches -design test-invert

### Relevant log output

```shell
[STEP 7]
[INFO]: Running Global Placement (log: designs/test-invert/runs/RUN_2022.12.08_18.06.42/logs/placement/7-global.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: designs/test-invert/runs/RUN_2022.12.08_18.06.42/logs/placement/7-global.log
[ERROR]: Last 10 lines:
[INFO GPL-0068] 2.0%RC: 0.5719298453707444
[INFO GPL-0069] 5.0%RC: 0.4985507350900899
[INFO GPL-0070] 0.5rcK: 1.0
[INFO GPL-0071] 1.0rcK: 1.0
[INFO GPL-0072] 2.0rcK: 0.0
[INFO GPL-0073] 5.0rcK: 0.0
[INFO GPL-0074] FinalRC: 0.6666667
[ERROR GPL-0306] RePlAce diverged at wire/density gradient Sum.
Error: gpl.tcl, 69 GPL-0306
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

maliberty, 2022-12-08:
In the future please note:
```
[INFO]: Reproducible packaged: Please tarball and upload 'designs/openlane-test-invert/runs/RUN_2022.12.08_18.52.08/issue_reproducible' if you're going to submit an issue.
```
providing that issue_reproducible is the best way to make a test case that is independent of OL.  I've done that for myself in this case.

mattvenn, 2022-12-08:
Apologies

maliberty, 2022-12-08:
np

maliberty, 2022-12-08:
I've fixed this case.  If there are others that still fail please reopen with a new case.

closed 2022-12-08 by maliberty