The-OpenROAD-Project/OpenLane #1562: DRC error inside macro area

smit27signoff, 2022-12-09:
### Description

Hello 

The run fails DRC with 11326640 errors. It is showing violations in the design after magic DRC.
Please find attached the drc.tr report and config.tcl for the same.

### Expected Behavior

Flow completes without drc error.

### Environment report

```shell
[smitch.p@signoff1 OpenLane]$ python3 ./env.py issue-survey
Kernel: Linux v3.10.0-1160.76.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.18 (OK)
OpenLane Git Version: 41ed0036c3dc77961a34005fc7e4494881f0a07f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

41ed003 %cI Fix get_design_path (#1450) - Liu Yihua -  (%D)
63a6d02 %cI Fix Synthesis Exploration (#1437) - vijayan -  (%D)
9197534 %cI fix usage command in docs (#1449) - Liu Yihua -  (%D)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[image]

Here is the link for drc.tr report
https://1drv.ms/u/s!ArS_MaFFW4KqgRfa1TO0PBQ7FWxL?e=eeJ0BN

### Relevant log output

```shell
[INFO]: Run Directory: /openlane/designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running Synthesis (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/synthesis/1-synthesis.log)...
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/synthesis/2-sta.log)...
[INFO]: Creating a netlist with power/ground pins.
[INFO]: Running Initial Floorplanning (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 2908.58 and height 3497.92.
[INFO]: Running IO Placement (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/floorplan/4-place_io.log)...
[INFO]: Performing Manual Macro Placement (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/placement/5-macro_placement.log)...
[INFO]: Running Tap/Decap Insertion (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/floorplan/6-tap.log)...
[INFO]: Power planning with power {vccd1 vccd2 vdda1 vdda2} and ground {vssd1 vssd2 vssa1 vssa2}...
[INFO]: Generating PDN (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/floorplan/7-pdn.log)...
[INFO]: Running Global Placement (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/placement/8-global.log)...
[INFO]: Running Placement Resizer Design Optimizations (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/placement/9-resizer.log)...
[INFO]: Running Detailed Placement (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/placement/10-detailed.log)...
[INFO]: Running Clock Tree Synthesis (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/cts/11-cts.log)...
[INFO]: Running Placement Resizer Timing Optimizations (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/cts/12-resizer.log)...
[INFO]: Running Global Routing Resizer Timing Optimizations (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/13-resizer.log)...
[INFO]: Running Detailed Placement (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/14-diode_legalization.log)...
[INFO]: Running Global Routing (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/15-global.log)...
[INFO]: Starting OpenROAD Antenna Repair Iterations...
[INFO]: Starting antenna repair iteration 1 with 146 violations...
[INFO]: [Iteration 1] Failed to reduce antenna violations (146 -> 146), stopping iterations...
[INFO]: Writing Verilog (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/15-global_write_netlist.log)...
[INFO]: Running Fill Insertion (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/17-fill.log)...
[INFO]: Running Detailed Routing (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/routing/18-detailed.log)...
[INFO]: No DRC violations after detailed routing.
[INFO]: Running SPEF Extraction at the min process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/19-parasitics_extraction.min.log)...
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/20-rcx_mcsta.min.log)...
[INFO]: Running SPEF Extraction at the max process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/21-parasitics_extraction.max.log)...
[INFO]: Running Multi-Corner Static Timing Analysis at the max process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/22-rcx_mcsta.max.log)...
[INFO]: Running SPEF Extraction at the nom process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/23-parasitics_extraction.nom.log)...
[INFO]: Running Multi-Corner Static Timing Analysis at the nom process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/24-rcx_mcsta.nom.log)...
[INFO]: Running Single-Corner Static Timing Analysis at the nom process corner (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/25-rcx_sta.log)...
[INFO]: Creating IR Drop Report (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/26-irdrop.log)...
[INFO]: Running Magic to generate various views...
[INFO]: Streaming out GDSII with Magic (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/27-gdsii.log)...
[INFO]: Generating MAGLEF views...
[INFO]: Streaming out GDSII with KLayout (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/28-gdsii-klayout.log)...
[INFO]: Running XOR on the layouts using KLayout (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/29-xor.log)...
[INFO]: Running Magic Spice Export from LEF (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/30-spice.log)...
[INFO]: Writing Powered Verilog (logs: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/31-write_powered_def.log, designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/31-write_powered_verilog.log)...
[INFO]: Writing Verilog (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/31-write_powered_verilog.log)...
[INFO]: Running LVS (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/33-lvs.lef.log)...
[INFO]: Running Magic DRC (log: designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/logs/signoff/34-drc.log)...
[INFO]: Converting Magic DRC database to various tool-readable formats...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 11326640
[INFO]: Saving current set of views in 'designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/Tunga_SP/runs/RUN10_MACRO_ABS_MACRO_CHANGE_vdda1_vssd1_new/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

smit27signoff, 2022-12-15:
@d-m-bailey Thanks for the suggestion, but I am getting same error as above with *.maglef.

kareefardi, 2023-01-24:
@smit27signoff any chance to get the design files to reproduce the problem?