The-OpenROAD-Project/OpenLane #1563: [ERROR]: Failed to package reproducible

Baungarten-CINVESTAV, 2022-12-09:
### Description

Hello 
I want to use the "-gui" command to see the "drc" errors. The drc problem is under review in #1553 . The current problem is because no matter what design I am running when I added "-gui" an error occurs and the reproducible package is not created. The original error cannot be solved because there is no reproducible material.

To reproduce the problem:
1. Clone and make the latest version of openlane
```
cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
make
```
2. Run OpenLane environment
`make mount`

3. Execute any design by adding the -gui flag.

e.g. `./flow.tcl -design inverter -gui
`

### Expected Behavior

OpenLane Container (73bc7f8):/openlane$ ./flow.tcl -design inverter -gui
OpenLane 73bc7f8a736d6f2a2d68168daea3e7718d4b6208
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/inverter/config.json'...
[INFO]: PDK Root: /home/baungarten/OpenLane/pdks
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/inverter/runs/RUN_2022.12.09_17.55.51
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[ERROR]: during executing openroad script /openlane/scripts/openroad/gui.tcl
[ERROR]: Log: ../dev/null
[ERROR]: Last 10 lines:
13# 0x00007FC47F4D73F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007FC4875DC3B5 in /lib64/libQt5DBus.so.5
19# 0x00007FC4838F8E71 in /lib64/libQt5Core.so.5
20# 0x00007FC48705DEA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
child killed: SIGABRT

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

[ERR] 0 not found.
[ERROR]: Failed to package reproducible.
[INFO]: Saving current set of views in 'designs/inverter/runs/RUN_2022.12.09_17.55.51/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/inverter/runs/RUN_2022.12.09_17.55.51/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/inverter/runs/RUN_2022.12.09_17.55.51/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

### Environment report

```shell
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-56-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 73bc7f8a736d6f2a2d68168daea3e7718d4b6208
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

73bc7f8 2022-12-09T13:20:53+02:00 Add new community guide for using RgGen (#1561) - Cra2yPierr0t -  (HEAD -> master, origin/master, origin/HEAD)
772a224 2022-12-05T15:31:29+02:00 [BOT] Update PDK (#1551) - Openlane Bot -  (tag: 2022.12.06)
010f5ad 2022-12-04T12:20:40+02:00 Add PDK_FAMILY for `volare enable` (#1545) - Kareem Farid -  (tag: 2022.12.05)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

- No reproducible material

### Relevant log output

```shell
[ERR] 0 not found.
[ERROR]: Failed to package reproducible.
[INFO]: Saving current set of views in 'designs/inverter/runs/RUN_2022.12.09_17.55.51/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/inverter/runs/RUN_2022.12.09_17.55.51/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/inverter/runs/RUN_2022.12.09_17.55.51/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

kareefardi, 2023-01-23:
This should print out an error now:
```
[ERROR]: CURRENT_ODB is unset.
```

kareefardi, 2023-01-24:
Exception handled in #1625

closed 2023-01-24 by kareefardi