The-OpenROAD-Project/OpenLane #1564: Example CoLab notebook in documentation is missing dependency

JarrettR, 2022-12-09:
### Description

In this link, from the READMD.md:

https://colab.research.google.com/github/chipsalliance/silicon-notebooks/blob/main/digital-inverter-openlane.ipynb

Step number 11 fails with:

```
ModuleNotFoundError                       Traceback (most recent call last)
[<ipython-input-11-f3322b103f89>](https://localhost:8080/#) in <module>
      1 import pathlib
----> 2 import gdstk
      3 import IPython.display
      4 
      5 gdss = sorted(pathlib.Path('runs').glob('*/results/final/gds/*.gds'))

ModuleNotFoundError: No module named 'gdstk'
```

### Expected Behavior

It runs successfully.

Presumably, `gdstk` just needs to be installed in the first step

### Environment report

```shell
Running on Windows 10 in Chrome
```

### Reproduction material

.

### Relevant log output

```shell
.
```

vijayank88, 2022-12-10:
@proppy 
Can you please look into this issue?

proppy, 2022-12-10:
See https://github.com/chipsalliance/silicon-notebooks/issues/30

donn, 2023-04-10:
Fixed.

closed 2023-04-10 by donn