The-OpenROAD-Project/OpenLane #1565: Need a safe way to disable power connections for a inner macro

growly, 2022-12-10:
### Description

I have an unpowered macro in an otherwise-powered design. The standard OpenLane flow fails at the "Writing Powered Verilog" step because the sub-macro fails with

```
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/power_utils.py write_powered_def --output /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/tmp/signoff/24-bfg_mux_test_small.p.def --input-lef /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/tmp/merged.nom.lef --power-port vdd --ground-port vss --powered-netlist {} /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/results/routing/bfg_mux_test_small.def |& tee /dev/null /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/logs/signoff/25-write_powered_def.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Top-level design name: bfg_mux_test_small
Found default power net 'vdd'
Found default ground net 'vss'
Found 1 power ports.
Found 1 ground ports.
[WARN] No pins in the LEF view of bfg_mux marked for use as power.
[WARN] Attempting to match power pin by name (using top-level port name) for bfg_mux.
[WARN] No pins in the LEF view of bfg_mux marked for use as ground.
[WARN] Attempting to match power pin by name (using top-level port name) for bfg_mux.
child process exited abnormally
```

I discovered that I can turn this off by manually adding `--ignore-missing-pins` to the `power_utils.py` script invocation in `lvs.tcl`, but I worry that this turns off checking for the entire hierarchy. My top-level macro is a standard digital macro with standard cells and such.

```
index 5b17882..cc89e23 100755
--- a/scripts/tcl_commands/lvs.tcl
+++ b/scripts/tcl_commands/lvs.tcl
@@ -99,6 +99,7 @@ proc write_powered_verilog {args} {
         --power-port $arg_values(-power) \
         --ground-port $arg_values(-ground) \
         --powered-netlist $arg_values(-powered_netlist) \
+        --ignore-missing-pins \
         $arg_values(-def)\
         |& tee $::env(TERMINAL_OUTPUT) $log_def
```

Is there a safe way to do this for a macro specifically? I couldn't find a relevant configuration variable or any mention in the docs.

### Expected Behavior

I expected an unpowered macro (no power/ground pins) to not cause an error.

### Environment report

```shell
Kernel: Linux v5.10.0-8-amd64
Distribution: debian
Python: v3.9.12 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 235fa7a4a2872e779588919c58fc4fa32568e075
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

235fa7a 2022-11-28T17:17:32+02:00 [BOT] Update PDK (#1516) - Openlane Bot -  (grafted, HEAD, tag: 2022.11.29)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

N/A

### Relevant log output

```shell
OpenROAD 7c85c140308f01b73f57ea1117f3e43f39abd437
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/tmp/merged.nom.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 60 technology vias
[INFO ODB-0225]     Created 230 library cells
[INFO ODB-0226] Finished LEF file:  /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/results/routing/bfg_mux_test_small.def
[INFO ODB-0128] Design: bfg_mux_test_small
[INFO ODB-0130]     Created 10 pins.
[INFO ODB-0131]     Created 425 components and 917 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 848 connections.
[INFO ODB-0133]     Created 18 nets and 69 connections.
[INFO ODB-0134] Finished DEF file: /home/aryap/src/bfg_mux_test_gf180mcu/openlane/bfg_mux_test_small/runs/22_12_09_20_44/results/routing/bfg_mux_test_small.def
Either power or ground (or both) pins not found for bfg_mux.
Top-level design name: bfg_mux_test_small
Found default power net 'vdd'
Found default ground net 'vss'
Found 1 power ports.
Found 1 ground ports.
[WARN] No pins in the LEF view of bfg_mux marked for use as power.
[WARN] Attempting to match power pin by name (using top-level port name) for bfg_mux.
[WARN] No pins in the LEF view of bfg_mux marked for use as ground.
[WARN] Attempting to match power pin by name (using top-level port name) for bfg_mux.
```

maliberty, 2022-12-10:
What is the purpose of an unpowered macro?

vijayank88, 2022-12-10:
similar issue found one of MPW CI design and reported GH issue: https://github.com/The-OpenROAD-Project/OpenLane/issues/1475

mattvenn, 2022-12-12:
could also have a use case for GDS art