The-OpenROAD-Project/OpenLane #1568: Provide a Spice model of final circuit that includes parasitics

growly, 2022-12-12:
### Description

It seems useful to provide a Spice model that includes extracted parasitics at the end of the flow. Proprietary tools readily make this available, and it makes sense to provide the parasitics in a format readily available for the user's simulations.

### Proposal

I don't know if existing tools in the flow can do this, but I [BigSpicy ](https://github.com/google/bigspicy) can. It can merge Verilog, Spice and SPEF netlists. It can also do things like extract capacitively-coupled networks, paths between given input/output nodes, and such.

It can easily be run as a (possibly optional) step in the flow to dump the merged Spice view for the user.

Ideally we wouldn't have to export to SPEF and Spice and then re-merge them, but absent another option this does work.

donn, 2022-12-12:
Absolutely chuffed that its name is "BigSpicy." We can integrate it, sure, that does sound incredibly useful.

maliberty, 2022-12-12:
Most blocks will be far too large to spice.  You might find write_path_spice in opensta more useful.

growly, 2022-12-12:
I don't agree. Without making any assumptions about what "most" blocks look like, consider that:

- Simulators like [Xyce](https://xyce.sandia.gov/) are designed to make full-chip simulation feasible
- Blocks which do not benefit from large scale Spice simulation do not need to use this feature
- Blocks which do benefit will gain greatly from this feature
- I have used BigSpicy and Xyce to analyse sub-blocks with success (I could simulate a floating point multiplier in a couple of minutes on my workstation)
- Last I checked, `write_path_spice` did not include parasitics

maliberty, 2022-12-13:
If this is useful then great.

write_path_spice does include parastics, otherwise it wouldn't useful.  That's not new.

growly, 2022-12-13:
My mistake, I misremembered why I couldn’t use write_path_spice. I needed a way to find all nets coupled to a path by some capacitance threshold and it couldn’t give me that.

It might be better to enable OpenSTA to dump the whole netlist with parasitics? But that would be more work