The-OpenROAD-Project/OpenLane #1574: [ERROR DPL-0036] Detailed placement failed -- error is not actionable

xobs, 2022-12-14:
### Description

When synthesizing a design, OpenLane produces the error `[ERROR DPL-0036] Detailed placement failed.`. It does not say why placement failed, nor does it explain how to address the error. It also doesn't report how to iterate to address the error, which is a problem when synthesis takes a lot of time.

### Expected Behavior

When a step fails, it should report *why* the step failed. Ideally it should also give suggestions on how to fix the issue.

### Environment report

```shell
user@Ondo:/opt/Si/work/caravel-gf180-pio$ python3 ./env.py issue-survey
python3: can't open file '/opt/Si/work/caravel-gf180-pio/./env.py': [Errno 2] No such file or directory
user@Ondo:/opt/Si/work/caravel-gf180-pio$
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10224306/issue_reproducible.tar.gz)

The repository is available at https://github.com/xobs/caravel-gf180-pio

### Relevant log output

```shell
[INFO]: Running Placement Resizer Design Optimizations (log: ../opt/Si/work/caravel-gf180-pio/openlane/wb_pio/runs/22_12_14_12_14/logs/placement/8-resizer.log)...
  [ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: ../opt/Si/work/caravel-gf180-pio/openlane/wb_pio/runs/22_12_14_12_14/logs/placement/8-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  _075635_
[INFO DPL-0035]  _075635_
[INFO DPL-0035]  _088150_
[INFO DPL-0035]  _087790_
[INFO DPL-0035]  _087459_
[INFO DPL-0035]  _088111_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 62 DPL-0036
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
```

xobs, 2022-12-14:
Note that there is https://github.com/The-OpenROAD-Project/OpenLane/issues/1010 which seems to have run into a similar problem. However, that issue is filled mostly with people attempting to debug the symptoms of the error. I've opened this Issue because the error itself has no meaning -- it's just a bunch of numbers with no context and no indication of why placement failed.

Possible solutions are:

* Print something about... `DIE_AREA`? `PL_TARGET_DENSITY`? `FP_CORE_UTIL`? I'm still not sure what these mean, or why there are two different knobs for die (what's a core and what's not-core?)
* Show a visualisation of the placement process so that a user can intuit where the failure is
* Suggest a different `SYNTH_STRATEGY` (does this affect things?)
* Suggest that maybe the design is too complicated to be routed

maliberty, 2022-12-14:
The simplest thing is to open the gui and look at where those instances are placed.  There are many possible causes and it hard to say what the exact cause is.

xobs, 2022-12-14:
Hmm... that would be a good thing to add to the output:

1. How to open the GUI
2. Which instances are causing the error
3. What to look for

maliberty, 2022-12-14:
I don't think an error message should be a how-to book.   The instances that are a problem are right there in the error message, as shown in the description above:
```
[INFO DPL-0035]  _075635_
[INFO DPL-0035]  _088150_
[INFO DPL-0035]  _087790_
[INFO DPL-0035]  _087459_
```
I would look at the area in which they are placed and how congested it is.  Is there an excess of buffering (perhaps bad constraints)?  Is it a channel between macros with limited space?  Is the utilization very high overall?  etc.

xobs, 2022-12-14:
I do think an error message should explain what the error is. The error I'm seeing is:
```
OpenROAD 7f00621cb612fd94e15b35790afe744c89d433a7 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 76 input buffers.
[INFO RSZ-0028] Inserted 97 output buffers.
[INFO RSZ-0058] Using max wire length 22815um.
[INFO RSZ-0039] Resized 16830 instances.
[INFO RSZ-0042] Inserted 143 tie gf180mcu_fd_sc_mcu7t5v0__tiel instances.
[INFO DPL-0034] Detailed placement failed on:
[INFO DPL-0035]  _063554_
[INFO DPL-0035]  _091720_
[INFO DPL-0035]  _055760_
...
[INFO DPL-0035]  _065550_
[INFO DPL-0035]  _074226_
[INFO DPL-0035]  _073498_
[INFO DPL-0035]  _073498_
[INFO DPL-0035]  _064475_
[INFO DPL-0035]  _064475_
[INFO DPL-0035]  _064475_
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 62 DPL-0036
```

What are those numbers? Are they wires? Are they cells? Are they logic elements? Are they indexes into some table? Are they distances? How do I cross-reference them with something useful?

The error message might not need to be a how-to book, but there should be something there that I can use with a search engine to be able to understand the problem better. Maybe a link to a page that describes how to open the GUI to investigate the error.

maliberty, 2022-12-14:
The placer works on instances so they are instance names.  The lousy names can be blamed on yosys; OR just uses whatever it gets as input.

xobs, 2022-12-14:
Alright, so it might make sense to rewrite the error as:

`Detailed placement failed on the following instances:`

ANd the last message should read:

`message limit reached, this message will no longer print -- there are %d remaining instances that were unable to be placed`

Additionally, I should open an Issue with Yosys to try to get them to improve the names that are generated.

You mentioned that it's possible to open the GUI to inspect things. Can you show me where that's documented? The documentation at https://openroad.readthedocs.io/en/latest/main/src/gui/README.html#add-buttons-to-the-toolbarhttps://openroad.readthedocs.io/en/latest/main/src/gui/README.html#add-buttons-to-the-toolbar seems to just go over the TCL interface without mentioning any of the following:

* What sorts of files does OpenROAD support viewing?
* How can you open files in OpenROAD?
* Which files should be inspected in order to troubleshoot the issue?

vijayank88, 2022-12-14:
@xobs which mpw tag you're cloned and running this project?

xobs, 2022-12-14:
Collecting the issues:

For OpenLane:

* `PL_BASIC_PLACEMENT` should have an explanation of what it does and what are the implications of skipping it
* `PL_TARGET_DENSITY` should have documentation on "vaguely" what acceptable parameters there are
* A check should be added to ensure `VDD_NETS` and `GND_NETS` are valid
* Documentation should be added to show how to use the GUI (how does one generate the image you made?)

For OpenROAD:

* The error message should indicate that the entity names that follow are the names of instances that were unable to be placed
* The error message should indicate *why* they were unable to be placed (did it exceed the number of interations?)
* The error message should indicate how many additional entities were unable to be placed

For Yosys:

* The generated net names should be improved

vijayank88, 2022-12-14:
Replace with following `config.json` and try it
```
{
    "DESIGN_NAME": "wb_pio",
    "DESIGN_IS_CORE": 0,
    "VERILOG_FILES": ["dir::../../verilog/rtl/defines.v", "dir::../../verilog/rtl/wb_pio_top.v", "dir::../../verilog/rtl/fpga_pio/src/*.v"],
    "CLOCK_PERIOD": 10,
    "CLOCK_PORT": "wb_clk_i",
    "CLOCK_NET": "pio1.clk",
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 1900 1600",
    "FP_PIN_ORDER_CFG": "dir::pin_order.cfg",
    "PL_BASIC_PLACEMENT": 0,
    "PL_TARGET_DENSITY": 0.50,
    "SYNTH_NO_FLAT": 0,
    "VDD_NETS": ["vdd"],
    "GND_NETS": ["vss"],
    "DIODE_INSERTION_STRATEGY": 4,
    "RUN_CVC": 0,
    "SYNTH_STRATEGY": "DELAY 4",
    "pdk::sky130*": {
        "FP_CORE_UTIL": 75,
        "RT_MAX_LAYER": "met4",
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "SYNTH_MAX_FANOUT": 5
        },
        "scl::sky130_fd_sc_ms": {
            "CLOCK_PERIOD": 10
        }
    },
    "pdk::gf180mcuC": {
        "STD_CELL_LIBRARY": "gf180mcu_fd_sc_mcu7t5v0",
        "CLOCK_PERIOD": 24.0,
        "FP_CORE_UTIL": 20,
        "RT_MAX_LAYER": "Metal4",
        "SYNTH_MAX_FANOUT": 4
    }
}
```

xobs, 2022-12-14:
That `config.json` does get to clock tree synthesis, but fails with the same sort of error (Detailed placement failed.).

Is there any documentation on how to get that nice graphic you showed? A search of https://openlane.readthedocs.io/en/latest/ does not yield any documentation on the `-gui` option.

vijayank88, 2022-12-14:
Use `openroad -gui` and load odb file with `read_db <path-to>/name.odb`

xobs, 2022-12-14:
I've been updating https://github.com/The-OpenROAD-Project/OpenLane/issues/1574#issuecomment-1350453510 with issues and pull requests that I've been following.

The specific thing that was missing was `read_db`, which is never called out as being the GUI's answer to the question "How do I open a file?" so I've added documentation to the README to explicitly call that out.

maliberty, 2022-12-14:
The message limit is a property of the logger and is not specific to a particular message.  We could add a total number to the message.

maliberty, 2022-12-14:
Message improved in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2626

maliberty, 2022-12-14:
"The error message should indicate why they were unable to be placed" is impractical and can't be implemented

maliberty, 2022-12-14:
You can run flow.tcl with -gui to open the gui.

vijayank88, 2022-12-17:
@xobs Are you able to complete flow?

xobs, 2022-12-18:
I haven't, but that isn't related to the issues I've opened -- the routing failure is due to a design problem, so now I can work on simplifying the Verilog to get it to route.

kareefardi, 2023-01-29:
I think this is completed no

closed 2023-01-29 by kareefardi