The-OpenROAD-Project/OpenLane #1579: fix the issue that's causing pdk mismatches

mattvenn, 2022-12-14:
### Description

To use openlane with mpw8 tagged tools I have to run with -ignore_mismatches otherwise the flow fails with a warning about matching pdk.

As the tools were installed following the instructions, either this is a problem and should be fixed, or it's not a problem and the warning should be removed.

radhe-2022, 2022-12-16:
Im also facing the same issue.. @mattvenn @proppy @vijayank88
[image]

proppy, 2022-12-16:
@mattvenn shouldn't we also file an issue against https://github.com/efabless/caravel_user_project ?

As the `OPEN_PDK_COMMITS` they use: https://github.com/efabless/caravel_user_project/blob/main/Makefile#L53 is different from the one blessed in https://github.com/The-OpenROAD-Project/OpenLane/blob/master/dependencies/tool_metadata.yml#L71 ?

Baungarten-CINVESTAV, 2022-12-16:
To remove that warning you need to install volare from https://github.com/efabless/volare
One time volare is installed run:

`volare enable --pdk sky130 3af133706e554a740cfe60f21e773d9eaa41838c`

Verify the installation with:
`volare ls --pdk sky130`

after that just run your project

kareefardi, 2022-12-19:
The tools were installed per caravel user project instructions. The mismatch is caused by caravel user project installing a different version of the pdk than the one recommend in the openlane version used.

closed 2022-12-19 by kareefardi

mattvenn, 2022-12-19:
> @mattvenn shouldn't we also file an issue against https://github.com/efabless/caravel_user_project ?
>
> As the `OPEN_PDK_COMMITS` they use: [...]

yes, I probably should have opened the issue there rather than here.