The-OpenROAD-Project/OpenLane #1582: process got killed for some unknown reason when trying to synthesize the Small BOOM core

JiaDYuan, 2022-12-17:
### Description

the synthesis process of small BOOM core got killed for unknown reason:  child killed: kill signal

### Expected Behavior

I want to synthesize the small BOOM core successfully through OpenLane.

### Environment report

```shell
Kernel: Darwin v22.1.0
Distribution: macOS 13.0.1
Python: v3.10.9 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 90d369bcce5cf1133c28291bc3c8b6473a19ff62
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (grafted, HEAD -> master, tag: 2022.12.14, origin/master, origin/HEAD)
---
Git Remotes

origin	http://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	http://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[DigitalTop.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10251532/DigitalTop.zip)

./flow.tcl -design DigitalTop -tag synthesis_only -to synthesis -overwrite

### Relevant log output

```shell
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/DigitalTop/runs/synthesis_only/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_596 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_593 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_563 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_558 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_555 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_544 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_539 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_536 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_525 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_520 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_517 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_506 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_501 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_498 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_487 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_468 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_432 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._l_mask_mask_T_105 } = 20'00010001001001001001
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_563 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_558 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_555 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_544 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_539 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_536 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_525 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_520 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_517 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_506 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_501 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_498 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_487 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_468 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_432 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._l_mask_mask_T_105 } = 18'000001001001001001
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_615 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_563 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_558 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_555 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_544 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_539 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_536 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_525 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_520 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_517 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_506 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_501 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_498 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_487 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_468 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_432 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._l_mask_mask_T_105 } = 19'0100001001001001001
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_615 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_577 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_574 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_563 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_558 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_555 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_544 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_539 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_536 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_525 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_520 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_517 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_506 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_501 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_498 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_487 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_468 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_432 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._l_mask_mask_T_105 } = 21'010001001001001001001
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_577 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_574 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_563 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_558 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_555 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_544 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_539 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_536 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_525 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_520 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_517 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_506 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_501 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_498 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_487 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_468 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_432 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._l_mask_mask_T_105 } = 20'00001001001001001001
      Activation pattern for cell $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$shl$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:290177$61616: { \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_601 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_596 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_593 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_582 \tile_prci_domain.tile_reset_domain.boom_tile.lsu._T_577 \tile_prci_doma [... cut ...] enlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:291736$63459_Y $flatten\tile_prci_domain.\tile_reset_domain.\boom_tile.\lsu.$eq$/openlane/designs/DigitalTop/src/chipyard.TestHarness.SmallBoomConfig.top.v:289253$60661_Y \tile_prci_domain.tile_reset_domain.boom_tile.lsu.commit_load \tile_prci_domain.tile_reset_domain.boom_tile.lsu.commit_store \tile_prci_domain.tile_reset_domain.boochild killed: kill signal

[ERROR]: Creating issue reproducible...
[ERROR]: Flow failed.
```

vijayank88, 2022-12-17:
@JiaDYuan 
Have you tried to include sram as `VERILOG_FILES_BLACKBOX`?

JiaDYuan, 2022-12-17:
@vijayank88  yes, it is included.

{
    "DESIGN_NAME": "DigitalTop",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 25.0,
    "DESIGN_IS_CORE": true,

    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 2000 3750",
    "PL_TARGET_DENSITY": 0.5,

    "VDD_NETS": "vccd1",
    "GND_NETS": "vssd1",

    
    "MACRO_PLACEMENT_CFG": "dir::macro_placement.cfg",

    "EXTRA_LEFS":      "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/lef/sky130_sram_1kbyte_1rw1r_32x256_8.lef",
    "EXTRA_GDS_FILES": "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/gds/sky130_sram_1kbyte_1rw1r_32x256_8.gds",
    "EXTRA_LIBS":      "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/lib/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib",

    "RUN_KLAYOUT_XOR": false,
    "MAGIC_DRC_USE_GDS": false,
    "QUIT_ON_MAGIC_DRC": false
}

vijayank88, 2022-12-17:
I see only `VERILOG_FILES` pointing all `src/*.v`
Not found `VERILOG_FILES_BLACKBOX` this variable in `config.json`

JiaDYuan, 2022-12-17:
yes, now I include the blackbox and re-run it. now I got this error messages:

wrong # args: should be "set varName ?newValue?"
    while executing
"set ::env(VERILOG_ FILES_BLACKBOX) {/openlane/designs/DigitalTop/bb/sky130_sram_1kbyte_1rw1r_32x256_8.v}"
    (file "/openlane/designs/DigitalTop/runs/synthesis_only/config_in.tcl" line 20)
    invoked from within
"source $config_in_path"
    (procedure "source_config" line 70)
    invoked from within
"source_config -run_path $run_path $::env(DESIGN_CONFIG) -expose $arg_values(-expose_env)"
    (procedure "prep" line 201)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401

JiaDYuan, 2022-12-17:
@vijayank88 
I include sram now,: 
{
    "DESIGN_NAME": "DigitalTop",
    "VERILOG_FILES": "dir::src/*.v",
    "CLOCK_PORT": "clk",
    "CLOCK_PERIOD": 25.0,
    "DESIGN_IS_CORE": true,

    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 2000 3750",
    "PL_TARGET_DENSITY": 0.5,

    "VDD_NETS": "vccd1",
    "GND_NETS": "vssd1",

    
    "MACRO_PLACEMENT_CFG": "dir::macro_placement.cfg",

    "EXTRA_LEFS":      "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/lef/sky130_sram_1kbyte_1rw1r_32x256_8.lef",
    "EXTRA_GDS_FILES": "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/gds/sky130_sram_1kbyte_1rw1r_32x256_8.gds",
    "EXTRA_LIBS":      "/openlane/pdks/sky130B/libs.ref/sky130_sram_macros/lib/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib",
     "VERILOG_FILES_BLACKBOX": "dir::bb/*.v",
    "RUN_KLAYOUT_XOR": false,
    "MAGIC_DRC_USE_GDS": false,
    "QUIT_ON_MAGIC_DRC": false
}

But  I got this error :

wrong # args: should be "set varName ?newValue?"
    while executing
"set ::env(VERILOG_ FILES_BLACKBOX) {/openlane/designs/DigitalTop/bb/sky130_sram_1kbyte_1rw1r_32x256_8.v}"
    (file "/openlane/designs/DigitalTop/runs/synthesis_only/config_in.tcl" line 20)
    invoked from within
"source $config_in_path"
    (procedure "source_config" line 70)
    invoked from within
"source_config -run_path $run_path $::env(DESIGN_CONFIG) -expose $arg_values(-expose_env)"
    (procedure "prep" line 201)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 12)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)

JiaDYuan, 2022-12-17:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10251639/issue_reproducible.zip)

maliberty, 2022-12-18:
Your reproducible doesn't work:
```
Successfully finished Verilog frontend.
ERROR: Can't open input file `openlane/designs/DigitalTop/src/ClockDividerN.v' for reading: No such file or directory
```

vijayank88, 2022-12-18:
@maliberty 
Its bug from `issue_reproducible`. Created recursive folders for verilog source file names.

@JiaDYuan Can you share updated `config.json` will try again at my end?

JiaDYuan, 2022-12-18:
@vijayank88 
[config.json.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10253170/config.json.zip)

vijayank88, 2022-12-18:
@JiaDYuan 
In the `DigitalTop.zip` I've only `src` directory. 
I'm not found `bb/*.v`. Can you re-attach the `DigitalTop.zip` again with updated files?

JiaDYuan, 2022-12-18:
[DigitalTop.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10253325/DigitalTop.zip)
@vijayank88

JiaDYuan, 2022-12-18:
So it is the problem of syntax error? I tried to synthesize the design in the Vivado, it was synthesized successfully.

vijayank88, 2022-12-20:
Even ORFS unable to finish synthesis stage.
```
10.14. Executing ALUMACC pass (create $alu and $macc cells).
10.15. Executing SHARE pass (SAT-based resource sharing).
Elapsed time: 2:49:35[h:]min:sec. CPU time: user 9786.52 sys 377.98 (99%). Peak memory: 124899820KB.
make: *** [results/sky130hd/DigitalTop/base/1_1_yosys.v] Error 9
```
@JiaDYuan 
Just for confirmation have you replaced all register into fixed SRAM macro in source RTL?

vijayank88, 2022-12-21:
@JiaDYuan 
Reply got from yosys team
```
Seems to be running out of memory. Try running synthesis without flattening, and make sure the caches are being mapped to SRAM.
```
Can you try the same?

JiaDYuan, 2022-12-21:
@vijayank88  
Yes, I have set the synthesis without the Flattening.   "SYNTH_NO_FLAT": true.
And now I am checking if all the caches and memories are mapped to SRAM. But it is not easy due to the large scale of BOOM core.

closed 2023-04-13 by donn