The-OpenROAD-Project/OpenLane #1583: Cli argument `-lvs` is broken

marwaneltoukhy, 2022-12-19:
### Description

Trying to only run LVS on a design that was previously hardened breaks the flow. command used:
```
./flow.tcl -lvs -design APU -gds designs/APU/runs/RUN_2022.12.19_10.20.21/results/signoff/APU.gds -net designs/APU/runs/RUN_2022.12.19_10.20.21/tmp/signoff/29-APU.pnl.v
```

### Expected Behavior

Run LVS on gds vs netlist

### Environment report

```shell
Kernel: Linux v5.15.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: improve-yosys-naming-dev
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

78015a1 2022-12-14T21:12:51+08:00 scripts: yosys: generate readable names for instances - Sean Cross -  (HEAD -> improve-yosys-naming)
90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (tag: 2022.12.14)
0d13487 2022-12-13T19:05:19+02:00 Add `report_parasitic_annotation` to sta (#1569) - Kareem Farid -  ()
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

```
./flow.tcl -lvs -design APU -gds designs/APU/runs/RUN_2022.12.19_10.20.21/results/signoff/APU.gds -net designs/APU/runs/RUN_2022.12.19_10.20.21/tmp/signoff/29-APU.pnl.v
```

### Relevant log output

```shell
[ERROR]: during executing: "openroad -exit -no_init -python /openlane/scripts/odbpy/power_utils.py write_powered_def --output /openlane/designs/APU/runs/RUN_2022.12.19_10.36.55/tmp/signoff/1-APU.p.def --input-lef /openlane/designs/APU/runs/RUN_2022.12.19_10.36.55/tmp/merged.nom.lef --power-port VDD --ground-port VSS --powered-netlist {} 0 |& tee /dev/null /openlane/designs/APU/runs/RUN_2022.12.19_10.36.55/logs/signoff/2-write_powered_def.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/odbpy/power_utils.py", line 84, in write_powered_def
    print(f"Top-level design name: {reader.name}")
AttributeError: 'OdbReader' object has no attribute 'name'
child process exited abnormally
```
```

d-m-bailey, 2022-12-19:
@marwaneltoukhy The `flow.tcl` script uses the `<design>/config.tcl` (or maybe `<design>/config.json`) file to retrieve many of the parameters for LVS. It was not designed for standalone LVS of gds and verilog outside of the openlane flow.

Is there an `APU/config.*` file with the correct parameters?

marwaneltoukhy, 2022-12-19:
@d-m-bailey Yes it does, this is a design in openlane. The documentation for this option [can be found here ](https://github.com/The-OpenROAD-Project/OpenLane/blob/5c58d413a3e6ddf134a0d89a96b25c025a6f47d1/docs/source/reference/cli.md)

d-m-bailey, 2022-12-19:
@marwaneltoukhy I believe I'm the one that originally added the lvs subroutine. There were changes later. 
The error messages look like it's trying to add the power nets which it will attempt to do if `LVS_INSERT_POWER_PINS` is set. If your verilog already has the power nets included, you want to set this to `0` or delete it (maybe).

See the `run_lvs` routine of `scripts/tcl_commands/lvs.tcl`

closed 2022-12-21 by marwaneltoukhy