The-OpenROAD-Project/OpenLane #1588: LVS errors, no matching pin & no matching net

Baungarten-CINVESTAV, 2022-12-22:
### Description

Hi, I am trying to harden a small eFPGA (4 x 4 CLB), I have used OpenFPGA to create the FPGA architecture.
This FPGA has basic modules like CLB, switching blocks, and connection blocks. To harden the whole FPGA, I first harden each module separately and then try to harden the whole FPGA but get the following error.

[STEP 33]
[INFO]: Running LVS (log: designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log)...
[ERROR]: There are LVS errors in the design: See 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/reports/signoff/33-fpga_top.lvs.rpt' for a summary and 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log' for details.

I previously opened a issue related to routing violations but that error was solved, now the problem is about LVS.

I hope you can help me, I want to submit this project to the MPW-8.

Thanks in advance

### Expected Behavior

[STEP 33]
[INFO]: Running LVS (log: designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log)...
[ERROR]: There are LVS errors in the design: See 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/reports/signoff/33-fpga_top.lvs.rpt' for a summary and 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log' for details.
[INFO]: Saving current set of views in 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

### Environment report

```shell
python3 ./env.py issue-survey
Kernel: Linux v5.15.0-56-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: 931398e1d869d6b6863e1406494b641c04bdfc31
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

931398e 2022-12-21T15:48:06+02:00 Fixes for lvs_batch (#1584) - Kareem Farid -  (HEAD -> master, origin/master, origin/HEAD)
245e625 2022-12-15T12:30:59+02:00 [BOT] Update magic (#1558) - Openlane Bot -  (tag: 2022.12.16)
90d369b 2022-12-13T22:59:28+02:00 Iterate over `$libs` in read_libs (#1570) - Kareem Farid -  (tag: 2022.12.14)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

1. `make  mount`
2. `./flow.tcl -design fpga_top`
[fpga_top.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10289700/fpga_top.zip)
[logs.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/10289706/logs.zip)

### Relevant log output

```shell
[STEP 33]
[INFO]: Running LVS (log: designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log)...
[ERROR]: There are LVS errors in the design: See 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/reports/signoff/33-fpga_top.lvs.rpt' for a summary and 'designs/fpga_top/runs/RUN_2022.12.22_17.58.17/logs/signoff/33-lvs.lef.log' for details.
-----------------------------------------------------------------------------------------------

File 33-fpga_top.lef.lvs.log

top_left_grid_right_width_0_height_0_subti |top_left_grid_right_width_0_height_0_subti 
top_left_grid_right_width_0_height_0_subti |top_left_grid_right_width_0_height_0_subti 
top_right_grid_left_width_0_height_0_subti |top_right_grid_left_width_0_height_0_subti 
top_right_grid_left_width_0_height_0_subti |top_right_grid_left_width_0_height_0_subti 
vccd1                                      |(no matching pin)                          
vssd1                                      |(no matching pin)                          
---------------------------------------------------------------------------------------
File 33-fpga_top.lef.lvs.log

Net: gfpga_pad_GPIO_PAD[28]                |Net: gfpga_pad_GPIO_PAD[28]                
  sky130_fd_sc_hd__buf_6/A = 1             |  sky130_fd_sc_hd__buf_6/A = 1             
  grid_io_top/gfpga_pad_GPIO_PAD[3] = 1    |  grid_io_top/gfpga_pad_GPIO_PAD[3] = 1    
  sky130_fd_sc_hd__diode_2/DIODE = 1       |  sky130_fd_sc_hd__diode_2/DIODE = 1       
                                           |  sky130_fd_sc_hd__buf_6/X = 1             
                                           |                                           
Net: net25                                 |Net: net25                                 
  sky130_fd_sc_hd__buf_12/A = 3            |  sky130_fd_sc_hd__buf_12/A = 3            
  sky130_fd_sc_hd__buf_12/X = 1            |  sky130_fd_sc_hd__buf_12/X = 1            
                                           |                                           
Net: PHY_1548/VPB                          |(no matching net)                          
  sky130_fd_sc_hd__decap_3/VPB = 1         |                                           
                                           |                                           
Net: PHY_15961/VPB                         |(no matching net)                          
  sky130_fd_sc_hd__decap_3/VPB = 1         |                                           
                                           |                                           
Net: PHY_1579/VPB                          |(no matching net)
```

kareefardi, 2023-01-22:
There are a couple of adjustments needed in order to complete your design LVS clean:

- In pdn log file, you will notice a warning about instances not connected. Following this, you will notice that `submodule.` prefix in `FP_PDN_MACRO_HOOKS` should be removed. 
- I assume you are going for macros only in your design, since you have `RUN_FILL_INSERTION` disabled in your configuration, however you would also need to disable tap/decap insertion, diode insertion,  design optimizations, clock tree synthesis and pdn rails for stdcells through:
  ```
    "PL_RESIZER_DESIGN_OPTIMIZATIONS": 0,
    "PL_RESIZER_TIMING_OPTIMIZATIONS": 0,
    "CLOCK_TREE_SYNTH": 0,
    "DIODE_INSERTION_STRATEGY": 0,
    "FP_PDN_ENABLE_RAILS": 0,
  ```
- You also need to disable the following checks (still investigating):
  ```
    "FP_PDN_ENABLE_RAILS": 0,
    "FP_PDN_CHECK_NODES": 0,
  ```
- Finally, `vccd1` and `vssd1` are declared in rtl as design pins, which will create additional pins at the boundary so they need to be removed.

Baungarten-CINVESTAV, 2023-01-22:
Ok, I will give it a try.

- Just one question, in the [OL documentation](https://openlane.readthedocs.io/en/latest/usage/advanced_power_grid_control.html?highlight=vccd1) in the **Macro Level** section, it says the following:

[image]

- so is the documentation wrong? or when can I or can I not add the power ports?

kareefardi, 2023-01-22:
If you wish to include it in the rtl, then it should be guarded with
```
`ifdef USE_POWER_PINS 
`endif
```

Baungarten-CINVESTAV, 2023-02-07:
Sorry to take much time to answer, I made multiple changes in each macro and took much care in voltage and ground connection, finally, I implemented the FPGA into the user_project_wrapper.

the precheck was successful but the following warning appears, how bad is that?
[image]

vijayank88, 2023-02-07:
@Baungarten-CINVESTAV you can skip that warning as it passed important checks already

closed 2023-02-07 by Baungarten-CINVESTAV