The-OpenROAD-Project/OpenLane #1593: RUN_KLAYOUT generates invalid command line

xobs, 2022-12-26:
### Description

When `magic` crashes (e.g. https://github.com/The-OpenROAD-Project/OpenLane/issues/1587), it can be useful to try using `klayout` by setting `RUN_MAGIC=0` and `RUN_KLAYOUT=1` in the config file. However, OpenLane appears to generate invalid command line arguments to klayout, and fails to run.

### Expected Behavior

klayout should run normally

### Environment report

```shell
Kernel: Linux v5.15.79.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

cb59d1f 2022-11-18T18:42:38+02:00 fix return values in `equally_spaced_sequence` (#1503) - Kareem Farid -  (grafted, HEAD, tag: 2022.11.19)
---
Git Remotes

origin  git@github.com:The-OpenROAD-Project/OpenLane (fetch)
origin  git@github.com:The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

There is no reproduction material (see Relevant log output) -- there is no "Reproducible packaged" message.

### Relevant log output

```shell
[STEP 23]
[INFO]: Streaming out GDSII with KLayout (log: ../si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/logs/signoff/23-gdsii-klayout.log)...
[ERROR]: during executing: "klayout -b -rm /openlane/scripts/klayout/def2gds.py -rd out_gds=/si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/results/signoff/wb_pio.klayout.gds -rd tech_file=/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.tech/klayout/gf180mcuC.lyt -rd design_name=wb_pio -rd in_def=/si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/results/routing/wb_pio.def -rd {in_gds=/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/gds/gf180mcu_fd_sc_mcu7t5v0.gds } -rd config_file= -rd seal_gds= -rd lef_file=/si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/tmp/merged.nom.lef |& tee /dev/null /si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/logs/signoff/23-gdsii-klayout.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:

Input: /si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/results/routing/wb_pio.def
Output: /si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/results/signoff/wb_pio.klayout.gds
Design: wb_pio
Technology File: /si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.tech/klayout/gf180mcuC.lyt
GDS File List: ['/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/gds/gf180mcu_fd_sc_mcu7t5v0.gds']
LEF File: /si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/tmp/merged.nom.lef

Unable to open file: /si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.tech/klayout/gf180mcuC.lyt (errno=2) in Technology.load
child process exited abnormally

[INFO]: Saving current set of views in '../si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../si/work/caravel-gf180-pio/openlane/wb_pio/runs/gf180mcuC-2022-12-26_16_08/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
make[1]: *** [Makefile:73: wb_pio] Error 255
make[1]: Leaving directory '/si/work/caravel-gf180-pio/openlane'
make: *** [Makefile:107: wb_pio] Error 2

real    39m51.943s
user    0m0.115s
sys     0m0.348s
$
```

xobs, 2022-12-26:
The following changes need to be made, after which klayout runs successfully:

1. The script passes `in_gds` as `-rd {in_gds=/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/gds/gf180mcu_fd_sc_mcu7t5v0.gds }`, but the curly braces seem like they should be omitted
2. The technology file `/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.tech/klayout/gf180mcuC.lyt` does not exist -- the path should be `/si/work/caravel-gf180-pio/deps/pdks/gf180mcuC/libs.tech/klayout/tech/gf180mcu.lyt` (i.e. add a `/tech/` directory and remove the `C`)

xobs, 2023-01-10:
Is there a standard that should be followed that gf180mcu is not adhering to? Or is it more along the lines of sky130 is the de-facto standard and I should reference that in the issue?

donn, 2023-01-10:
It doesn't actually matter: OpenLane looks for the lyt and lyp wherever the `libs.tech/openlane/config.tcl` variables tell it to. That just means that variable is set incorrectly for gf180.

proppy, 2023-01-10:
I think we also need to make sure that the internally reference fro mthe `lyt` to the `lyp` file also conform to the final open_pdks layout, see:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/74e4ec59b55bcf5be2f153abff8519d15ebe21fa/tech/klayout/gf180mcu.lyt#L23-L24

donn, 2023-01-16:
Resolved via #1609

closed 2023-01-16 by donn

proppy, 2023-02-15:
Related https://github.com/RTimothyEdwards/open_pdks/issues/346