The-OpenROAD-Project/OpenLane #1598: set_io_pin_constraint should be able to constrain ordering of pins

oharboe, 2022-12-29:
### Description

It should be possible to specify ordering of pins on an edge using ```set_io_pin_constraint```

### Proposal

Idea on how to express ordering, add a parameter

```
set_io_pin_constraint -region up:* -pin_names foo -order clockwise
set_io_pin_constraint -region up:* -pin_names foo -order anti-clockwise
```

eder-matheus, 2023-01-02:
@oharboe Just to clarify, the pins don't need to be adjacent, right?

oharboe, 2023-01-02:
> @oharboe Just to clarify, the pins don't need to be adjacent, right?

What I'm really trying to do is illustrated in this pull request with the designs/asap7/mock-array-big/config.mk.

I'm creating an array of macros and I want the pins to align to the neighbouring macros: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/743

In the example, all the macros are identical, but that's just for simplification, the macros could have identical routing, but different content/function.

What I want is to have a symmetry in the design where I can create macros that are placed next to eachother and where the routing is done at the edges of the macros only, no separate routing layer on top of the macros as the routing is done inside the macros.

oharboe, 2023-01-02:
So in light of the above, pins should not be ordered, but paired up in x or y position.

A named pin on the left should have the same y position as a paired pin on the right.

Same for top/bottom.

So perhaps an argument with an list of pairs to names?

eder-matheus, 2023-01-02:
This sounds like an easy update. I can devise a quick implementation for you to test and check if it satisfies your requirements.

oharboe, 2023-01-03:
> This sounds like an easy update. I can devise a quick implementation for you to test and check if it satisfies your requirements.

Great!

The next thing I'm going to ask is how to be able to access the layout of pins in one macro and use it in other macros. The macros are not all of the same type, or shape.

An illustration of possibilities, not a real example, would be:

```
AAAA
AAAA
AAAA
BBBB
CCCC
DDEE
FFFFF
```

Here the edges that must match are:

A left-right B
A top-bottom A
A bottom-top B
B left-right B
B bottom-top C
C bottom-top D
C bottom-top E
D bottom - top F
E bottom - top F

eder-matheus, 2023-01-03:
@oharboe Could you try this branch: https://github.com/eder-matheus/OpenROAD/tree/ppl_mirrored_pins? You can check a sample on how to use the new option for the constraints command here: https://github.com/eder-matheus/OpenROAD/blob/ppl_mirrored_pins/src/ppl/test/add_constraint9.tcl#L6

I choose the name "-mirrored_pins" for now. If you have a suggestion for a better name, please let me know.

oharboe, 2023-01-03:
I tried, but got an error:

eder-matheus, 2023-01-03:
Could you provide a testcase with make io_placement_issue?

oharboe, 2023-01-03:
> Could you provide a testcase with make io_placement_issue?

[io_placement_mock-array-big_Element_asap7_base_2023-01-03_19-58.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10339447/io_placement_mock-array-big_Element_asap7_base_2023-01-03_19-58.tar.gz)

eder-matheus, 2023-01-05:
@oharboe Sorry for my delay, but the error should be fixed now. Could you try the same branch again? I'm also looking at some weird behavior where some mirrored pins don't look optimized. I'll keep you updated on it too.

eder-matheus, 2023-01-05:
@oharboe FYI, I've fixed the optimization issue. You should have good results now, and your testcase is not failing anymore.

oharboe, 2023-01-06:
> @oharboe FYI, I've fixed the optimization issue. You should have good results now, and your testcase is not failing anymore.

Segfault:

```
Using 2 tracks default min distance between IO pins.
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = ppl::Slot; _Alloc = std::allocator<ppl::Slot>; std::vector<_Tp, _Alloc>::reference = ppl::Slot&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x0000562F8D071B9B in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F6CE423BCF0 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x0000562F8D06FBA8 in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# ppl::HungarianMatching::getFinalAssignment(std::vector<ppl::IOPin, std::allocator<ppl::IOPin> >&, std::unordered_map<odb::dbBTerm*, odb::dbBTerm*, std::hash<odb::dbBTerm*>, std::equal_to<odb::dbBTerm*>, std::allocator<std::pair<odb::dbBTerm* const, odb::dbBTerm*> > >&, bool) const in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# ppl::IOPlacer::findPinAssignment(std::vector<ppl::Section, std::allocator<ppl::Section> >&) in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# ppl::IOPlacer::run(bool) in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# 0x0000562F8D1F77F7 in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007F6CE874D028 in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
14# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# main in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# 0x00007F6CE4223510 in /lib/x86_64-linux-gnu/libc.so.6
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in /home/oyvind/ascenium/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 6
Elapsed time: 0:01.06[h:]min:sec. CPU time: user 0.97 sys 0.03 (94%). Peak memory: 180736KB.
make: *** [Makefile:497: results/asap7/mock-array-big_Element/base/3_2_place_iop.odb] Error 134
```

Updated issue file:

[io_placement_mock-array-big_Element_asap7_base_2023-01-06_08-47.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/10358480/io_placement_mock-array-big_Element_asap7_base_2023-01-06_08-47.tar.gz)

oharboe, 2023-01-06:
Another thing: I think it would be better to have a syntax that is more Tcl idiomatic, e.g. a list of pairs.

If we look at other commands:

```
add_pdn_connect -grid {top} -layers {M1 M2}
```

Here the list is a Tcl list.

So the mirror pins would be something like:

```
-mirror_pins {a b c d}
```

This would simplify the implementation, no need for regex to parse the arguments. Also, it simplifies the calling code, such as in my case.

eder-matheus, 2023-01-06:
@oharboe, I've updated the branch with the syntax you suggested for the option. I can't reproduce the segfault on my end with the attached testcase. Are you using the env var `PLACE_PINS_ARGS` in your run? It was not loaded on the .tar file.

oharboe, 2023-01-06:
> @oharboe, I've updated the branch with the syntax you suggested for the option. 

Thanks!

>I can't reproduce the segfault on my end with the attached testcase. Are you using >the env var `PLACE_PINS_ARGS` in your run? It was not loaded on the .tar file.

No, I'm not using PLACE_PIN_ARGS, it is not as flexible as a .tcl script.

eder-matheus, 2023-01-06:
@oharboe Running the flow in a CentOS machine doesn't fail during pin placement. It is currently in detailed routing, with about 200 violations in the last iterations. Could you try it again with the latest commit in my branch?

oharboe, 2023-01-06:
> @oharboe Running the flow in a CentOS machine doesn't fail during pin placement. It is currently in detailed routing, with about 200 violations in the last iterations. Could you try it again with the latest commit in my branch?

Did u modify my code to match the new syntax?

eder-matheus, 2023-01-06:
> Did u modify my code to match the new syntax?

Yes, I did.

oharboe, 2023-01-06:
Now... soon as I have this, I want to copy the layout of the pins to a different macro that has the same interface, but different contents.

I would be very interested in a .tcl snippet to do so or at least some pointers on how to do so...

eder-matheus, 2023-01-06:
> Now... soon as I have this, I want to copy the layout of the pins to a different macro that has the same interface, but different contents.
> 
> I would be very interested in a .tcl snippet to do so or at least some pointers on how to do so...

I'm not sure if I understand your scenario. Do you have something like a DEF file where your different macro is described? If that's the case, you can write a DEF and copy the PINS section to the DEF of your macro.

oharboe, 2023-01-06:
> I'm not sure if I understand your scenario. Do you have something like a DEF file where your different macro is described? If that's the case, you can write a DEF and copy the PINS section to the [...]

I have two macros I'm making in OpenROAD and I want the pin layouts to be identical, because I want to use them in an array.

Also, I'd like to automate the layout.

How can I use OpenROAD to automate the layout?

Persumably I can load the .odb file for the first macro in the .tcl file that places pins in the second?

maliberty, 2023-01-09:
You could use read_def -floorplan_initialize to copy over the data from one block to another.

oharboe, 2023-01-09:
> You could use read_def -floorplan_initialize to copy over the data from one block to another.

I can't just copy everything, if you see in https://github.com/The-OpenROAD-Project/OpenLane/issues/1598#issuecomment-1369429072, I need to copy the positions of pins on horizontal or vertical edges.

Should I be able to write a bit of .tcl to do that?

maliberty, 2023-01-09:
You could use the apis from db.h (in tcl) to get/set the pin posititions.

oharboe, 2023-01-09:
@eder-matheus So, then I think this feature is complete. I should be able to copy the mirrored pins from one macro in the vertical or horizontal direction.

As a separate feature, it would be nice to be able to specify the ordering of the pins.

oharboe, 2023-01-09:
> > @eder-matheus So, then I think this feature is complete. I should be able to copy the mirrored pins from one macro in the vertical or horizontal direction.
> > As a separate feature, it would be nice to be able to specify the ordering of the pins.
> 
> Okay, I'll merge the mirrored pins option and start working on ordering pins. It may take a while since it is more complex, but I'll keep you updated.

Excellent!

donn, 2023-01-10:
I'll move this to OR as this is mostly OR-related for now- when the feature's ready, can you just ping me? I'll merge OR and try to surface this in OL.